libretro/Lakka-LibreELEC #677: Esperanza EGG107K TROOPER gamepad not working

dudaerich, 2019-06-13:
- Which version of Lakka are you using?
2.2.2 but I tested it also on nightly build from 2019-06-12

- What system hardware are you using?
General PC x86_64

- What did you do?
Plugged in the gamepad.

- What did you expect to happen?
That I will be able to use the gamepad out of the box or after some configuration.

- What happened instead?
The gamepad is firstly recognized as `SHANWAN PS3/PC Gamepad` than it is immediately disconnected and reconnected as `SHANWAN Android Gamepad` _(see dmesg below)_.

When the gamepad is plugged in:
- all buttons except one do nothing. _R2_ behaves like _Back_ button
- in binding configuration it seems that the gamepad sends regularly signal -4. When I choose to bind a button, it is immediately bound to -4.
- keyboard starts to behave strangely. Some keys don't work, some keys work badly. E.g. right arrow chooses the top item and left arrow the bottom item, etc.

**dmesg on Lakka after that gamepad is connected:**
```
[  110.793394] usb 7-1: new full-speed USB device number 4 using ohci-pci
[  110.955573] input: SHANWAN PS3/PC Gamepad as /devices/pci0000:00/0000:00:16.0/usb7/7-1/7-1:1.0/0003:2563:0575.000A/input/input15
[  110.955784] hid-generic 0003:2563:0575.000A: input,hidraw5: USB HID v1.10 Gamepad [SHANWAN PS3/PC Gamepad] on usb-0000:00:16.0-1/input0
[  111.116108] usb 7-1: USB disconnect, device number 4
[  111.621625] usb 7-1: new full-speed USB device number 5 using ohci-pci
[  111.781594] input: SHANWAN Android Gamepad as /devices/pci0000:00/0000:00:16.0/usb7/7-1/7-1:1.0/0003:2563:0526.000B/input/input16
[  111.781734] hid-generic 0003:2563:0526.000B: input,hidraw5: USB HID v1.10 Gamepad [SHANWAN Android Gamepad] on usb-0000:00:16.0-1/input0
[  111.789601] input: SHANWAN Android Gamepad as /devices/pci0000:00/0000:00:16.0/usb7/7-1/7-1:1.1/0003:2563:0526.000C/input/input17
[  111.841718] hid-generic 0003:2563:0526.000C: input,hidraw6: USB HID v1.01 Device [SHANWAN Android Gamepad] on usb-0000:00:16.0-1/input1
```

I tested the gamepad on standard linux distributions like Fedora and Ubuntu and it works fine. I even installed Retroach on Ubuntu and the gamepad worked correctly. On the standard linux distributions the gamepad is recognized as `SHANWAN PS3/PC Gamepad` and it is not reconnected as `Android` one. I believe that the root cause is the reconnection.

**dmesg on Fedora after that gamepad is connected:**
```
[ 2102.996662] usb 3-1: new full-speed USB device number 8 using xhci_hcd
[ 2103.124295] usb 3-1: New USB device found, idVendor=2563, idProduct=0575, bcdDevice= 2.00
[ 2103.124297] usb 3-1: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[ 2103.124298] usb 3-1: Product: PS3/PC Gamepad
[ 2103.124299] usb 3-1: Manufacturer: SHANWAN
[ 2103.127652] input: SHANWAN PS3/PC Gamepad as /devices/pci0000:00/0000:00:14.0/usb3/3-1/3-1:1.0/0003:2563:0575.0002/input/input25
[ 2103.127936] hid-generic 0003:2563:0575.0002: input,hidraw1: USB HID v1.10 Gamepad [SHANWAN PS3/PC Gamepad] on usb-0000:00:14.0-1/input0
[ 2103.166180] usb 3-1: USB disconnect, device number 8
[ 2103.771702] usb 3-1: new full-speed USB device number 9 using xhci_hcd
[ 2103.899459] usb 3-1: New USB device found, idVendor=2563, idProduct=0575, bcdDevice= 2.00
[ 2103.899463] usb 3-1: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[ 2103.899465] usb 3-1: Product: PS3/PC Gamepad
[ 2103.899467] usb 3-1: Manufacturer: SHANWAN
[ 2103.903184] input: SHANWAN PS3/PC Gamepad as /devices/pci0000:00/0000:00:14.0/usb3/3-1/3-1:1.0/0003:2563:0575.0003/input/input26
[ 2103.903503] hid-generic 0003:2563:0575.0003: input,hidraw1: USB HID v1.10 Gamepad [SHANWAN PS3/PC Gamepad] on usb-0000:00:14.0-1/input0
```

Ntemis, 2019-06-13:
Try to switch to xinpuut from direct if you have the option

dudaerich, 2019-06-13:
Do you mean `Settings -> Drivers -> Input` set to `x`? I've tried that but it didn't help. It works the same as I described.

dudaerich, 2019-06-13:
It is worth to mention that I tested it also on Fedora with kernel 4.11 and it worked. And I tried `jstest /dev/input/js0` directly on Lakka and it worked as well. So kernel driver looks fine.

gouchi, 2019-06-13:
Did you try with other input driver `linuxraw` or `hid` ?

Ntemis, 2019-06-13:
when a gamepad is ps3 compatible and PC -> Product: PS3/PC Gamepad it uses different protocols to communicate with each platform :)
Windows ->x-input
PS3 -> direct input
You should have a switch or something on the gamepad that selects it

dudaerich, 2019-06-14:
@gouchi yes I tried but it didn't help. `linuxraw` behaves same as I described. `hid` is a bit different. When the gamepad is connected it is still recognized as `Android` but keyboard works OK and I am able to bind gampad buttons `B`, `Y`, `select` and `start`. When I want to bind `arrow up`, it returns `-4` signal and everything freezes. Neither gamepad nor keyboard works. When I disconnect gamepad, keyboard starts working. When I tried to use `Bind All`, it returns `-4` immediately for all buttons.

@Ntemis Ah I see, unfortunately there is no such switch on the gamepad.

Do you know what can cause that the gamepad is recognized as `Android` one? Could it be caused by incorrect `udev` rule? As I said I tested Kernel and Retroach separately and they worked fine so I suspect that it is something with Linux settings.

Thanks guys for your help.

gouchi, 2019-06-15:
Can you try this [testing image](https://drive.google.com/drive/folders/128MaH09bjVcB2pR4Kj2XxdBabR-oDP75) based on LE 9 ?

Thank you.

Ntemis, 2019-06-15:
I did a research on your pad and is indeed a dual mode controller.
It automatically switches internally as d-input and x-input dependable on os.
Somehow it misinterprets Lakka and tries to go direct input mode by it self.
Your issue needs an eeprom firmware update of the controller to be fixed or heavy debugging.
Due to the fact that it seems its a cheaply made frankenbuilded ps3 lookalike cloned controller with low to none interest what so ever(not many users have it afaik)  it might be wiser to report your issue and seek support to esperanza http://esperanza.pl/szczegolyProduktu.php?kat=176&item=1695&lan=4&cur=2

dudaerich, 2019-06-16:
@gouchi it works :) The gamepad is recognized as `SHANWAN PS3/PC Gamepad` and it is not reconnected. I am able to bind all buttons. Thank you!

Do you plan to use LE 9 in next Lakka release?

gouchi, 2019-06-18:
I am glad it is working.

Yes we will work on using LE9/10 but there is no ETA yet.

I will close this issue as there is a workaround.

closed 2019-06-18 by gouchi