libretro/Lakka-LibreELEC #694: Freezes and don't save

niccco, 2019-06-26:
This issue tracker is NOT for help or support. Please use the Lakka forum (https://forums.libretro.com/c/libretro/lakka-tv-general) or IRC @ #lakka on freenode.net for those kinds of posts.

BEFORE you report a bug make sure you have tried the latest nightly version of Lakka at http://le.builds.lakka.tv/. Your bug might be already fixed.

If you are sure that it's a bug in Lakka, please answer the following questions:
- Which version of Lakka are you using?
Lakka 2.2.2 RPi2
- What system hardware are you using?
Raspberry PI 3 B+
- What did you do?
Open the pause menu in Tomb Raider 2, and the system freezes
- What did you expect to happen?
To open the game menu, and save the game, but the game freezes and the system freezes
- What happened instead?
The game freezes, the OS freezes, and all I could do was to unplug the RPi3B+

The OS freezes, and after the freezes I can press R3 + L3 (for the menu), and after I press some movement pad, everything freezes.

I have to unplug the RPi3B+ and plug it again, after I did that, the memory card reset until the last good save (what do I mean with this? Well, this error sometime happens, sometime don't, yesterday I was playing for like 1 hours and nothing happened, I save the game, turn off the RPi and everything was good. Today I loaded the game, and everything was ok, after playing for like 30 minutes, I did saved like 2 o 3 times, I tried to save the game again, but the game freezes, the OS freezes, I had to reset the RPi, and when I did that, the game didn't save anything from those 30 minutes, and bring me back the save from yesterday).

ToKe79, 2019-06-26:
have you been saving by:

    a) in game save function, or
    b) save state?

in case of a) the "memory card" / save data / SRAM content is written when the "close content" is used, unless you change `autosave_interval` to a value > 0 (in seconds, e.g. 60 = every minute). this is how RetroArch works, so it is not a bug of Lakka.
In case of b) it is really strange, the savestates are written immediately and in case you did not change the slot, it would overwrite the same file.

niccco, 2019-06-26:
I have been saving with a) in game save function.

Well, that might be it, sometimes I just press esc from keyboard, or turn off the raspberry pi.

But, how can this work like this? If I save the game in the memory card (in game function I mean), it should be instant saved, not waiting for me to close de application.

Give me a sec, I will try.

And thanks for the help!!

niccco, 2019-06-27:
Well, look's like it was it, when the game freezes, the save freezes and there isn't a "close content" and the memory is never saved, but, if the game doesn't freezes and I save the game in the memory card and I close the content, everything works fine.

So, my error now, is that Tomb Raider 2 is freezing when I press Start or Select button.

I know that it isn't the game, because I had it in my desktop computer, and I could play it just fine.

I will have to try with other games, to test if there is other game with same issue or particular game issue.

Any idea how to debug or get logs to see why the game freezes?

ToKe79, 2019-06-27:
Which core / emulator have you been playing Tomb Raider with? My first guess is pcsx_rearmed (Sony Playstation). [How to get logs.](https://github.com/libretro/Lakka-LibreELEC/wiki/Troubleshooting-Lakka)

niccco, 2019-06-27:
I'm using Lakka default PSX emulator, if i'm not wrong is pcsx_rearmed. I will check the how to get logs doc, and I will try to post here the results.

niccco, 2019-06-28:
[log.txt](https://github.com/libretro/Lakka-LibreELEC/files/3337428/log.txt)

Update, this is the log file, when the Tomb Raider 2 freezes, it was when I tried to enter to the pause menu.

ToKe79, 2019-07-01:
With my humble knowledge I do not see any errors - the core is correctly unloaded. The freezing might be also caused by wrong dump / incompatibility - see https://pyra-handheld.com/boards/threads/pcsx-rearmed-tomb-raider-2-freezes.70671/.

closed 2019-07-01 by ToKe79