libretro/Lakka-LibreELEC #705: [Requests] Add/Remove somes cores

wwark, 2019-07-15:
Hello,

Do you think it is possible to add/remove following cores in the master branch?

- Add Play- (PS2 Emulator) - https://github.com/libretro/Play-
- Add Mesen-s (Snes Emulator) - https://github.com/libretro/libretro-super/commit/077c5e0981067ad9e11ad24eb6fd897fc4f27dff
- Remove Mupen64plus (archived in libretro side : https://github.com/libretro/mupen64plus-libretro)

Thank you,
Wwark

natinusala, 2019-07-15:
We have a package.mk for Play it will be added eventually

Mupen can't be removed as some platforms can't use mupen-next

pcs3rd, 2019-07-18:
Also https://github.com/libretro/minibrowser. 
Would make a good addition to lakka

RobLoach, 2019-07-18:
> Also https://github.com/libretro/minibrowser.

Last time I tried that, it didn't work.

pcs3rd, 2019-07-18:
Oh :(. So no way to authorize captive portals?

natinusala, 2019-07-19:
I understand your request, but I really don't want people to exit RetroArch when running Lakka, so using a standalone browser is a no-go.

However, if you can find / create a lightweight browser libretro core that works with controllers (that's very important), feel free to notice us.

wwark, 2019-07-27:
Thank you for adding play- core !

gouchi, 2019-07-27:
- Play!: new libretro core #707

wwark, 2019-08-02:
Thanks again for adding mesen-s.

I closed this ticket !

closed 2019-08-02 by wwark

gouchi, 2019-08-02:
- mesen-s : add mesen-s libretro core  #740