libretro/Lakka-LibreELEC #710: Feature Request: local coop without internet

pcs3rd, 2019-07-17:
For those places that don't have convenient for compatible networks, have one capable device host wifi (and the net play session) and set the other to just connect to the wifi then the hosted net play session

RobLoach, 2019-07-17:
This should be possible right now?

- Disable Publically Anounce Netplay
- Disable Relay Server
- Start the server
- Input the IP of the host in your client
- Connect

Haven't tried this before, but should work just fine.

If not, feelf ree to create an issue on the [RetroArch Issue Queue](https://github.com/libretro/retroarch/issues).

gouchi, 2019-07-17:
Moreover, you can use [Lakka as access point](https://www.lakka.tv/doc/lakka-as-accesspoint/) if needed.

pcs3rd, 2019-07-18:
@gouchi don't realize that... Thanks for bringing it up. Are there any plans to implement settings for that into the gui

closed 2019-07-18 by pcs3rd