libretro/Lakka-LibreELEC #715: Lakka 3.0 checklist

natinusala, 2019-07-19:
- [ ] **Platforms - either ported or removed**
  - [ ] Raspberry Pi
    - [ ] RPi0
    - [ ] RPi1
    - [ ] RPi2/3
    - [ ] RPi3
  - [ ] Generic
    - [ ] Standard
    - [ ] VK only nvidia
  - [ ] IMX
    - [ ] Cuboxi
    - [ ] UDOO
  - [ ] HummingBoard
  - [ ] Odroid
    - [ ] Odroid C1
    - [ ] Odroid C2
    - [ ] Odroid Xu3/4
  - [ ] WeTek
    - [ ] Play 1/2
    - [ ] Hub
  - [ ] Allwinner
    - [ ] Banana Pi
    - [ ] Cubieboard
    - [ ] Cubiebord 2
    - [ ] Cubietruck
    - [ ] Orange Pi
  - [ ] Amlogic
    - [ ] S805
    - [ ] S802
    - [ ] S812
    - [ ] S905
    - [ ] S912
  - [ ] Asus Tinkerboard
  - [ ] Rockchip
    - [ ] RK3328 (ROCK64)
    - [ ] RK3399 (ROCKPro64)
  - [ ] Other platforms unlisted on the Lakka.tv website
    - [ ] MiQi
    - [ ] Gamegirl
    - [ ] Slice
    - [ ] Slice3
- [ ] **Features**
  - [ ] SSH/SCP
  - [ ] Bluetooth
  - [ ] Wi-Fi
  - [ ] Samba Share
  - [ ] USB Automount
  - [ ] Timezone
  - [ ] Lakka updater
  - [ ] Overlays
    - [ ] Assets
    - [ ] Cores
    - [ ] Database
    - [ ] Joypad autoconfig
    - [ ] Shaders
    - [ ] Overlays
    - [ ] System (for cores needing assets such as PPSSPP)
- [ ] working aarch64 builds on relevant platforms (with properly configured cores)
- [ ] don't rip xorg out of LibreELEC (for the L4T port (Switch and Jetson Nano) that I hope to upstream one day)

SwedishGojira, 2019-07-20:
RPi3 is listed twice...

ToKe79, 2019-07-20:
fixed, the other RPi3 should have been RPi4 ;-)

ToKe79, 2019-07-20:
I merged (in OP) RPi0 and RPi1 in one - they share the image. Same for IMX cuboxi and HummingBoard (download links to the same cuboxi image on lakka.tv).

What is already in LE9.2:
- project Allwinner
   - device A64 (new)
   - device H3 (covers all above + some more)
   - device H6 (new)
- project Amlogic
   - device AMLGX (covers S905, S912, Odroid C2 and others)
   - device AMLG12 (did not try to build yet)
- project Generic
   - both 32/64-bit architectures, no VULKAN yet
- project Rockchip
   - device MiQi
   - device RK3328 (covers Rock64, ROC-CC and others)
   - device RK3399 (covers RockPro64, Rock960 and others)
   - device TinkerBoard (Asus TinkerBoard)
- project RPi
   - device Gamegirl (does not build yet)
   - device RPi (covers RPi0/1)
   - device RPi2 (covers RPi2/3)
   - device RPi4
- project Qualcomm (new)
   - device Dragonboard (new)

Images: http://nightly.builds.lakka.tv/Lakka-LE9.2/latest/

SethBarberee, 2019-07-29:
I can help test the A64 port on my Pine A64+ in the coming month. Also, have a Pi3 that I can volunteer to test too

rome1983, 2019-10-09:
Will the odroid n2 be a supported board?

yyyreal, 2019-10-21:
Does it mean that Jetson nano will receive Lakka in the future?

natinusala, 2019-10-21:
I hope eventually that both will, I already have the code for Jetson Nano so it will be easier

ezar, 2019-12-06:
> Does it mean that Jetson nano will receive Lakka in the future?

I will see it too. @natinusala :)

SethBarberee, 2019-12-06:
Just got my Pine64 and was able to test the image from `Lakka-9.2` branch. Booted perfectly fine and recognized my steam controller on boot. Gonna put my GBA roms to test performance but test setup is ready! :)

ezar, 2019-12-07:
> Just got my Pine64 and was able to test the image from `Lakka-9.2` branch. Booted perfectly fine and recognized my steam controller on boot. Gonna put my GBA roms to test performance but test setup is ready! :)

Where are this test builds?

natinusala, 2019-12-07:
Right here : http://nightly.builds.lakka.tv/Lakka-LE9.2/latest/

closed 2020-08-18 by Ntemis