libretro/Lakka-LibreELEC #723: ethernet not working on s912, only Wi-Fi

farauk, 2019-07-24:
Good day/night. This is my very first ever submitting a bug. 

I just tried the last development version Lakka-S912.arm-2.2-devel-20190702092302-r28309-ga3bbc74f52.img.gz. I had previously tested this on the stable versions and some other development ones. I just found issue #450, but it was closed out back on June 2018, and still issue persists. When using the dtb that is within the lakka image (MD5 F38FE0F9298392783D2503C35C970BD2), lakka only shows Wi-Fi connections. No eth0 information is shown when disconnected from Wi-Fi.

I have a Tanix Tx92, based on s912, the 16GB/2GB version. I have tried the following dtbs with no avail: gxm_q200_2g, gxm_q200_3g, gxm_q200_k3_pro, gxm_q201_2g_1gbit, gxm_q201_2g_mb2, gxm_q201_3g_1gbit. I know the one I should be using is the gxm_q201_2g_1gbit one. 

The lakka documentation says we should try first the dtbs on the libreelec.tv site. I have tried gxm_q200_2g (MD5 FC22DF96E2F6029319701F64F3FA9F68), but it doesn´t work. Lakka doesn't load retroarch. It shows the booting screen and then it goes all black. 

I do find that both the "stock" dtb and the one in the libreelec.tv have different size. The one included in the lakka image has 134KB, while the dtbs downloaded from the trees are around 41-44KB. 

I don't think I'm using the wrong dtb - [according to this](https://coreelec.org/dtb/), the one for Tanix Tx92 is gxm_q201_2g_1gbit. I have tried [the ones from here](https://github.com/LibreELEC/device-trees-amlogic) as well. 

To sum up, the dtbs on the device trees needed to have eth0 working do not seem to boot properl. System only boots properly with the dtb that is included in the lakka image (which is almost three times the size of the ones included in the device trees), but that one doesn't have a working ethernet.

ToKe79, 2019-07-24:
The DTB included in the image tries to cover most possible devices, but in some cases a specific DTB for the device is needed. Please test with the dtbs [here](http://kszaq.libreelec.tv/s912/8.2/device_trees/1Gbit_Ethernet/).

farauk, 2019-07-25:
Hello. Yes, I have tried gxm_q200_2g from the link in the documentation before as well. In fact, I have tried all of those just in case.  None of them seem to be working.

Ntemis, 2019-07-25:
We only support gbit lans if yours is 10/100 then simply isn't supported whatever device tree you use :)

ToKe79, 2019-07-25:
But according to http://www.tanix-box.com/project-view/tanix-tx92-android-tv-box-powered-amlogic-s912/ it should have Gbit LAN. When I used Lakka on my S912 box (Beelink GT1) the ethernet worked.

Ntemis, 2019-07-25:
Hardware failure then?

ToKe79, 2019-07-25:
@farauk to rule out any issue on Lakka side, can you try [LibreELEC 8.2 for S912](http://kszaq.libreelec.tv/s912/8.2/)?

ToKe79, 2019-07-25:
you can also try [LE9.2 based image](http://nightly.builds.lakka.tv/Lakka-LE9.2/latest/AMLGX.arm/) - download the file with suffix `-box.img.gz` - this is generic build for S905/S912 devices.

Ntemis, 2019-07-25:
Also tell us your lan chipset/maker. Could be a simple kernel fix if we dont have your lan driver enabled.

farauk, 2019-07-25:
> But according to http://www.tanix-box.com/project-view/tanix-tx92-android-tv-box-powered-amlogic-s912/ it should have Gbit LAN. When I used Lakka on my S912 box (Beelink GT1) the ethernet worked.

I don't recall if previous versions of lakka worked with it. I think the first version I tried on the box was 2.1, and I am not sure if it ever worked. And yes, according to the specs, this has GBit lan. I downloaded some apps to see what chipset it might use, but none of them show the lan information, only Wi.Fi. According to CPU-Z, the board is q201 (that is why I tried gxm_q201_2g_1gbit dtb as well).

> Also tell us your lan chipset/maker. Could be a simple kernel fix if we dont have your lan driver enabled.

I tried to get that information from multiple apps, None of them show anything. I tried `dmesg | grep -i lan` using Android Shell and got a lot of lines, but didn't show any information I think. Also tried `dmesg | grep -i eth0` and it showed nothing. With `dmesg` only, I got a bunch of lines and could only find wlan info. I do not of any other command that could show the chipset.

I have been doing some research on the net, but could not find anything. However, [I did find pics of the board here](https://forum.freaktab.com/forum/tv-player-support/amlogic-based-tv-players/s912/664645-tanix-tx92-amlogic-s912-cpu-android-7-1-os-2-4g-5g-wifi-1000m-lan-bt4-1). There is a picture of the board that shows the chip, I think it is SG24002 G. A quick Google search says it is a "10/100/1000Base Transformer", so probably not a lan card?

Will try those LE images and post again with the results. Thank you, guys!

natinusala, 2019-07-26:
FWIW, all q201 devices that I've tried have a working Ethernet port

farauk, 2019-07-26:
> FWIW, all q201 devices that I've tried have a working Ethernet port on Android

Yes, when I boot up Android, ethernet is working fine. 

The main reason I wanted to use ethernet on lakka is speed. I stream everything from an RPi3-NAS in Kodi using Android, throug ethernet. I tried to do the same using lakka, loading from the NAS, as the PCE chds I tried take way too long to load - I don't know if it is because somehow lakka "copies" the file to a temp folder and then loads it in a core, probably a separate issue. In any case, ethernet is working using Android.

> @farauk to rule out any issue on Lakka side, can you try [LibreELEC 8.2 for S912](http://kszaq.libreelec.tv/s912/8.2/)?

Image booted fine, but there is no ethernet either. It directly prompted me to select the Wireless connection I wanted to use on the initial setup. I assume that if would directly be connected to the internet had it detected the ethernet connection? I unplugged the cable and plugged it in again, but doesn't recognize it.

> you can also try [LE9.2 based image](http://nightly.builds.lakka.tv/Lakka-LE9.2/latest/AMLGX.arm/) - download the file with suffix `-box.img.gz` - this is generic build for S905/S912 devices.

This one didn't boot at all. Couldn't even expand the contents. I could trying using the same dtbs I already used, but I'm afraid the result would be the same.

Ntemis, 2019-07-26:
You possibly have an unsupported lan chipset

ToKe79, 2019-07-27:
the other DTBs are compiled for different kernel (3.14-amlogic I guess), the LE9.2 images use 5.1 for Amlogic. I have found that in u-boot is config for khadas vim2, which is compatible with s912, so you can try this image http://nightly.builds.lakka.tv/special_builds/vudiq/le92_s912/

farauk, 2019-07-28:
> the other DTBs are compiled for different kernel (3.14-amlogic I guess), the LE9.2 images use 5.1 for Amlogic. I have found that in u-boot is config for khadas vim2, which is compatible with s912, so you can try this image http://nightly.builds.lakka.tv/special_builds/vudiq/le92_s912/

This one didn´t boot either. The TV box booted into Android, just like the previous LE9.2 image I tried. 

Interestingly enough, though, I just tried the latest dev build Lakka-S912.arm-2.2-devel-20190727210657-r28318-g1433b76af3 and it SEEMS to be working. I will try now to see if everything loads fine. So far I am connected through wired and loading roms just fine from the NAS.

Thank you, guys!

farauk, 2019-07-28:
For the record, the new DTB that comes on the new build has this MD5 checksum: FC22DF96E2F6029319701F64F3FA9F68. It doesn't match with neither of the DTBs that are found on the trees. I was about to get the DTB from my box to see if it worked with that one, but the new build works just fine.

ToKe79, 2019-08-01:
good to know that the 2.2-devel build works.

I just found out that for the LE9.2 builds for AMLGX devices (image with `-box` suffix) you have to edit first line of `uEnv.ini` file in the root folder of the LAKKA partition:
```
dtb_name=/dtb/@@DTB_NAME@@
```
just replace the `@@DTB_NAME@@` with the desired file name in the folder `/dtb/`, i.e. `meson-gxm-q200.dtb`.
I did not try myself yet.

closed 2020-06-14 by farauk