libretro/Lakka-LibreELEC #734: Lakka comes with a faulty lr-fuse core

brightentayle, 2019-07-30:
The lr-fuse that has been bundled with Lakka for a while (at least with both Raspberry Pi 0/1 and Raspberry Pi 2/3 builds) crashes every time it attempts to make a savestate. To my knowledge, this has been the case for every version of Lakka from 2.0 to the more recent 2.2.2, but _never_ with RetroPie, which comes with a properly working lr-fuse out of the box.

For the record, I did test RetroPie's lr-fuse on Lakka, and it still handles savestates pretty well.

gouchi, 2019-09-10:
Hi,

Do you reproduce the issue with [nightly builds](http://nightly.builds.lakka.tv/Lakka-LE9.2/latest/) ?

Thank you.

brightentayle, 2019-09-13:
I've tested it on Lakka-RPi.GPICase.arm-2.3-devel-20190912092804-r28338-g3c10941986.img.gz, and the issue persists there, too.

Lucius1202, 2019-12-30:
Hi, same problem here, raspberry pi 4 with lakka 2.3.1, every fuse savestate crash and restart retroarch, in the .cache directory pop a big file every crash 90 - 100 mb.

closed 2020-08-18 by Ntemis