libretro/Lakka-LibreELEC #743: Lakka on RPi4 composite output bootloo

pjft, 2019-08-07:
Apologies for reporting this here, but I was unable to find a solution elsewhere.

- Which version of Lakka are you using?
(Lakka-RPi2.RPi4.arm-2.2-devel-20190803002701-r28321-gc3108f0a77.img) (03/08) and the one from 03/07 as well (Lakka-RPi2.RPi4.arm-2.2-devel-20190731001831-r28320-g30c72500fa.img).

- What system hardware are you using?
RPi 4 with via composite output. PS3 controller connected, but it's the same if not connected.

- What did you do?
Flash the image via Etcher. Edit config.txt to add enable_tvout=1 . Turn on Pi.

- What did you expect to happen?
For Lakka to eventually boot and run.

- What happened instead?
Black screen for a non-negligible amount of time (1 or more minutes). Then after a bit the Lakka flower shows up, and a few messages show up asking not to reboot, a few paths (/storage?), and finally something suggesting it will reboot after 15 seconds or so. 
Screen loses signal on reboot, and after that it reboots, same wait for 1 or more minutes, then Lakka flower - this time without any messages - for a few minutes, and then just repeats the black screen for a few minutes, Lakka flower pattern indefinitely. I left it like this for about 1 hour or so and didn't look as if anything was changing so then I gave up.

I believe that the long "1 or more minutes" before the Lakka flower shows up is a firmware issue, https://www.raspberrypi.org/forums/viewtopic.php?t=245655 but I would expect that the repeated reboot pattern is not?

If I remove the "enable_tvout=1" flag, the display doesn't work but on the flipside I can see the yellow light from SD card access to start flashing immediately, suggesting that it is not waiting for that "1 or more minute" before progressing. That being said I don't know if it stays in the reboot cycle or not as I have no visual feedback :)

I cannot generate logs as I haven't yet managed to set up the wifi connection to the Pi (I don't have ethernet here, apologies).

Let me know if there's anything else I can help with here.

natinusala, 2019-08-07:
I'm afraid that without logs we won't be able to diagnose the issue.

Can you look on the Lakka website, in the Documentation section there is a guide to generate logs and I think that there's a way to grab them without using WiFi or Ethernet.

pjft, 2019-08-07:
Thanks. I have mounted the SD card on my current pi, but can't seem to find any folders where any system logs would be stored.

I read https://www.lakka.tv/doc/troubleshooting-lakka/ but unless I'm missing something clear I'm not sure that's feasible. Also, I do have a keyboard, but direct access doesn't seem to be possible in a Raspberry Pi? 

https://www.lakka.tv/doc/accessing-lakka-command-line-interface/

So, maybe a different workaround: any way I can set up ssh and wifi in headless mode (in raspbian I recall being able to add a wpa_suplicant.conf file, but I'm not sure this is working here).

Or could anyone point me to system logs somewhere?

Thanks.

pjft, 2019-08-10:
Just a comment, then, so that this can at least help someone else in digging into this: if someone can install Lakka from a fresh install from one of these builds and then just add the  "enable_tvout=1" entry to config.txt, I imagine that it'd be as replicable as here, though someone might be better able to debug or trace it, if it's useful.

Will keep installing future nightlies in case something changes. By the way, is there an expectation for Lakka builds to include specific RPi firmware updates? As there have been some firmware updates to address the enable_tvout performance at some point, it seems, hence the question.

https://github.com/raspberrypi/firmware/issues/1223

Thanks, and have a great weekend.

pjft, 2019-08-16:
Ok, managed to get an ethernet cable, then plugging it in, enabling ssh in cmdline.txt and logging in, this is what I get in terms of logs:

```
[INFO] RetroArch 1.7.7 (Git ef5f448f41)
[INFO] === Build =======================================
[INFO] Capabilities: NEON VFPv3 VFPv4 
[INFO] Built: Jul 29 2019
[INFO] Version: 1.7.7
[INFO] Git: ef5f448f41
[INFO] =================================================
[INFO] Environ SET_PIXEL_FORMAT: RGB565.
[INFO] Redirecting save file to "/storage/savefiles/.srm".
[INFO] Redirecting savestate to "/storage/savestates/.state".
[INFO] Version of libretro API: 1
[INFO] Compiled against API: 1
[INFO] [Audio]: Set audio input rate to: 30000.00 Hz.
[INFO] [Video]: Video @ fullscreen
[INFO] [Video]: Starting threaded video driver ...
[WARN] [DRM]: Couldn't get device resources.
[WARN] [DRM]: Couldn't get device resources.
[INFO] [DRM]: Found 1 connectors.
[INFO] [DRM]: Connector 0 connected: yes
[INFO] [DRM]: Connector 0 has 1 modes.
[INFO] [DRM]: Connector 0 assigned to monitor index: #1.
[INFO] [DRM]: Mode 0: (LCD_MODE) 720 x 480, 60 Hz
[INFO] [GL]: Found GL context: kms
[INFO] [GL]: Detecting screen resolution 720x480.
[INFO] [EGL] Found EGL_EXT_platform_base, trying eglGetPlatformDisplayEXT
[INFO] [EGL]: EGL version: 1.4
[INFO] [EGL]: Current context: 0xb3f087a8.
[INFO] [KMS]: New FB: 720x480 (stride: 2880).
[INFO] [GL]: Vendor: Broadcom, Renderer: V3D 4.2.
[INFO] [GL]: Version: OpenGL ES 3.0 Mesa 19.1.1.
[INFO] [GL]: Using resolution 720x480
[INFO] [GL]: Default shader backend found: glsl.
[INFO] [Shader driver]: Using GLSL shader backend.
[WARN] [GL]: Stock GLSL shaders will be used.
[INFO] [GLSL]: Found GLSL vertex shader.
[INFO] [GLSL]: Found GLSL fragment shader.
[INFO] [GLSL]: Linking GLSL program.
[INFO] [GLSL]: Found GLSL vertex shader.
[INFO] [GLSL]: Found GLSL fragment shader.
[INFO] [GLSL]: Linking GLSL program.
[INFO] [GLSL]: Found GLSL vertex shader.
[INFO] [GLSL]: Found GLSL fragment shader.
[INFO] [GLSL]: Linking GLSL program.
[INFO] Setting up menu pipeline shaders for XMB ... 
[INFO] [GLSL]: Compiling ribbon shader..
[INFO] [GLSL]: Found GLSL vertex shader.
[INFO] [GLSL]: Found GLSL fragment shader.
[INFO] [GLSL]: Linking GLSL program.
[INFO] [GLSL]: Compiling simple ribbon shader..
[INFO] [GLSL]: Found GLSL vertex shader.
[INFO] [GLSL]: Found GLSL fragment shader.
[INFO] [GLSL]: Linking GLSL program.
[INFO] [GLSL]: Compiling snow shader..
[INFO] [GLSL]: Found GLSL vertex shader.
[INFO] [GLSL]: Found GLSL fragment shader.
[INFO] [GLSL]: Linking GLSL program.
[INFO] [GLSL]: Compiling modern snow shader..
[INFO] [GLSL]: Found GLSL vertex shader.
[INFO] [GLSL]: Found GLSL fragment shader.
[INFO] [GLSL]: Linking GLSL program.
[INFO] [GLSL]: Compiling bokeh shader..
[INFO] [GLSL]: Found GLSL vertex shader.
[INFO] [GLSL]: Found GLSL fragment shader.
[INFO] [GLSL]: Linking GLSL program.
[INFO] [GLSL]: Compiling snowflake shader..
[INFO] [GLSL]: Found GLSL vertex shader.
[INFO] [GLSL]: Found GLSL fragment shader.
[INFO] [GLSL]: Linking GLSL program.
[INFO] [GL]: Using 4 textures.
[INFO] [GL]: Loaded 1 program(s).
[INFO] [Font]: Using font rendering backend: freetype.
[INFO] [Video]: Graphics driver did not initialize an input driver. Attempting to pick a suitable driver.
[WARN] [udev]: Couldn't open any keyboard, mouse or touchpad. Are permissions set correctly for /dev/input/event*?
[INFO] [Joypad]: Found joypad driver: "udev".
[WARN] [udev]: Full-screen pointer won't be available.
[INFO] [Video]: Found display server: null
[INFO] Found shader "/tmp/shaders/bilinear.glslp"
[INFO] Found shader "/tmp/shaders/nearest.glslp"
[INFO] Found shader "/tmp/shaders/slang-conversion-template.glsl"
[INFO] Found shader "/tmp/shaders/stock.glsl"
[INFO] ALSA: Using signed 16-bit format.
[INFO] ALSA: Period size: 444 frames
[INFO] ALSA: Buffer size: 1411 frames
[INFO] [Menu]: Found menu display driver: "gl".
[INFO] [Font]: Using font rendering backend: freetype.
[INFO] [Font]: Using font rendering backend: freetype.
[INFO] [Menu]: Found menu display driver: "gl".
[INFO] [Font]: Using font rendering backend: freetype.
[INFO] [Font]: Using font rendering backend: freetype.
[INFO] [LED]: LED driver = 'null' 0x34659c
[INFO] [MIDI]: Initializing ...
ALSA lib /mnt/data/PROJECTS/Lakka-LibreELEC/build.Lakka-RPi4.arm/alsa-lib-1.1.9/src/seq/seq_hw.c:466:(snd_seq_hw_open) open /dev/snd/seq failed: No such file or directory
[ERROR] [MIDI]: snd_seq_open failed with error -2.
[ERROR] [MIDI]: Initialization failed (list of input devices unavailable).
[INFO] SRAM will not be saved.
[INFO] Loading history file: [/storage/.config/retroarch/content_history.lpl].
[INFO] Loading history file: [/storage/.config/retroarch/content_favorites.lpl].
[INFO] Loading history file: [/storage/.config/retroarch/content_music_history.lpl].
[INFO] Loading history file: [/storage/.config/retroarch/content_video_history.lpl].
[INFO] Loading history file: [/storage/.config/retroarch/content_image_history.lpl].
[INFO] [Font]: Using font rendering backend: freetype.
[INFO] [Font]: Using font rendering backend: freetype.
[INFO] [KMS]: New FB: 720x480 (stride: 2880).
[INFO] [KMS]: New FB: 720x480 (stride: 2880).
[INFO] [KMS]: New FB: 720x480 (stride: 2880).
Segmentation fault (core dumped)
```

54dfg254, 2019-09-14:
This issue is seemingly fixed as of yesterday in Raspbian. 

https://github.com/raspberrypi/firmware/issues/1223

Commenting here to ensure this firmware change makes it into the next Lakka release, as the nightly still has this bug.

natinusala, 2019-09-14:
I have trouble understanding what exactly changed in theGitHub issue you linked. "Firmware" can refer to many things, and I don't know what change is necessary to fix this issue. Did they update the dt overlays? Or did they update the actual bootloader firmware?

54dfg254, 2019-09-15:
I'm not sure, however, they left that issue open for a few days, so I just asked them and referenced this issue. Thanks for responding.

54dfg254, 2019-09-18:
This was the initial kernel fix
https://github.com/Hexxeh/rpi-firmware/commit/2096a8d567eed47cb742ddb1ce9bf9ac25726980

And there was also a problem with the fkms driver
https://github.com/raspberrypi/linux/pull/3223

Sorry for any imprecise descriptions. I'm a guy who plays a lot of old games on a CRT with Lakka, not a kernel developer.

natinusala, 2019-09-18:
No worries, I know it's a kernel update that's required now. Thanks

natinusala, 2019-09-25:
There is a temporary fix available here: 

https://github.com/libretro/Lakka-LibreELEC/issues/802#issuecomment-534960672

closed 2020-08-18 by Ntemis