libretro/Lakka-LibreELEC #744: Lakka starts but doesn't reach menu on Samsung 4k TV.

pedrev, 2019-08-09:
- Which version of Lakka are you using?
Lakka-Allwinner.orangepi_lite.arm-2.2.2.img

- What system hardware are you using?
Orange Pi Lite (Allwinner)

- What did you do?
Flashed the image of Lakka on the SD card, connected it to the board and a desktop monitor (HDMI). After the initial process of "expanding the file-systems", I was able to see Lakka menu.

- What did you expect to happen?
To work, and it worked fine on my desktop LG monitor.

- What happened instead?
When I connect it to my Samsung 4k TV, I never get past the splash screen (Lakka flower). Also, in that splash screen I have access to the command line, but that's it. Nothing else happens. Seems to me that there is a problem with the Lakka image and this TV ? 

What might be the problem?

Thank you

ToKe79, 2019-08-10:
```
systemctl stop retroarch
retroarch -v
```
gives what?

pedrev, 2019-08-10:
# Lakka: systemctl stop retroarch
Lakka: #retroarch -u
CINFO] RetroArch 1.7.7 (Git ef5f448f41 )
INFO] Build ==== ===- =
CINFO] Capabilities: NEON UFPU3 UFPU4
[INFO] Built: Jul 29 2019
INFO] Version: 1.7.7
CINFO] Git: ef5f448f41
INFO) Enuiron SET PIXEL FORMAT: RGB565.
INFO] Redirecting saue file to "/storage/sauefiles/.srm".
CINFO] Redirecting sauestate to "/storage/sauestates/.state".
INFO] Version of libretro API: 1
[INFO) Compi led against API: 1
[INFO] [Audio]: Set audio input rate to: 30000.00 Hz.
CINFO) [Video]: Uideo fullscreen
DINFO) [Video]: Starting threaded uideo driver ..
INFO] [DRM]: Found 1 connectors.
CINFO] EDRM]: Connector 0 comnected: yes
INFO] IDRM]: Connector 0 has 48 modes.
INFO [DRM]: Connector 0 assigned to monitor index: #1
INFO] DRM]: Mode 0: (3840x2160) 3840 x 2160, 30 Hz
INFO] DRM]: Mode 1: (4096x2160) 4096 x 2160, 30 Hz
CINFO] EDRM]: Mode 2: (4096x2160 ) 4096 x 2160, 30 Hz
INFO] [DRM]: Mode 3: (4096x2160) 4096 x 2160, 25 Hz
CINFO] CDRM]: Mode 4: (4096x2160) 4096 x 2160, 24 Hz
[INFO] CDRM]: Mode 5: (4096x2160) 4096 x 2160, 24 Hz
[INFOJ IDRM]: Mode 6: (3840x2160) 3840 x 2160, 30 Hz
[INFO] CDRM]: Mode 7: (3840x2160) 3840 x 2160, 25 Hz
[INFO] CDRM]: Mode 8: (3840x2160) 3840 x 2160, 24 Hz
DINFO)
[INFO] CDRM]: Mode 10: (1920x1080) 1920 x 1080, 60 Hz
LINFO) CDRM]: Mode 11: 1920x1080) 1920 x 1080, 60 Hz
[INFO] CDRM1: Mode 12: (1920x1080 i) 1920 x 1080, 60 Hz
[INFO] DRM1: Mode 13: (1920x1080i) 1920 x 1080, 60 Hz
[INFO] IDRM]: Mode 14: (1920x1080) 1920 x 1080, 50 Hz
[INFO] CDRM]: Mode 15: (1920x1080 i) 1920 x 1080, 50 Hz
LINFO) IDRM]: Mode 16: (1920x1080) 1920 x 1080, 30 Hz
[INFOJ DRM]: Mode 17: (1920x1080) 1920 x 1080, 30 Hz
[INFOJ IDRM]: Mode 18: (1920x1080) 1920 x 1080, 25 Hz
INFOJ CDRM]: Mode 19: (1920x1080) 1920 x 1080, 24 Hz
CINFO) IDRM]: Mode 20: (1920x1080) 1920 x 1080, 24 Hz
INFO] DRM1: Mode 21: (1680x1050) 1680 x 1050. 60 Hz
CINFO] IDRM]: Mode 22: (1600x900) 1600 x 900, 60 Hz
[INFO) [DRM]: Mode 23: (1280x1024) 1280 x 1024, 75 Hz
CINFO] [DRM1: Mode 24: (1280x1024) 1280 x 1024, 60 Hz
[INFO) [DRM]: Mode 25: (1440x900) 1440 x 900, 60 Hz
CINFO] IDRM]: Mode 26: (1366x768) 1366 x 768, 60 Hz
CINFOJ [DRM]: Mode 27: (1280x800) 1280 x 800, 60 Hz
CINFO] [DRM]: Mode 28: (1152x864) 1152 x 864, 75 Hz
CINFO) [DRM]: Mode 29: (1280x720) 1280 x 720, 60 Hz
[INFOJ CDRM]: Mode 30: (1280x720) 1280 x 720, 60 Hz
[INFOJ IDRM]: Mode 31: (1280x720) 1280 x 720, 50 Hz
CINFO] [DRM]: Mode 32: (1024x768) 1024
INFO] CDRM]: Mode 33: (1024x768) 1024
INFO] CDRM]: Mode 34: (1024x768) 1024 x 768, 60 Hz
[INFO] CDRM]: Mode 35: (832x624) 832 x 624, 75 Hz
INFO] EDRM]: Mode 36: (800x600) 800 x 600, 75 Hz
CINFO] CDRM]: Mode 37: (800x600 ) 800 x 600, 72 Hz
[INFO] [DRM]: Mode 38: (800x680) 800 x 600, 60 Hz
[INFO] CDRM]: Mode 39: (720x576) 720 x 576, 50 Hz
CINFO] CDRM]: Mode 40: (720x480) 720 x 480, 60 Hz
DINFO] CDRMJ: Mode 41: (72z0x480) 720 x 480, 60 Hz
INFO] [DRM]: Mode 42 (640x480) 640 x 480, 75 Hz
[INFO] CDRM: Mode 43: (640x480) 640 x 480, 73 Hz
[INFO] [DRM1: Mode 44: (640x480) 640 x 480, 67 Hz
CINFO] CDRM]: Mode 45: (640x480) 640 x 480, 60 Hz
CINFO] CDRM]: Mode 46: (640x480) 640 x 480, 60 Hz
[INFO] CDRM]: Mode 47: (720x400) 720 x 400, 70 Hz
CINFO) [GL]: Found GL context: kms
9: (3840x2160) 384
x 2160,
768, 75 Hz
768, 70 Hz
INFO] [GLI: Detecting screen resolution 3840x2160
EINFO] CEGL] Falling back to eglGetDisplay
INFO] CEGL1: EGL version: 1.4
Segmentation fault (core dumped)

So basically: _**Segmentation fault (core dumped)** _ 
 
Anything I can do?

natinusala, 2019-08-10:
You need a debug build (only debug retroarch will suffice) and GDB.

I would gladly provide you such a build but I cannot as I'm away from home

pedrev, 2019-08-16:
> You need a debug build (only debug retroarch will suffice) and GDB.
> 
> I would gladly provide you such a build but I cannot as I'm away from home

Can someone else provide me more info on how to proceed with debugging? 
I would really love to see Lakka working on my TV :)

Thank you for everything

closed 2020-08-18 by Ntemis