libretro/Lakka-LibreELEC #745: Lakka Generic PC issues

Gitlab1111, 2019-08-10:
CPU: Intel Haswell Intel© Core™ i5-4670 CPU @ 3.40GHz × 4
GPU: N/A
RAM: 4gb 
SSD 60gb

1.One of the usual annoyances I find with Lakka OS is right off the bat when it is successfully installed is that you have to fiddle with the audio setting because there is no sound and it was not letting me cycle through the list untill I rebooted it.

2. PPSSPP has issues with audio and game saving. It's as if the function has been removed and so far I had to resort to using save states instead but this does not work right with certain titles that need multiple playthroughs for new game+. When attempting to save the screen goes black and you can hear the emulation dropping in quality such as audio and FPS.
note: i have the audio on low latency setting. 

3. Lakka OS lacks documentation for bios placement. For example PSx do I make a folder under system/ or do I just dump the rest of the bios there.  For example would it work like this /system/psx  

4. No mention of bios placement for Atari 5200 as well.

5. Lakka site does not list what emulator cores are available, where is PS2 emulation core at? The latest Retroarch has it included.

6.I wanted to delete a entry from a game rom that was on a different drive. But now I'm stuck with a double of the listed game rom.

7. Screenshots show 0% when taking one through the menu but they actually are taken despite never showing 100%

8.PPSSPP sometimes when I load a save state it crashes the core. save state loading might start from 75% to 85%  when it does happen. 

So far I have found Lakka OS to be frustrating  and I doubt this is all the issues I will find so far with version  Lakka-Generic x86_64-2.2.2

I don't want to install retroarch on Linux Mint because i find it a waste of resources with all the other OS programs/resources running will take away from game emulation. Which is why I like the idea of something like Lakka OS

Mr-Jake, 2019-08-11:
Lakka does have documentation for BIOS. It is located here:
https://www.lakka.tv/doc/bioses/
https://www.lakka.tv/doc/accessing-lakka-filesystem/

The PS2 core Play is new so it would not be found in 2.2.2 since that image was created months ago. But the core and infos can be downloaded manually and added to Lakka. Perhaps it would work better in the updated Lakka nightly builds.

Mr-Jake, 2019-08-11:
Lakka does have documentation for BIOS. It is located here:
https://www.lakka.tv/doc/bioses/
https://www.lakka.tv/doc/accessing-lakka-filesystem/

BIOS Information Hub:
http://docs.libretro.com/library/bios/

Atari 5200:
http://docs.libretro.com/library/atari800/

The PS2 core Play is new so it would not be found in 2.2.2 since that image was created months ago. But the core and infos can be downloaded manually and added to Lakka. Perhaps it would work better in the updated Lakka nightly builds.

ToKe79, 2019-08-13:
I would suggest downloading latest nightly - it has also the assets for PPSSPP core included, which might solve the issue with black screen - no assets were found so nothing was displayed - if you would press circle (as would be displayed if assets were present), you would return to the game. as pointed above, check the [Lakka documentation](https://www.lakka.tv/doc/home/) for more help or join us in channel #lakkatv on [Libretro Discord server](https://discordapp.com/invite/chjzzQx) or via IRC (#lakkatv on Freenode).

Gitlab1111, 2019-08-16:
After further reading and placing the bios I have still have the problems for Atari5200 and Dreamcast which uses Reicast.  Reicast is not on the Retroarch site but instead they have flycast, which I don't know why Reicast is not listed there.  I think something might be wrong with the cores because after trying out other bios sources for further testing  I still cannot get those two cores to work.  Then again the PPSSPP core functions worse than the one that was available on the linux version of Retroarch about 2 months ago. As for the Dreamcast core, I used a few other games to test as well and yet still nothing, the screen goes black then it exits as if the core shutdown.

I'm not going to test this out any further I have already put in enough time and gained enough frustration with it.

ToKe79, 2019-08-16:
If you are using "stable" version, it is quite outdated. For PC I would recommend using the devel version based on new (9.2) build system provided by LibreELEC team - http://nightly.builds.lakka.tv/Lakka-LE9.2/latest/Generic.x86_64/

crazypink9999, 2019-08-20:
Lakka-LE9.2-X86_64 System has been able to run FlyCast perfectly, but all SS cores don't work properly.

crazypink9999, 2019-08-21:
Do not update the new Flycast core, otherwise it will not run. Update the new Yabause core to run SS games properly.

Gitlab1111, 2019-08-25:
@ToKe79 I tried the Nightly version you mentioned. An odd thing is that it isn't  recognizing bios for PSx. I get a Bios missing under core info. I think the same is going on for the PS2 core.

ToKe79, 2019-08-25:
I did not yet try the PS2 emulator core (Play!), but according to https://github.com/libretro/libretro-super/blob/master/dist/info/play_libretro.info it does not require a BIOS file.
regarding psx bioses - can you post result of `md5sum /storage/system/scph*.bin /tmp/system/scph*.bin`? I am not sure, if the latest builds from LE9.2 base include a fix for the overlayfs (https://github.com/libretro/Lakka-LibreELEC/commit/9131149377c153fa2366935daffe6db85329549a). if the hashes are different (i.e. hashes from /storage/system != /tmp/system), it means, the fix is not included. i would then rebuild the images asap.

Gitlab1111, 2019-08-31:
@ToKe79 I noticed that the issue also extends to me not being able to upscale the resolution on the PSx core either.  For PSx, I use  USA region games. I will try again on the PS2 core using USA region games.

490f666e1afb15b7362b406ed1cea246  scph5001.bin

PS2 d5ce2c7d119f563ce04bc04dbc3a323e  scph39001.bin

gouchi, 2019-09-01:
Hi,

You should  use [Beetle PSX HW core](https://www.youtube.com/watch?v=xVRTwIHoNoo) on Generic platform as Haswell gpu can [handle OpenGL 3.3](http://docs.libretro.com/library/beetle_psx_hw/#requirements) for the [enhancements](http://docs.libretro.com/library/beetle_psx_hw/#core-options).

Please bear in mind that Play! is still a WIP and just some [games are playable](https://github.com/jpd002/Play-Compatibility/issues?q=is%3Aissue+is%3Aopen+label%3Astate-playable). You don't need bios for it. But indeed, I tried with a playable game (which works with Libretro buildbot core) and I got segfault. This will be maybe fixed with [this update ](https://github.com/libretro/Lakka-LibreELEC/pull/764) when we had it to the LE 9.2 tree.

iokzero, 2019-09-04:
Hi

@ToKe79 , sense you mentioned LE9.2, Will Lakka go native Vulkan support on the Generic PC edition?

Couple thing might worth a mention, in case people didn't know:

Flycast is basically a newer Reicast, I believe the leadership has change, mainly from the retroarch side of thing. Retroarch did NOTHING to notice anyone that Flycast is formerly known Reicast! Thumb down for that, but now you know. 

Reicat from Lakka 2.2.2 Generic PC x64 is broken, I know because I reported that bug. (It should be fixed by now? I got a special built to fix mine.)

PPSSPP has been hit or miss for a very long time, (at least for a year or so). Not sure what's going, always problems here and there. More testing is definitely needed in the future for stable release.

Play! is fairly new (like months), and I would say, don't even bother with it yet. Just like Dolphin for Wii. It's getting a lot better, but I wouldn't depended on them. (And the reason I mention that is because it really isn't Lakka's fault they aren't not ready... )

Generally speaking, Retroarch is very picky, both on "naming of Bios", the "authenticity of the Bios", as well as ROM/ISO it can run. As you can see, as @Gitlab1111 need to rename the PSx Bio to scph5501, from scph5001, for REARMED core to recognized it. This can mean two things: REARMED only recognized a BIOS name scph5501 (even if the actual file is BIOS scph5001). or That BIOS is actually scph5501 but got named scph5001 by accident. 

My recommendation on that is to verify all your BIOS with the RetroArch WM by pushing F5 (I think it only work if RetroArch is installed on top of an OS). Once you are in the RetroARch WM, on the Right, it will tells you all the Required / Optional BIOS files you need for that core, and GREEN it ,if the BIOS pass the checksum and being named currently. So you know for sure which one should work correctly, and which one is needs works or to be added. 

Yes, getting everything setup currently is a lot of works

 But I must say, the Lakka team did a fantastic job to getting everything together and making it as plug and play as possible. BIOS setup being annoying is really because emulators require the correct file to be run.... and the best both RetroArch and Lakka can do is to provide tools to make it as simple as possible.

natinusala, 2019-09-04:
> Will Lakka go native Vulkan support on the Generic PC edition?
Probably once mesa has vulkan support for all drivers (currently only radeon is supported IIRC)

> PPSSPP has been hit or miss for a very long time
The core has been left unmaintained for a very, very long time. m4xw is taking care of it now, and I hope to be able to use his version soon while waiting for all the fixes to be upstreamed.

iokzero, 2019-09-04:
@natinusala 

That explain a lot.

Surprised that Intel didn't get supported yet, but it does seem to me that LE9.2 final will be using Kernel 5.3 along with Mesa 19.1+ (I assume) which should have Vulkan support for Gen8+ integrated graphic from Intel. But I guess that's Lakka 3.0 and beyond.

Anyways, looking forward to your next release!

Gitlab1111, 2019-09-08:
Im still using a nightly build for 8-25-2019 at the moment. How do I enable cheats? I am trying to use it for PSx HW core. I might be doing it wrong but I added them to system/cheats/ which is a folder I created and then had the directory redirect itself to it.  I then tried adding it to a folder that appeared after testing that with 2 games.  
2. Is there a way to udpate to new nightly builds without doing a format and fresh install?
3. When using the PSx HW core I noticed with this USA region Resident Evil 3 that there's a glitch in the inventory screen. Bad Rom? Upscaling issue (4x)?
[image]
[image]
[image]
[image]

Am I doing something wrong here?

gouchi, 2019-09-10:
Please open one subject per issue.

@Gitlab1111 Your initial issue is fixed or not ?

Gitlab1111, 2019-09-11:
For my first post when I had the stable  Lakka-Generic x86_64-2.2.2 version. Most of those have been fixed upon nightly releases.  At the time of this writing the stable versions are still not available.  I have been using nightly builds since it was recommended. 

 3. I still find believe Lakka OS needs to have some Documentation updated. I have been resorting to skimming through Lakka and Retroarch documentation for my troubleshooting. IE recently the cheat section. I figured it out myself after going through some of the options and advance options as well.
4. Ties in with issue 3, still no Atari 5200 bios explanation. I had to read Retroarchs before deciding to dump the bios in my system folder.
5. The PS2 core I was not able to get working. Even in the other nightly builds I was not able to get the games to run. 

I'm not returning to that stable 2.2.2 version for further testing/playing. I am however going to update it to the newer 9-11 2019 nightly build.

gouchi, 2019-09-11:
Hi,

Thank you for your feedback.

3. & 4. Indeed some documentations need some update and you have to switch between RA and Lakka documentation. If you want to submit some update to [Lakka documentation](https://github.com/libretro/Lakka-LibreELEC/wiki), you can try to use this repository 
`https://github.com/libretro/Lakka-LibreELEC.wiki.git`

5. Please open a new issue for Play! core and remember to use [playable games](https://github.com/jpd002/Play-Compatibility/issues?q=is%3Aissue+is%3Aopen+label%3Astate-playable).

Thank you.

closed 2019-09-11 by gouchi