libretro/Lakka-LibreELEC #749: Raspberry Pi 4 - Online Update

microadam, 2019-08-14:
Hello,

Using the latest RPI 4 nightly (13/08) and trying to update the cores to make sure I have the latest (pretty sure flycast is out of date from reading the issue tracker and seeing missing features).

Whenever I try and update, I just get a "Task Failed" message. I have seen quite a few posts about this and everyone says to check the "core_updater_buildbot_url" in the retroarch.cfg file. I have checked this and it looks correct (http://buildbot.libretro.com/nightly/linux/armhf/latest/) I have also tried using http://buildbot.libretro.com/nightly/linux/armv7-neon-hf/latest/ but still the same issue.

I have checked the logs and nothing comes out when using verbose mode when I click the update button (literally nothing, the logs don't change).

Is this a bug? Or have I just done something wrong, or is this just too new to work at present?

Many thanks

P.s I did try posting this on the forum, but as a new user I can't seem to post new topics

slav0nic, 2019-08-14:
manual upgrade also don't work:
`wget -P /storage/.update http://nightly.builds.lakka.tv/....img.gz ; reboot` (rpi2.rpi4.arm) got error `Missing KERNEL or SYSTEM!`

natinusala, 2019-08-14:
You need to use the .tar to update, not the .img.gz

Also the online updater is purposely disabled on Lakka

slav0nic, 2019-08-14:
http://nightly.builds.lakka.tv/2019-08-13/RPi2.RPi4.arm/ here is no any `tar` across noob

natinusala, 2019-08-14:
Oh right, @ToKe79 would it be costly to add update archives to the nightlies? Maybe only for RPi4 for the time being?

ToKe79, 2019-08-14:
I am updating usually using the .img.gz file (amlogic, generic), but okay, I will leave the tarball with kernel/system for upcoming builds.

ToKe79, 2019-08-14:
@slav0nic I have uploaded the update tarball to the latest RPi4 nightly folder. please, let us know if the update with tarball works.

microadam, 2019-08-15:
Hi @natinusala out of interest, how come the online update is purposely disabled?  What you guys have been discussing looks like the process of upgrading from one nightly to another nightly without having to re-image the whole SD card. Is that correct?

I am curious as to how I go about updating cores from http://buildbot.libretro.com/nightly/linux/armhf/latest/

Specifically, I am curious how I update flycast to the latest version mentioned here:

https://www.libretro.com/index.php/flycast-progress-report-hle-bios-no-longer-requires-external-bios/

As that version didn't appear to be in the last nightly for RPI4 (unless its just not available for RPI 4 yet?)

Many yhanks

slav0nic, 2019-08-15:
@ToKe79 system update - ok, kernel - no, got error like `stat: Can't stat /flash/KERNEL ; No such file on directory` looks like `UPDATE_KERNEL=KERNEL` in init file

natinusala, 2019-08-15:
Oh, the raspberry pi doesn't use KERNEL but kernel.img instead right? It's the same as the RPi 3, it should still work

As for the core updater, we would need to provide each core for each platform we support and we don't want to setup such an infrastructure

slav0nic, 2019-08-15:
> Oh, the raspberry pi doesn't use KERNEL but kernel.img instead right?

yep
```

# ls /flash/KERNEL /flash/kernel.img
ls: /flash/KERNEL: No such file or directory
/flash/kernel.img
```

microadam, 2019-08-15:
@natinusala 

> As for the core updater, we would need to provide each core for each platform we support and we don't want to setup such an infrastructure

Forgive me, I thought that was the whole point of the buildbot site (http://buildbot.libretro.com/nightly/). That has every platform, you go into each platform and that lists all of the cores that are available for that platform?

If we are unable to update via the Online Updater, is there a way to manually update each core? Or are each of the cores only updated every so often and then bundled as part of a nightly release (e.g here http://nightly.builds.lakka.tv/2019-08-15/RPi2.RPi4.arm/).

Apologies for the questions, just trying to understand how this is all supposed to work (couldn't find too much info on updating in the docs)

ToKe79, 2019-08-15:
@microadam you can manually override provided cores by placing your own or those from buildbot into /storage/cores.

ToKe79, 2019-08-15:
and the reason why we do not "support" online updater - if we provide support, we must know what version which core is. if user updates their core(s), they have to ask for support in the respective core repository. and again, Lakka is quite different from other OSes - no window manager, read-only rootfs, no package manager, everything runs under the user root, etc. - it is more like an embedded system, not full featured OS.

microadam, 2019-08-15:
@ToKe79 thanks, I thought that might have been the case in terms of manually updating. From the looks of things though, there is no Flycast core in the buildbot for ARM (http://buildbot.libretro.com/nightly/linux/armv7-neon-hf/latest/) Does it have a different name under ARM? I can see flycast under windows and OSX, so I had assumed it would be the same name. Or do we have to manually compile ourselves for ARM?

ToKe79, 2019-08-15:
for the kernel - this is the final [init script from busybox](http://nightly.builds.lakka.tv/members/vudiq/rpi4-busybox-init.txt).

slav0nic, 2019-08-15:
@microadam Online Core updater available in nightly in `Load Core` - `Download a Core`

ToKe79, 2019-08-15:
@slav0nic nice find
@microadam there seems to be no flycast recipe for armhf:
https://github.com/libretro/libretro-super/blob/053769bf82eff1cd601b4f24d58cb414f7d9c511/recipes/linux/cores-linux-armhf-generic#L22-L31

microadam, 2019-08-15:
@slav0nic ah! Awesome, is that from the most recent nightly build? I will try that later.
@ToKe79 That would explain it then I guess! Do you know if there is a reason its missing? or is it simple enough just to add a new line to get builds of it being generated?

ToKe79, 2019-08-15:
also - each lakka device uses different OPENGLES provider - I am no developer, but I think if we compile core X against mali-rockchip, it is different than compiled against e.g. opengl-meson. so maybe that is also one of the reasons why Lakka does not support online updater by design. it is not the case for non-3D cores, but if somebody really wants to update the cores, they can put them into /storage/cores, for the rest - the Lakka should stay as simple as possible (also some features/menu items are disabled by default in Lakka).

gouchi, 2019-08-15:
[Lakka doesn’t expose RetroArch Online Updater (buildbot). Why?](https://www.lakka.tv/doc/faq/#lakka-doesnt-expose-retroarch-online-updater-buildbot-why)

microadam, 2019-08-15:
@gouchi thanks for pointing that out! I looked through the FAQ for other things and must have missed that! Looks like that may be changing with what @slav0nic mentioned though?

microadam, 2019-08-15:
https://www.lakka.tv/doc/upgrading-lakka/

This is what confused me as it mentions about an Online Updater that should work in latest versions

slav0nic, 2019-08-15:
@microadam maybe i manual set `menu_show_core_updater = "true"` at `/storage/.config/retroarch/retroarch.cfg`

nightly builds more for devs and advanced users

ToKe79, 2019-08-15:
the online updater is meant as update of the Lakka system, when new stable/rc version is released.
of you are running nighty build, this would do a downgrade, as latest stable < latest nightly. the online updater (of the Lakka system) does not have the information about the nightly devel builds.

br6, 2019-09-02:
Hello,
Does updating using the libreelec method of copying tar gz to the .update work on the following build?
http://nightly.builds.lakka.tv/Lakka-LE9.2/latest/RPi4.arm/ which is installed via noobs on rpi4

to any of the latest nightly rpi2.rpi4 builds, I have tried gz and tar files but they dont update, the files are removed on reboot

ToKe79, 2019-09-02:
Those are incompatible versions - the rpi2.rpi4 builds are based on LE8.2 (master branch of Lakka) - they have different `/etc/os-release` (or which file is checked during the update process), so the update fails. The update works only between the LE9.2 builds same way as on LibreELEC (put the `.img`/`.img.gz`/`.tar` file to the `/storage/.update` folder. Not sure if the fact, that the install is from noobs implies different procedure.

You can put `.nocompat` empty file to the `.update` folder (i.e. `touch /storage/.update/.nocompat)`, but I would not advise to do so. But if you are not afraid to break things (you can always re-install), feel free to experiment.

br6, 2019-09-02:
> Those are incompatible versions - the rpi2.rpi4 builds are based on LE8.2 (master branch of Lakka) - they have different `/etc/os-release` (or which file is checked during the update process), so [...]

Thanks that clears things up.

closed 2019-09-04 by natinusala

slav0nic, 2019-09-19:
just tried upgrade to 2.3 release from dev (pi4 platform) and still have error related to `Missing KERNEL`

natinusala, 2019-09-19:
This is normal, your old dev build still had the issue

Install 2.3 manually and subsequent updates should run fine

ToKe79, 2019-09-19:
isnt there a possibility to edit the commandline for RPi via a text file? adding KERNEL=/kernel.img should do the trick, right?

natinusala, 2019-09-19:
I suppose you can do it manually in config.txt

ToKe79, 2019-09-19:
`BOOT_IMAGE=/kernel.img`

PR #767 (2019-09-04): RPi4: fix CMDLINE
fixes #749