libretro/Lakka-LibreELEC #753: Issues with SSH for headless install

hjc25, 2019-08-18:
If you are sure that it's a bug in Lakka, please answer the following questions:
- Nightly 2019-08-15
- Raspberry Pi 4
- First time boot and attempt to access via SSH
- Access to RPi4 and Lakka over SSH
- Connection refused

There is no logs. I have no microHDMI cable to use so I wanted headless install. How can I enable SSH within the file system to then get something like VNC server going?

ToKe79, 2019-08-18:
on first boot you need to wait for storage partition to expand, then the system is rebooted. ssh (for security reasons) is not started by default, you have to enable it manually either via retroarch gui, or via command line arguments (add `ssh`), or by adding empty `sshd.conf` file to `/storage/.cache/services` folder. i also suggest adding `retroarch=0` to command line arguments, of you are running headless, as you probably do not want to start retroarch, before you solve the VNC part, which is currently not supported by Lakka ( = no issue). i suggest getting an hdmi cable for your rpi4.

closed 2019-08-18 by ToKe79

hjc25, 2019-08-18:
[image]

Thank you! It has worked well. Now it can be possible to try using VNC with it. My other components like microHDMI cables are still coming from China so I wanted to check for it to work.