libretro/Lakka-LibreELEC #754: Missing cores on RPi 1/0 image

duduke, 2019-08-24:
Following cores:
fbalpha2012_cps1_libretro.so
fbalpha2012_cps2_libretro.so
fbalpha2012_libretro.so
fbalpha2012_neogeo_libretro.so
fbalpha_libretro.so
mame2000_libretro.so

Are missing on the RPi 0/1 build (i've tried the latest nightly from Aug-15). 
I've tried adding them to **/storage/cores** from [https://buildbot.libretro.com/nightly/linux/armhf/latest/](url) but they cause a segmentation fault.
These cores run much better than mame2003 and fbaneo on the RPi0, is there a plan to add them back?

RobLoach, 2019-08-24:
fbalpha is now fbneo. Mame is a package, so I'm unsure where it has gone:
https://github.com/libretro/Lakka-LibreELEC/tree/master/packages/libretro

twinaphex, 2019-08-24:
I believe he says FB Neo is too slow though on his device. Even though it's natural we have a lot of disdain for FBAlpha considering what was done, I think it would be remiss for us to then simply no longer offer these standalone (CPS1/CPS2/Neogeo) cores of older FBA builds for the target hardware in question. It would be like not deciding to ship older MAME cores anymore for hardware that desperately needs it for performance. I just think in this case we need to meet the needs of the user, politics be damned.

natinusala, 2019-08-24:
But the fbalpha core doesn't exist anymore does it

duduke, 2019-08-24:
Maybe development had moved to FBNeo, but they still exists:
https://buildbot.libretro.com/nightly/linux/armhf/latest/

And those older cores are the only ones that I know that run most of the CPS1/2 library at good FPS on the Pi 0.

natinusala, 2019-09-10:
Will be solved with the 2.3 release

closed 2019-09-10 by natinusala