libretro/Lakka-LibreELEC #755: [Feature Request] Add Retroflag GPi Case Support

duduke, 2019-08-24:
As the title says.

Currently, to make it work add the following to **config.txt**:
`display_rotate=1
dtoverlay=dpi24
overscan_left=0
overscan_right=0
overscan_top=0
overscan_bottom=0
framebuffer_width=320
framebuffer_height=240
enable_dpi_lcd=1
display_default_lcd=1
dpi_group=2
dpi_mode=87
dpi_output_format=0x6016
#hdmi_timings=320 0 28 18 28 480 0 2 2 4 0 0 0 60 0 32000000 6
hdmi_timings=240 1 38 10 20 320 1 20 4 4 0 0 0 60 0 6400000 1
dtoverlay=pwm-2chan,pin=18,func=2,pin2=19,func2=2
disable_pvt=1
disable_audio_dither=1
dtoverlay=pwm-audio-pi-zero
`
Comment out `hdmi_driver=2` from **distconfig.txt**

Download [http://download.retroflag.com/Products/GPi_Case/GPi_Case_patch.zip](url)

Copy and overwrite from the above zip **patch_files/overlays/pwm-audio-pi-zero.dtbo** and **dpi24.dtbo** to **overlays** on the Lakka SD Card.

Once booted, go to Lakka Audio settings and fill in **Audio Device**:
`default:CARD=ALSA`

I haven't been able to test shutdown scripts yet.

mlnlbrt, 2019-08-28:
Hi @duduke 

I've started working on this, however my GPiCase didn't arrive yet, so I didn't have a chance to test it.

Here is the first image I've built today:
http://buildserver3.lakka.tv/Lakka-RPi.GPICase.arm-2.2-devel-20190828174428-r28326-gb03cafd.img.gz

If you are brave enough you can try to test it. Basically this is Lakka for RPi0 minus all cores plus  modified distroconfig.txt and dtbo overlays, so it should run out-of-the-box after burning the image to a SD card. No shutdown scripts yet.

Thanks

duduke, 2019-08-28:
Just tested it, and it seems to be working correctly.
systemctl complains that it Failed to mount directory:
[image]

duduke, 2019-08-28:
This can be used as a base for the shutdown script. its for NESpi, but can be adapted for GPi.
https://github.com/marcelonovaes/lakka_nespi_power

mlnlbrt, 2019-08-30:
Hi @duduke 

Fixed systemctl error and added safe shutdown scripts:
http://buildserver3.lakka.tv/Lakka-RPi.GPICase.arm-2.2-devel-20190830235724-r28331-gdd4c1e0.img.gz

Would you be able to test this and let me know if safe shutdown works correctly?

Thanks

siuying, 2019-08-31:
The screen, controller and shutdown script work as expected. Great job @mlnlbrt

mlnlbrt, 2019-08-31:
Thanks! Is sound working too?

siuying, 2019-08-31:
Can confirm sound working too. 

One problem I had with GPi case + Lakka is the RetroArch interface cannot find the AP of my home (It will only scan once and it will never return my AP). I tried to find way to preconfig Wifi but no luck. I cant really connect keyboard/ethernet with disassemble everything. Any idea how I can workaround this?

duduke, 2019-09-01:
Maybe the WIFI issue is local? I have WIFI working fine.

siuying, 2019-09-02:
Yep, I don't think it is this specific to this change. It just happen LibreELEC not pick up my local wifi and I have no way to setup manually. It do work fine on other distro.

duduke, 2019-09-02:
mlnlbrt, i've opened another issue which is somewhat related: https://github.com/libretro/Lakka-LibreELEC/issues/754
For LAKKA to viable on the GPi Case, I believe older mame/fba cores should be bundled.
Can this be done as part of this PR or it requires a bigger change to LAKKA?

mlnlbrt, 2019-09-02:
@duduke I've been planning to do a separate PR for this thing. It would contain some of the cores you mentioned and stuff like using faster gpu plugin for pcsx_rearmed.

ngsoftwaredev, 2019-09-03:
May I suggest to add the following two lines to distroconfig.txt : 

# Disable the ACT LED on the Pi Zero.
dtparam=act_led_trigger=none
dtparam=act_led_activelow=on

As per [https://www.jeffgeerling.com/blogs/jeff-geerling/controlling-pwr-act-leds-raspberry-pi](url), this will deactivate the ACT led and save a bit of battery. it has been applied on other distros such as Recalbox, Super Retropie.

Best regards.

duduke, 2019-09-03:
Also this:
extra_transpose_buffer=2
As per the following thread:
https://www.google.com/amp/s/amp.reddit.com/r/retroflag_gpi/comments/cuujvz/screen_tearing_fix/

mlnlbrt, 2019-09-03:
@ngsoftwaredev thanks for the tip, I've just added it:
https://github.com/libretro/Lakka-LibreELEC/pull/763/commits/897e9170994256c9fd508c7fbfc90fc3df94b9c4

@duduke it's already here, line 11:
https://github.com/libretro/Lakka-LibreELEC/pull/763/files#diff-b13e70cfd98772cf51623b8f40cf03b2#11

duduke, 2019-09-03:
Didn’t see it, Thanks.

duduke, 2019-09-03:
If we're in the business of conserving power (considering this runs on 3AAs), maybe this would be useful as well:
`/usr/bin/tvservice -o` Will powerdown HDMI for an extra 25mA.

Source: https://www.jeffgeerling.com/blogs/jeff-geerling/raspberry-pi-zero-conserve-energy

duduke, 2019-09-06:
@mlnlbrt I've seen that there's a new PR build: http://nightly.builds.lakka.tv/PRs/pr_763_897e917/
Tried that, but it is not booting as it doesn't have all the GPi Case specific settings in distoconfig.txt and I think is missing the right overlay files.

mlnlbrt, 2019-09-06:
@duduke for some reason this build does not include RPi.GPICase image, don't know why.

Anyway, here is a new build:
http://buildserver3.lakka.tv/Lakka-RPi.GPICase.arm-2.2-devel-20190906184157-r28340-g5c8a02d.img.gz

changelog:
* added mame2000 and fbalpha2012 cores
* removed some heavyweight cores like beetle-psx
* added HDMI powerdown
* fixed swapped buttons, now they should appear correctly in Retropad core
* GUI tweaks (switched to RGUI and adjusted fontsize in notifications)
* settings tweaks (integer scale = on, threaded video = off, Start + Select as menu combo)
* added splash screen for 240-line LCDs

duduke, 2019-09-07:
@mlnlbrt, The new build seems to be working fine:

- added mame2000 and fbalpha2012 cores
Performance looks great for the games i've tested:
CPS1/2: Caddis & Dinos, Final Fight, SF Alpha (SF Alpha 3 didn't load).
MAME2000: TMNT2, ThunderFox
For the sake of support, can you keep the previous MAME cores?
- removed some heavyweight cores like beetle-psx
- added HDMI powerdown
No issues observed.
- fixed swapped buttons, now they should appear correctly in Retropad core
A matter of taste :)
- GUI tweaks (switched to RGUI and adjusted fontsize in notifications)
Font size for notifications makes them now readable instead of taking the entire screen.
I think XMB works fine, for sake of performance I've disabled the ribbon.
- settings tweaks (integer scale = on, threaded video = off, Start + Select as menu combo)
Integer scale looks good considering the native screen res is 320x240.
Isn't threaded video suppose to provide more performance?
- added splash screen for 240-line LCDs
Looks great.

One final comment that might be useful: 
This commit: https://github.com/libretro/FBNeo/commit/8b53cc1a5ba795a988442efd26769f8794271b63
for FBNeo had re-enabled the CYCLONE 68k core which is what I think makes FBA2012 faster on RPi0/1 than FBNeo. 
Can you compile the latest FBNeo and i'll test it? if that works and performs adequately, then there is no need for the older FBA cores.

Thanks. Awesome work.

closed 2019-09-10 by natinusala