libretro/Lakka-LibreELEC #756: N64 and PS1 on RPi4

boschili, 2019-08-24:
I‘m currently running the august 13th Nightly Release on a Raspi 4 (2gb).
I have issues with the PS1 Core. 

1. I can‘t add Bioses manually. The Core loads a default Bios After reporting that no proper one was found, despite I loaded propper Bioses according to the DOC.

2. every now and then the Core freezes and I Need to hard-reset My Pi (unplug from and replug to the mains).

3. I generally have the impression of the entire system to be quite laggy. I mostly use the N64 and PS1 Emulators, which should be run by the Pi4 rather well , but These two emulators in particular Tend to be slow except for some  Games like SuperMarioKart64  or FIFA99, which tend to be Running mostly without 

Sorry for the weird small and capital letter-Thing in the post, but I‘m from Germany and My iPad refuses to write english, by Not beging set to do so :)

RobLoach, 2019-08-24:
> every now and then the Core freezes

That's often caused by overheating. What case are you using?

> generally have the impression of the entire system to be quite laggy.

N64 is very CPU intensive. I have doubts on it running smoothly on the hardware. Try a few of the different plugins and see if you have better performance with another plugin.

boschili, 2019-08-24:
[image]
I actually use the retailbox with heatsinks on cpu and ram and a 1.5“ Fan (Stolen from an old Pentium 4, sucking the air out of the box; still waiting for my flirc-case to be shipped) on top of it, as you can probably see in the picture. So I don’t think overheating is an issue. 

I will post a log, when I can get my hands in the Pi, cause I‘m not home at the moment. 

If you mean core by saying plugin, I‘m using the muppen-core, because the other one, of wich I forgot the name, works worse in comparison. If not, plz tell me what you mean, cause I‘d like to get it working desperately.

natinusala, 2019-08-24:
Try parallel-n64 on the RPi

boschili, 2019-08-24:
That‘s the name of the other core, thanks natinusala!
I did, as mentioned the core works better. As to be seen in a YouTube-video by ETA Prime the Pi4 is capable of emulating N64 well. He used the muppen-core, as far as I remember.

boschili, 2019-08-28:
so, now I put the bioses for the PS1 into the sytemfolder on the Pi's sd card, instead of redirecting the directory. bioses are being found now and the core stopped crashing.

by here I'd like to make a request to the developers:
please make either this redirecting functionality work, or remove it. I think it might be a little confusing to other users besides me to have an option avalable, that pretends to be working, and giving non-developers a hard time troubleshooting.

natinusala, 2019-08-28:
What do you call the `sytemfolder on the Pi's sd card`? What is ̀redirecting the directory`?

boschili, 2019-08-28:
quotation from the DOC:

"BIOSes must be placed in the system folder although there are certain exceptions described in the table below."

there is a folder on the sd card called "system" in which the bioses are to be put. 

by redirecting I mean (sorry, need to guess the names, my UI is german) 
->settings->directories->bioses
here you can tell retroarch where to look for bioses. in my case it did entirely not work to set the path to an external drive.

natinusala, 2019-08-28:
What value did you put exactly in that setting?

boschili, 2019-08-28:
the exact path was 
->roms (name of the drive)->bioses

ToKe79, 2019-08-28:
could be checksum issue - can you compare
```
$ md5sum /storage/roms/<yourdrive>/<bioses_folder>/scph*.bin
```
against
```
$ md5sum /storage/system/scph*.bin
```

natinusala, 2019-08-28:
Can you do it again, check the config file (`/storage/.config/retroarch/retroarch.cfg`) and send the full path here for the `system_directory` entry?

boschili, 2019-08-28:
@ToKe79 
I don't think it's a checksum-issue, because now it works with the bioses in the place they're supposed to be originally. 

@natinusala 
you want me to redirect to the external drive again, and then check the retroarch configfile for the directory the bioses are directed to. did I get that right?
the systemfolder I have the bioses in working state put now is
Lakka partition root ->system

natinusala, 2019-08-28:
Yes please, I'd like to see the full path as seen in the config file

boschili, 2019-08-28:
ok, but it'll take approx. 9 hours for me to post it. I'm at my desk at work right now :P

ToKe79, 2019-08-28:
@boschili no rush, we will wait - it can be either some weird character in the path or the checksum - I had this issue when overlayfs was compiled in kernel instead as a module - `md5sum /storage/system/scph5502.bin` returned different value than `md5sum /tmp/system/scph5502.bin` :)

boschili, 2019-08-28:
short question nearby: can I update to a later nightly without the need to reflash the sd-card?

natinusala, 2019-08-28:
You should be able to download it and update like any new version (download the file and put it in .updater, you have the process on the website)

boschili, 2019-08-28:
so, now I redirected the bios source to the external drive in the UI-settings. like before the core does not find its cores. 
as wished here's what the  config says:

system_directory = "~/system"

looks to me that you CAN change the setting in the UI, but it does not update the retroarch.cfg, right?

natinusala, 2019-08-28:
Do you exit RA properly after changing the config?

boschili, 2019-08-28:
do you mean rebooting the pi?

ToKe79, 2019-08-29:
@boschili change it in RA UI, quit/restart RA - the config is saved during RA shutdown (not device shutdown). then check what value is as `system_directory`.

natinusala, 2019-08-29:
Uh, the settings should be saved on device shutdown, if not it's a major bug

boschili, 2019-08-29:
I will try again and see if a restart of the RA UI solves the problem. You definitely will get an update!

ToKe79, 2019-08-29:
@natinusala settings should be saved also at device shutdown, but it is not necessary to do shutdown for the settings to be saved - that's what I meant :-)

boschili, 2019-08-29:
@RobLoach 
> N64 is very CPU intensive. I have doubts on it running smoothly on the hardware. Try a few of the different plugins and see if you have better performance with another plugin.

Little update on that:
After overclocking the cpu to 1.95GHz, gpu to 600MHz and v3d zu 750MHz the laggynes on the N64 mostly disappeared. Only games that are supposed to use the memory-Extension on the original N64 are still a little slow or frameskip a little. Seems as if the Pi4 is still a little to weak out of the box for a smooth experience with the provided N64 cores. Concearning the Choice of cores it did still not matter mich which one I used.

@natinusala 
Tried again several times to redirect the source for the Bioses to the external drive. 
No matter if I restarted RA or Lakka/Pi after changing the setting on the UI, the config was never updated corectly and remained at the path mentioned above, although the UI says, it did. I have not tried to alter the config directly via SSH in nano, because I was apparently a little scared to do so and it‘s obviously not the supposed way to go.
Looks like I found an issue :P

boschili, 2019-08-29:
Is it possible that it has something to do with the „read-only filesystem“ design of LibreELEC?

natinusala, 2019-08-29:
No, the config isn't in the read only filesystem

boschili, 2019-08-31:
Hey, me again!
I‘d like to note, that redirecting the directory for the thumbnails to somewhere else than default isn‘t working either, behaving the same way discussed above. Checked the config after several reboots of RA and Lakka, and the corresponding changes were not written to the config. 
Thumbnails in the default directory on the SD card work, except for PS1, but that could still be a naming-issue

Renetrox, 2019-09-02:
> > I can‘t add Bioses manually
> 
> BIOS files are expected in the system folder:
> https://www.lakka.tv/doc/bioses/
> 
> Post logs?
> 
> > every now and then the Core freezes
> 
> That's often caused by overheating. What case are you using?
> 
> > generally have the impression of the entire system to be quite laggy.
> 
> N64 is very CPU intensive. I have doubts on it running smoothly on the hardware. Try a few of the different plugins and see if you have better performance with another plugin.

Sorry for sneaking into the subject, but with respect to N64, could the option of "frameskip" be included in the future ?, considering that the core options will have sub-levels with the advances of retroarch. I miss trying to configure as much as possible the emulator as it happens in android with mupen64plusFZ.

br6, 2019-09-02:
I cant even get n64 cores to work on 9.2 08/25 build, do i need to do something different

boschili, 2019-09-03:
@br6  
maybe yout should go ahead and try a non 9.2 version. 
I'm running (as mentioned) the august 13th nightly and besides some minor issues all cores that I tested (most 8, 16 and 32 bit cores) work fine now after some tweaking, except ppsspp. but that was to be expected.
muppen64 works fine also on that build, after I Overclocked my Pi4

closed 2020-08-18 by Ntemis