libretro/Lakka-LibreELEC #757: Booting with attached HD

boschili, 2019-08-24:
I‘m using an Rpi4 (2GB) with the nightly release Dated to august 13th.

In about 60% of attempts to Boot the System with an  external HD (WD Elements 2TB, USB3, NTFS) attached to one of the USB3 ports the OS does Not Boot to the UI. When I Unplug the drive before or during the boot-process it Boots up to the UI normally.
Automatic Mounting of any external Drive after Being fully booted up (no matter if it‘s a HD, SSD or flashdrive) works Fine.

Sorry Again for the smal/captial-letter Thing. Still writing on a german iPad :)

natinusala, 2019-08-24:
Can you provide retroarch logs? You have the guide in our documentation on the website

boschili, 2019-08-27:
strangely the issue did not occur since i had been posting it. maybe it was some kind of hickup in initializing the drive.

boschili, 2019-09-11:
strange new things going on at the external drive front:
I recently exchanged the WD-External HD mentioned above with an Intenso SSD. What I did is mirror the WD to the Intenso (both NTFS) and plugged it in to the Pi. strangely the SSD needs 2-3 minutes to be mounted until data can be read from it by the Pi, whereas with the WD it was a matter of seconds. for testing purposses formatting the SSD to fat32 reduced the mounting-time to approx. 30 sec. 
now mounted the data are being read reasonably faster than off of the HD. I just wonder if it's a matter of a shitty controller in the cheap SSD or if there's something wrong within lakka.
I think its strange enough that fat32 is being mounted faster than NTFS, on the SSD, and the HD is still mounting faster although being formatted to NTFS with (in my case) having 50% slower read/write times compared to the SSD in Win10 hooked pu to USB3.
I need to mention, that this occurs only on the Pi in lakka and neither in Raspbian, nor in Windows. 

Any suggestions anyone?

closed 2020-08-18 by Ntemis