libretro/Lakka-LibreELEC #759: Wrong screen-scaling on Rpi4

boschili, 2019-08-24:
Nightly august 13th release in RPi4 2gb

I‘m using the RPi on a tv with a resolution of 1280x720@60Hz via HDMI.

The OS recognizes 1440x1080@60Hz

Strange thing is: the scaling looks right but feels kinda wrong. Maybe there’s some strange kind of tearing or an issue with rescaling on the tv. If i manually set the right resolution for the tv the picture shown on screen feels right when playing but looks wrong, because of being kind of Pixeled.

RobLoach, 2019-08-24:
Does the same happen on Windows?

boschili, 2019-08-24:
No, my Win10 laptop recognizes the tv with its native resolution of 1280x720@60Hz

I‘m not home at the moment. Will post a log, as soon as I get my hands in the Pi.

boschili, 2019-08-28:
Problem solved. retroarch did a fualty scaling. I went ahead and fitted the picture to the screen manually in the video-settings. looks anf feels fine now!

closed 2019-08-28 by boschili