libretro/Lakka-LibreELEC #760: Bluetooth Controller Issues on RPi4

cscheib, 2019-08-25:
- **Which version of Lakka are you using?** Latest nightly: Lakka-RPi2.RPi4.arm-2.2-devel-20190815003103-r28324-g35a9e03144.img
- **What system hardware are you using?** Raspberry Pi 4B / 4GB memory at default clock speeds
- **What did you do?** Connected _8BitDo SN30 Pro_ via Bluetooth
- **What did you expect to happen?** Fully working controller
- **What happened instead?**:
D-Pad up/down arrow keys are essentially useless, they jump all the way up or all the was down on any given list on the main page of the user interface.  Basically the sensitivity is incredibly high.

Plugging in a keyboard to USB, or starting a game then exiting it, seem to (temporarily) resolve the issue.  Turning the controller fully off and then on does not resolve the issue.  A fresh boot causes the issue to recur.  I haven't been able to fully suss out the conditions upon which it occurs, but I'm slowly honing in on them.

jholloway, 2019-09-03:
I can confirm this issue persists on the 2019-09-02 nightly.

natinusala, 2019-09-03:
This is not the first time I see such an issue on Lakka, sadly I am not aware of any fix.

Can you try the 9.2 based version of Lakka in the meantime? You can find it here http://nightly.builds.lakka.tv/Lakka-LE9.2/latest/RPi4.arm/

ToKe79, 2019-09-03:
These controller issues seem like the gamepad is considered to be a keyboard (up / down => pgup / pgdn). I think I have seen an udev rule to fix this.

ToKe79, 2019-09-03:
https://discourse.coreelec.org/t/bluetooth-controllers-are-not-working/858

cscheib, 2019-09-03:
@ToKe79 that does make a certain amount of sense, considering the unplug/replug of the keyboard seems to kinda kick it into working (sometimes)

@natinusala what's the difference between the 9.2 based version and the normal nightlies?  I haven't really been following the project long, so not sure the intricacies of the development cycle.

What's the preferred upgrade method for pre-release versions?  Should I pull my configs off, reflash the card, then toss the configs back on?  Or can I drop the img/kernel/system(?) files in the normal upgrade spot?

ToKe79, 2019-09-03:
@cscheib the difference between the 9.2 builds and the regular nightlies is, that the regular nightlies are built from the master branch of Lakka - current development branch. The 9.2 builds are based on LibreELEC 9.2 (actually 9.1.xxx, future 9.2) build system/packages. It has newer kernel, newer libs, etc. - see the Lakka-LE9.2 branch. The master branch of Lakka is based on LibreELEC 8.2.

If you want to test the 9.2 branch / builds, I would suggest using separate SD card and see how it works for you. I have only tested the Generic builds, as I do not have any other HW I could test on (Amlogic builds - S905/S912 - are not ready for "prime time", which I tested too). For the Generic build I did not find any showstoppers / No-Goes.

cscheib, 2019-09-04:
Seems to work so far on 9.2 nightly, fresh SD card... but now wifi doesn't want to seem to want to come up after a hard boot.  I have to navigate to wifi in the menu and reconnect.  If there's an issue someone's aware of for that, it'd be awesome if someone could point me there.  Otherwise I'll open another when I get an opportunity (should one be opening issues against Lakka/9.2 nightly?)

jholloway, 2019-09-04:
I just tried the 9.2 based build as suggested, and I'm having the same issues as before. I'm using an Xbox One S controller via Bluetooth using the manual pairing methods suggested at lakka.tv. After connecting and pairing, I am still experiencing the jumpy menus where it appears to move multiple directional inputs when I just press an input once. @cscheib's original workaround of plugging in and unplugging a USB keyboard isn't making any difference.

natinusala, 2019-09-04:
Can you try on Raspbian Stretch with regular RetroArch to rule out any RetroArch / hardware issue?

jholloway, 2019-09-04:
@natinusala I am using a Raspberry Pi 4 board like @cscheib. My understanding is that Stretch will not run properly on this board. I tried installing Buster and RetroArch, but getting it to build properly on the Pi4 in its current state is finicky at best, so I'm having trouble testing it.

I will say that I've had similar issues with the selection jumping around more than intended when using RetroArch on a PlayStation Classic device, so it seems like it could be an underlying problem and not specific to Lakka.

natinusala, 2019-09-04:
Isn't stretch the latest one? Why doesn't it work?

jholloway, 2019-09-04:
No, Raspbian Buster is the latest official Raspbian release, based on Debian Buster. https://www.raspberrypi.org/downloads/raspbian/

natinusala, 2019-09-04:
Whoops I meant Buster then. The last one. Can you try installing latest RA? It should be easier now

CRP, 2019-09-14:
For what's it worth, I just tried replacing my 8Bitdo N30pro with a newer M30 controller. The first always displays the behaviour described by the original poster, while the newer one works flawlessly. So there might be an issue with older controllers' bluetooth protocol.

closed 2020-08-18 by Ntemis