libretro/Lakka-LibreELEC #765: No menus on Lakka nightly build 2019-09-02

kidcorpse, 2019-09-02:
Lakka nightly build released on 2019-09-02 for RPI2-RPI4 will not scan games to menu, not will it save to favorites menu.

natinusala, 2019-09-02:
Can you elaborate? Does it not scan, or does it scan but doesn't add the games to the menu?

Did you try restarting RetroArch after scanning?

kidcorpse, 2019-09-02:
The latter... It scans but doesn't add to the menu. Also doesn't scan the box arts like the previous versions. And yes, but restarting didn't help. The games with fine, just gotta go to them the long way compared to the previous versions. Not complaining, just felt it was a quirk worth nothing for dev-purps 🙂

natinusala, 2019-09-02:
Sure, thanks for reporting that issue, it sure is a problem!

Can you access the Lakka filesystem (see the website if you don't know how to) and send the `.config/retroarch/retroarch.cfg` file here (in a pastebin maybe)? Thanks!

kidcorpse, 2019-09-02:
I'll do my best. I'm still learning all this stuff, but I'm a fast learner and I've learned so much in the last couple months. (I'm building an arcade machine for my son so you guys are helping me so much with the process with all your hard work on the nightlys) Thanx so much BTW, for all you guys do, and for responding so quickly.

natinusala, 2019-09-02:
You're welcome, good luck for your project, the important part is to have fun

kidcorpse, 2019-09-02:
I think this is it...please tell me if I'm wrong.

ai_service_enable = "false"
ai_service_mode = "0"
ai_service_url = "http://localhost:4404/"
all_users_control_menu = "true"
apply_cheats_after_load = "false"
apply_cheats_after_toggle = "false"
aspect_ratio_index = "1"
assets_directory = "/tmp/assets"
audio_block_frames = "0"
audio_device = "hw:0,1"
audio_driver = "alsathread"
audio_dsp_plugin = ""
audio_enable = "true"
audio_enable_menu = "true"
audio_enable_menu_bgm = "true"
audio_enable_menu_cancel = "true"
audio_enable_menu_notice = "true"
audio_enable_menu_ok = "true"
audio_filter_dir = "/usr/share/audio_filters"
audio_latency = "64"
audio_max_timing_skew = "0.050000"
audio_mixer_mute_enable = "false"
audio_mixer_volume = "0.000000"
audio_mute_enable = "false"
audio_out_rate = "44100"
audio_rate_control = "true"
audio_rate_control_delta = "0.005000"
audio_resampler = "sinc"
audio_resampler_quality = "2"
audio_sync = "true"
audio_volume = "0.000000"
auto_overrides_enable = "true"
auto_remaps_enable = "true"
auto_screenshot_filename = "true"
auto_shaders_enable = "true"
automatically_add_content_to_playlist = "false"
autosave_interval = "0"
block_sram_overwrite = "false"
builtin_imageviewer_enable = "true"
builtin_mediaplayer_enable = "true"
bundle_assets_dst_path = ""
bundle_assets_dst_path_subdir = ""
bundle_assets_extract_enable = "false"
bundle_assets_extract_last_version = "0"
bundle_assets_extract_version_current = "0"
bundle_assets_src_path = ""
cache_directory = ""
camera_allow = "false"
camera_device = ""
camera_driver = "null"
cheat_database_path = "/tmp/database/cht"
check_firmware_before_loading = "false"
cheevos_auto_screenshot = "false"
cheevos_badges_enable = "false"
cheevos_enable = "false"
cheevos_hardcore_mode_enable = "false"
cheevos_leaderboards_enable = "false"
cheevos_password = ""
cheevos_test_unofficial = "false"
cheevos_token = ""
cheevos_username = ""
cheevos_verbose_enable = "false"
config_save_on_exit = "true"
content_database_path = "/tmp/database/rdb"
content_favorites_path = "~/.config/retroarch/content_favorites.lpl"
content_history_dir = ""
content_history_path = "~/.config/retroarch/content_history.lpl"
content_history_size = "100"
[... 161 lines cut ...]
input_load_state_axis = "nul"
input_load_state_btn = "nul"
input_load_state_mbtn = "nul"
input_max_users = "5"
input_menu_toggle = "f1"
input_menu_toggle_axis = "nul"
input_menu_toggle_btn = "nul"
input_menu_toggle_gamepad_combo = "2"
input_menu_toggle_mbtn = "nul"
input_movie_record_toggle = "o"
input_movie_record_toggle_axis = "nul"
input_movie_record_toggle_btn = "nul"
input_movie_record_toggle_mbtn = "nul"
input_netplay_game_watch = "i"
input_netplay_game_watch_axis = "nul"
input_netplay_game_watch_btn = "nul"
input_netplay_game_watch_mbtn = "nul"
input_netplay_host_toggle = "nul"
input_netplay_host_toggle_axis = "nul"
input_netplay_host_toggle_btn = "nul"
input_netplay_host_toggle_mbtn = "nul"
input_osk_toggle = "f12"
input_osk_toggle_axis = "nul"
input_osk_toggle_btn = "nul"
input_osk_toggle_mbtn = "

ToKe79, 2019-09-02:
@kidcorpse I have edited your comment for better readability, however, it looks like the config is not complete (mine has 3004 lines) and last line is
```
youtube_stream_key = ""
```
fyi - to insert text formatted in fixed width (like above), put three back ticks (```) in front and after the text you want to have formatted.

do you have any free space on your SD card?

kidcorpse, 2019-09-03:
'''ai_service_enable = "false"
ai_service_mode = "0"
ai_service_url = "http://localhost:4404/"
all_users_control_menu = "true"
apply_cheats_after_load = "false"
apply_cheats_after_toggle = "false"
aspect_ratio_index = "1"
assets_directory = "/tmp/assets"
audio_block_frames = "0"
audio_device = "hw:0,1"
audio_driver = "alsathread"
audio_dsp_plugin = ""
audio_enable = "true"
audio_enable_menu = "false"
audio_enable_menu_bgm = "false"
audio_enable_menu_cancel = "false"
audio_enable_menu_notice = "false"
audio_enable_menu_ok = "false"
audio_filter_dir = "/usr/share/audio_filters"
audio_latency = "64"
audio_max_timing_skew = "0.050000"
audio_mixer_mute_enable = "false"
audio_mixer_volume = "0.000000"
audio_mute_enable = "false"
audio_out_rate = "44100"
audio_rate_control = "true"
audio_rate_control_delta = "0.005000"
audio_resampler = "sinc"
audio_resampler_quality = "2"
audio_sync = "true"
audio_volume = "0.000000"
auto_overrides_enable = "true"
auto_remaps_enable = "true"
auto_screenshot_filename = "true"
auto_shaders_enable = "true"
automatically_add_content_to_playlist = "false"
autosave_interval = "0"
block_sram_overwrite = "false"
builtin_imageviewer_enable = "true"
builtin_mediaplayer_enable = "true"
bundle_assets_dst_path = ""
bundle_assets_dst_path_subdir = ""
bundle_assets_extract_enable = "false"
bundle_assets_extract_last_version = "0"
bundle_assets_extract_version_current = "0"
bundle_assets_src_path = ""
cache_directory = ""
camera_allow = "false"
camera_device = ""
camera_driver = "null"
cheat_database_path = "/tmp/database/cht"
check_firmware_before_loading = "false"
cheevos_auto_screenshot = "false"
cheevos_badges_enable = "false"
cheevos_enable = "false"
cheevos_hardcore_mode_enable = "false"
cheevos_leaderboards_enable = "false"
cheevos_password = ""
cheevos_test_unofficial = "false"
cheevos_token = ""
cheevos_username = ""
cheevos_verbose_enable = "false"
config_save_on_exit = "true"
content_database_path = "/tmp/database/rdb"
content_favorites_path = "~/.config/retroarch/content_favorites.lpl"
content_history_dir = ""
content_history_path = "~/.config/retroarch/content_history.lpl"
content_history_size = "100"
content_image_history_path = "~/.config/retroarch/content_image_history.lpl"
content_music_history_path = "~/.config/retroarch/content_music_history.lpl"
[... 2910 lines cut ...]
video_stream_scale_factor = "1"
video_stream_url = ""
video_swap_interval = "1"
video_threaded = "true"
video_vsync = "true"
video_window_opacity = "100"
video_window_save_positions = "false"
video_window_show_decorations = "true"
video_windowed_fullscreen = "false"
video_windowed_position_height = "720"
video_windowed_position_width = "1280"
video_windowed_position_x = "0"
video_windowed_position_y = "0"
vrr_runloop_enable = "false"
wifi_driver = "connmanctl"
xmb_alpha_factor = "75"
xmb_font = ""
xmb_layout = "0"
xmb_menu_color_theme = "8"
xmb_scale_factor = "100"
xmb_shadows_enable = "false"
xmb_theme = "5"
xmb_vertical_thumbnails = "false"
youtube_stream_key = ""
'''

kidcorpse, 2019-09-03:
I believe that's all of it, however I don't know why so many of these lines have black strikes through them. The original text I copied from doesn't appear to have the strikes...strange

natinusala, 2019-09-03:
I was looking for `content_show_playlists` and it is already set to true...

Do the same ROMs work on RetroArch PC (Windows or Linux)?

kidcorpse, 2019-09-04:
IDK, I don't use it on those. Only Raspberry Pi 4

natinusala, 2019-09-04:
Can you please try?

closed 2020-08-18 by Ntemis