libretro/Lakka-LibreELEC #771: Beeling GT1 Ultimate Wifi and Bluetooth don't works properly

guido-bruzadin, 2019-09-06:
* Lakka-S912.arm-nightly-2019-09-02.img (dtb: gxm_q200_3g.dtb)
* Beelink GT1 Ultimate (QCA9377)

Both Bluetooth and Wifi (2.4 and 5Ghz) doesn't work.
The 2.4GHz wifi network works for a few minutes and starts to fail.
I tried everything that is in the docs but unsuccessfully.

I saw a lot of issues with QCA9377 firmware and was thinking to recompile Lakka changing firmware files for this chipset as describe in some articles but I haven't enough knowledge to this yet.

There is another issues related to this? 

Here is some logs and extra info.
When system starts `dmesg | grep -i bluetooth` shows nothing.
After execution of:
```shell
touch /storage/.cache/services/bluez.conf
systemctl enable bluetooth
systemctl start bluetooth
```
Results are:
`dmesg | grep -i bluetooth`
```shell
[  543.285231@3] Bluetooth: Core ver 2.18
[  543.285440@3] Bluetooth: HCI device and connection manager initialized
[  543.285470@3] Bluetooth: HCI socket layer initialized
[  543.285481@3] Bluetooth: L2CAP socket layer initialized
[  543.285515@3] Bluetooth: SCO socket layer initialized
[  543.316381@3] Bluetooth: BNEP (Ethernet Emulation) ver 1.3
[  543.316384@3] Bluetooth: BNEP filters: protocol multicast
[  543.316414@3] Bluetooth: BNEP socket layer initialized
````
But still no controllers founded, checked with `bluetoothctl`:
```shell
[bluetooth]# devices
No default controller available
```
And there isn't any `lspci` command:
```shell
Lakka:~ # lspci
-sh: lspci: not found
````

Ntemis, 2019-09-10:
Doesn't make sense to even bother with this issue atm. Wait until we switch to mainline kernel and mesa.

guido-bruzadin, 2019-09-11:
> Doesn't make sense to even bother with this issue atm. Wait until we switch to mainline kernel and mesa.

Thank you @Ntemis. Do You have some expectative about this to happen? How can I follow up these changes?

Ntemis, 2019-09-11:
Kepp an eye on our le-9.2 branch

guido-bruzadin, 2019-09-11:
> Kepp an eye on our le-9.2 branch

Thank you very much.

closed 2019-09-11 by guido-bruzadin