libretro/Lakka-LibreELEC #785: RPi2 Build error freadahead.c

tonilsz, 2019-09-14:
Hi,

First at all lets get in context, I'm trying to make a lakka theme based on "Retrorama", but I wanted to change the loading/splash screen. I have tryied to find "splash.png" in both partitions and nothing was found, so finally, I have found that the only way is rebuilding lakka because these files are allocated in a read only partition:
https://forums.libretro.com/t/can-not-create-ah-custom-design/5643

I have a clean Debian Virtual machine, and I follow this steps:
https://www.lakka.tv/doc/compiling-lakka/

since:
DISTRO=Lakka PROJECT=RPi2 ARCH=arm make image

And when building I get this error:
...
/home/deb/Lakka-LibreELEC-2.3/build.Lakka-RPi2.arm-2.3-devel/toolchain/bin/host-gcc  -I. -I/home/deb/Lakka-LibreELEC-2.3/build.Lakka-RPi2.arm-2.3-devel/m4-1.4.17/lib     -O2 -Wall -pipe -I/home/deb/Lakka-LibreELEC-2.3/build.Lakka-RPi2.arm-2.3-devel/toolchain/include -Wno-format-security -MT freadahead.o -MD -MP -MF $depbase.Tpo -c -o freadahead.o /home/deb/Lakka-LibreELEC-2.3/build.Lakka-RPi2.arm-2.3-devel/m4-1.4.17/lib/freadahead.c &&\
mv -f $depbase.Tpo $depbase.Po
/home/deb/Lakka-LibreELEC-2.3/build.Lakka-RPi2.arm-2.3-devel/m4-1.4.17/lib/freadahead.c: In function 'freadahead':
/home/deb/Lakka-LibreELEC-2.3/build.Lakka-RPi2.arm-2.3-devel/m4-1.4.17/lib/freadahead.c:91:3: error: #error "Please port gnulib freadahead.c to your platform! Look at the definition of fflush, fread, ungetc on your system, then report this to bug-gnulib."
  #error "Please port gnulib freadahead.c to your platform! Look at the definition of fflush, fread, ungetc on your system, then report this to bug-gnulib."
   ^~~~~
/home/deb/Lakka-LibreELEC-2.3/build.Lakka-RPi2.arm-2.3-devel/m4-1.4.17/lib/freadahead.c:93:1: warning: control reaches end of non-void function [-Wreturn-type]
 }
 ^
make[4]: *** [Makefile:1837: freadahead.o] Error 1
make[4]: Leaving directory '/home/deb/Lakka-LibreELEC-2.3/build.Lakka-RPi2.arm-2.3-devel/m4-1.4.17/.i686-linux-gnu/lib'
make[3]: *** [Makefile:1602: all] Error 2
make[3]: Leaving directory '/home/deb/Lakka-LibreELEC-2.3/build.Lakka-RPi2.arm-2.3-devel/m4-1.4.17/.i686-linux-gnu/lib'
make[2]: *** [Makefile:1506: all-recursive] Error 1
make[2]: Leaving directory '/home/deb/Lakka-LibreELEC-2.3/build.Lakka-RPi2.arm-2.3-devel/m4-1.4.17/.i686-linux-gnu'
make[1]: *** [Makefile:1461: all] Error 2
make[1]: Leaving directory '/home/deb/Lakka-LibreELEC-2.3/build.Lakka-RPi2.arm-2.3-devel/m4-1.4.17/.i686-linux-gnu'
make: *** [Makefile:29: image] Error 2
...

In this link I have found that it can be something related with glibc version, but I don't know how to solve it and I'm stucked:
https://forum.openwrt.org/t/solved-build-from-master-on-archlinux-gives-error-for-freadahead-c/18693

I tryed with master and 2.3 versions

Well, I report this bug FYI, but if you know a easyest way to change splash I will be very pleased.

Thanks in advance

ToKe79, 2019-09-14:
Hi,
you can try the fix from #747 on top of your changes. Assuming that you did not fork the repo and cloned from libretro/Lakka-LibreELEC:
```
$ git fetch origin pull/747/head:pr747
$ git merge pr747
```
If you forked and cloned the fork, just add upstream and change `origin` to `upstream` in the above `git fetch` (do following before you run the above commands):
```
$ git remote add upstream https://github.com/libretro/Lakka-LibreELEC
```
After the `git merge` your default editor will open - here just save & exit.
Then just remove your build folder (`rm -rf build.Lakka-RPi2.arm-2.3-devel`) and start the build again. If the issue is related to the glibc version, it should fix it.

Another way is to unsquash the SYSTEM (the read only root file system in squashfs), make your changes, and squash. But I never did this, so I can't give you any instructions, but some folks did it already, maybe they can give you a hint.

natinusala, 2019-09-14:
Unsquashing and re-squashing works (early days of lakka-switch were rough).

@tonilsz which Linux version is your build PC on? Is it Ubuntu 18.10? If so, I recommend trying Ubuntu 18.04 LTS instead

tonilsz, 2019-09-14:
ok it was solved this first error, but I'mhaving more errors, at the moment all are dependency ones, except this one: 

make[3]: Entering directory '/home/deb/Lakka-LibreELEC/build.Lakka-RPi2.arm-2.3-devel/curl-7.58.0/.i686-linux-gnu/src'
  CC       curl-slist_wc.o
  CC       curl-tool_binmode.o
  CC       curl-tool_bname.o
  CC       curl-tool_cb_dbg.o
  CC       curl-tool_cb_hdr.o
  CC       curl-tool_cb_prg.o
  CC       curl-tool_cb_rea.o
  CC       curl-tool_cb_see.o
  CC       curl-tool_cb_wrt.o
  CC       curl-tool_cfgable.o
  CC       curl-tool_convert.o
  CC       curl-tool_dirhie.o
  CC       curl-tool_doswin.o
  CC       curl-tool_easysrc.o
  CC       curl-tool_formparse.o
  CC       curl-tool_getparam.o
  CC       curl-tool_getpass.o
  CC       curl-tool_help.o
  CC       curl-tool_helpers.o
  CC       curl-tool_homedir.o
  CC       curl-tool_hugehelp.o
  CC       curl-tool_libinfo.o
  CC       curl-tool_main.o
  CC       curl-tool_metalink.o
  CC       curl-tool_msgs.o
  CC       curl-tool_operate.o
  CC       curl-tool_operhlp.o
  CC       curl-tool_panykey.o
  CC       curl-tool_paramhlp.o
  CC       curl-tool_parsecfg.o
  CC       curl-tool_strdup.o
  CC       curl-tool_setopt.o
  CC       curl-tool_sleep.o
  CC       curl-tool_urlglob.o
  CC       curl-tool_util.o
  CC       curl-tool_vms.o
  CC       curl-tool_writeout.o
  CC       curl-tool_xattr.o
  CC       ../lib/curl-strtoofft.o
  CC       ../lib/curl-nonblock.o
  CC       ../lib/curl-warnless.o
  CCLD     curl
/usr/bin/ld: skipping incompatible /home/deb/Lakka-LibreELEC/build.Lakka-RPi2.arm-2.3-devel/toolchain/lib/libssl.so when searching for -lssl
/usr/bin/ld: skipping incompatible /home/deb/Lakka-LibreELEC/build.Lakka-RPi2.arm-2.3-devel/toolchain/lib/libssl.a when searching for -lssl
/usr/bin/ld: skipping incompatible /home/deb/Lakka-LibreELEC/build.Lakka-RPi2.arm-2.3-devel/toolchain/lib/libcrypto.so when searching for -lcrypto
/usr/bin/ld: skipping incompatible /home/deb/Lakka-LibreELEC/build.Lakka-RPi2.arm-2.3-devel/toolchain/lib/libcrypto.a when searching for -lcrypto
/usr/bin/ld: ../lib/.libs/libcurl.so: undefined reference to `sk_pop_free'
/usr/bin/ld: ../lib/.libs/libcurl.so: undefined reference to `OPENSSL_add_all_algorithms_noconf'
/usr/bin/ld: ../lib/.libs/libcurl.so: undefined reference to `sk_value'
/usr/bin/ld: ../lib/.libs/libcurl.so: undefined reference to `SSL_load_error_strings'
/usr/bin/ld: ../lib/.libs/libcurl.so: undefined reference to `CONF_modules_free'
/usr/bin/ld: ../lib/.libs/libcurl.so: undefined reference to `SSLv3_client_method'
/usr/bin/ld: ../lib/.libs/libcurl.so: undefined reference to `ERR_free_strings'
/usr/bin/ld: ../lib/.libs/libcurl.so: undefined reference to `EVP_cleanup'
/usr/bin/ld: ../lib/.libs/libcurl.so: undefined reference to `SSLv23_client_method'
/usr/bin/ld: ../lib/.libs/libcurl.so: undefined reference to `sk_num'
/usr/bin/ld: ../lib/.libs/libcurl.so: undefined reference to `SSL_library_init'
/usr/bin/ld: ../lib/.libs/libcurl.so: undefined reference to `sk_pop'
/usr/bin/ld: ../lib/.libs/libcurl.so: undefined reference to `SSL_COMP_free_compression_methods'
/usr/bin/ld: ../lib/.libs/libcurl.so: undefined reference to `SSLeay'
collect2: error: ld returned 1 exit status
make[3]: *** [Makefile:826: curl] Error 1
make[3]: Leaving directory '/home/deb/Lakka-LibreELEC/build.Lakka-RPi2.arm-2.3-devel/curl-7.58.0/.i686-linux-gnu/src'
make[2]: *** [Makefile:1818: all-recursive] Error 1
make[2]: Leaving directory '/home/deb/Lakka-LibreELEC/build.Lakka-RPi2.arm-2.3-devel/curl-7.58.0/.i686-linux-gnu/src'
make[1]: *** [Makefile:921: all-recursive] Error 1
make[1]: Leaving directory '/home/deb/Lakka-LibreELEC/build.Lakka-RPi2.arm-2.3-devel/curl-7.58.0/.i686-linux-gnu'
make: *** [Makefile:29: image] Error 2

I'm using debian10 4.19, downloaded few days ago. I can download ubuntu to build if there are less dependency problems. I will start downloading

ToKe79, 2019-09-14:
you can find the dependencies in the `Dockerfile`, but the build system should install the required ones.

I can suggest applying following patch:
```patch
diff --git a/packages/web/curl/package.mk b/packages/web/curl/package.mk
index c5ff9b1574..2f13c352d9 100644
--- a/packages/web/curl/package.mk
+++ b/packages/web/curl/package.mk
@@ -31,6 +31,7 @@ PKG_LICENSE="MIT"
 PKG_SITE="http://curl.haxx.se"
 PKG_URL="http://curl.haxx.se/download/$PKG_NAME-$PKG_VERSION.tar.bz2"
 PKG_DEPENDS_TARGET="toolchain zlib openssl rtmpdump"
+PKG_DEPENDS_HOST="toolchain zlib:host openssl:host"
 PKG_SECTION="web"
 PKG_SHORTDESC="curl: Client and library for (HTTP, HTTPS, FTP, ...) transfers"
 PKG_LONGDESC="Curl is a client to get documents/files from or send documents to a server, using any of the supported protocols (HTTP, HTTPS, FTP, FTPS, GOPHER, DICT, TELNET, LDAP or FILE). The command is designed to work without user interaction or any kind of interactivity."
@@ -94,6 +95,59 @@ PKG_CONFIGURE_OPTS_TARGET="ac_cv_lib_rtmp_RTMP_Init=yes \
                            --with-librtmp=$SYSROOT_PREFIX/usr \
                            --without-libidn"
 
+PKG_CONFIGURE_OPTS_HOST="ac_cv_lib_rtmp_RTMP_Init=yes \
+                         ac_cv_header_librtmp_rtmp_h=yes \
+                         --disable-debug \
+                         --enable-optimize \
+                         --enable-warnings \
+                         --disable-curldebug \
+                         --disable-ares \
+                         --enable-largefile \
+                         --enable-http \
+                         --enable-ftp \
+                         --enable-file \
+                         --disable-ldap \
+                         --disable-ldaps \
+                         --enable-rtsp \
+                         --enable-proxy \
+                         --disable-dict \
+                         --disable-telnet \
+                         --disable-tftp \
+                         --disable-pop3 \
+                         --disable-imap \
+                         --disable-smb \
+                         --disable-smtp \
+                         --disable-gopher \
+                         --disable-manual \
+                         --enable-libgcc \
+                         --enable-ipv6 \
+                         --enable-versioned-symbols \
+                         --enable-nonblocking \
+                         --enable-threaded-resolver \
+                         --enable-verbose \
+                         --disable-sspi \
+                         --enable-crypto-auth \
+                         --enable-cookies \
+                         --enable-symbol-hiding \
+                         --disable-soname-bump \
+                         --with-gnu-ld \
+                         --without-krb4 \
+                         --without-spnego \
+                         --without-gssapi \
+                         --with-zlib \
+                         --without-egd-socket \
+                         --enable-thread \
+                         --with-random=/dev/urandom \
+                         --without-gnutls \
+                         --with-ssl \
+                         --without-polarssl \
+                         --without-nss \
+                         --without-ca-path \
+                         --without-libpsl \
+                         --without-libmetalink \
+                         --without-libssh2 \
+                         --without-libidn"
+
 pre_configure_target() {
 # link against librt because of undefined reference to 'clock_gettime'
   export LIBS="-lrt -lm -lrtmp"
```

tonilsz, 2019-09-14:
Sorry but patching is out of my linux abilityes, have I to do something like that? 

curl -X PATCH -v -u root: -H "Accept: application/json" -H "Content-Type: application/json" -d @3674tcpatch http.//localhost:8080/objectserver/restapi/alerts/status/kf/3674%3ANCOMS

ToKe79, 2019-09-15:
just create a new file in the repository root (e.g. `curl.patch`), copy the above contents into the file, save & exit, apply the patch on top of the repository:
```bash
$ cat curl.patch | patch -d . -p1
patching file packages/web/curl/package.mk
$
```

tonilsz, 2019-09-15:
Ok, thanks I will try later, now I have moved to 18.04 LTS as suggested @natinusala and it seems to work, just some missing dependencies like zlib.

I just got "recipe for target 'image' failed" error but it seems to be a VirtualBox bug:
https://forums.virtualbox.org/viewtopic.php?f=7&t=91814

I will try and update ;)

Thanks

ToKe79, 2019-09-16:
I think you need to `sudo apt-get install u-boot-tools`.

natinusala, 2019-09-16:
I'm not sure why you would need the uboot tools for the RPi2 as it's not
using uboot?

What is the exact error you're having?

tonilsz, 2019-09-16:
OMG! it was u-boot-tools @Toke79, you are a real guru ;),  and I think that was the last step who crashed because now I have my custom build :)

I have to go work and I will try later, but the build problem is away

Thanks to both @ToKe79 @natinusala, we can close the issue, if you want I can reanude the debian VM to solve the other bugs, but I think that is an envirointment setup issue and it's better to ovoid lost time in that case.

natinusala, 2019-09-16:
GitHub strips the code blocks in the notification email, that's good to know because I didn't see your logs at first

Glad that it's sorted out! We should probably add uboot tools as a host dependency distribution-wide, I think we already have a package for it

tonilsz, 2019-09-16:
Confirmed, I just lunched the img and it worked,

Thanks again

closed 2019-09-17 by natinusala