libretro/Lakka-LibreELEC #787: [Feature Request] LAKKA Access point menu item

duduke, 2019-09-15:
Can this https://www.lakka.tv/doc/lakka-as-accesspoint/ be made into a menu item?
This is especially useful for devices without a keyboard (RetroFlag GPi for instance). This will allow an easy setup for adhoc local netplay.

Thanks.

ToKe79, 2019-09-18:
Hi, I am testing one patch, which should provide menu for enabling/disabling the localap.service via RetroArch front-end. Subscribe to #792 to see the progress. If it builds OK, you can then test it out, so I have some feedback, as I have no device I could test it on. If the test(s) will be positive, I will request a merge in RetroArch master.

ToKe79, 2019-09-19:
[These devel images based on Lakka 2.3](http://nightly.builds.lakka.tv/PRs/pr_792_000087/) should have the patch applied - you should be able to switch the localap.service on/off from the RetroArch menu (should be located where settings for other services - SSH / Samba / Bluetooth - reside).
Thank you for testing!

duduke, 2019-09-19:
Thanks, Will test later today.

duduke, 2019-09-19:
@ToKe79, I’ve done a quick test. 
Without connecting to any AP, I’ve enabled the LocalAP and it works (I’ve seen from the commit that the password is RetroArch). 
I was able to connect via SSH, but haven’t tested Netplay yet. 
I have noticed that I only got an IPv6 address, so I don’t know how that will affect netplay. 

Will update further later today. 

Thanks.

natinusala, 2019-09-19:
An ipv4 would be nice, especially seeing that it's local only so who cares

ToKe79, 2019-09-19:
Thanks, at least we know I did not break Retroarch 😁
Looking forward to hearing from you soon.

duduke, 2019-09-19:
A couple more notes:
1. For some reason, it doesn’t work on bootup. I need to go an toggle off/on and then the AP goes up.
I was the one who contributed to this in LAKKA initially. The way it was structured was in order for it to run at boot a kernel cmdline would need to be added. 
Maybe keeping this as it is - only working adhoc after boot is best, but maybe a disclaimer needs to be added that will alert users that this is not enabled on boot. 
2. If it’s on, WIFI scanning hangs. Again, maybe a disclaimer stating that. 

I’m now testing netplay, will report back.

ToKe79, 2019-09-19:
Thanks for your valuable feedback!

Regarding point nr. 1:
The localap.service is not enabled by default. To enable it at boot, `systemctl enable localap.service` is needed (or, as you mentioned, adding `localap` to cmdline), this is currently not done by retroarch (only `systemctl start|stop localap.service` + creating / removing the localap.conf file in /storage/.cache/services).

I think, to conserve the battery power, it is better not to start the tethering at boot time (in case of GPICase). If someone wants that at boot time, probably it is a not-portable device, it can be enabled (either cmdline or systemctl enable).

Regarding point nr. 2:
we could probably switch off tethering before scanning.

Have you tried it other way round, i.e. connect to a wifi network and then starting the local wlan ap?

duduke, 2019-09-20:
Trying to turn on LocalAP after a WIFI connection has been established does not work. 
Messing with the command line, I think the service should first use connmanctl to disconnect from the current AP before turning on the LocalAP. 
I have tried the following:
“connmanctl disconnect wifi_myuniqueserviceidentifier & connmanctl tether wifi on TEST RetroArch” it does work, but I requires the command to state the exact service identifier. I’m trying to find a workaround that will disconnect all WIFI before setting up the LocalAP. 

With regards to netplay, I wasn’t able to connect from another machine (a Windows laptop). I suspect this has to do with the fact that when I connect to the LocalAP both negotiate a link-local IPv6 address. And although SSH/SMB work, I’m not sure RetroArch netplay supports IPv6 (I’ve tried both lakka.local and the IPV6 address). Once the LocalAP works reliably via the on/off switch I’ll figure out how to get IPv4 addresses working.

natinusala, 2019-09-20:
@ToKe79 you can use `runloop_msg_queue_push` in RA to show a fancy notification, we can have one when you enable the AP to say "Current Wi-Fi connection will be stopped" (or anything more english).

Similarly, I think you can alter the Wi-Fi settings menu to "block" it when the AP is running, and show a notification saying that "Wi-Fi is currently in AP mode"

duduke, 2019-09-20:
I think that:
/etc/connman/main.conf:
PersistentTetheringMode = true should be changed to false.

Also:
Manually testing connmanctl tether off and on from ssh leads me to believe that something is wrong with the service or the menu toggle. as running tether on from ssh disconnects the AP and sets up the localap with ipv4, turning it off resumes the previously connected AP.

I'll dig further into this...

ToKe79, 2019-09-20:
@duduke thanks for your input and testing and reporting back

the toggle is based on the existence of the file `/storage/.cache/services/localap.conf` - if the file exists, the toggle (iirc read from `settings->bools.localap`) is set to true. this works correctly for other services like ssh/samba/bluetooth, as the units are set up to be enabled by default and the start is conditional on the existence of those .conf files (no file = do not start).

however localap.service unit is not enabled by default, so even when localap.conf exists at the bootup, it is not started, but RetroArch/my patch finds the file and sets the bool to true (even when the AP is not running). this means we cannot rely on arbitrary files to control the toggle. we need to really check (via connmanctl) what is the current state and set the toggle accordingly. also we have to disable the AP (if it is running) as soon as we want to connect to a wifi network (and vice versa - disconnect from a wlan network when we are starting up the AP mode). Also we should not delete the localap.conf file when switching the service off, as it contains envs, which the user might change and want to have set up different way than my arbitrary values. We should then only document that by default the access point name is ... and password is ... and can be changed by editing localap.conf.

I looked into wifi/drivers/connmanctl.c in RetroArch sources and I think we can get it work somehow.

then we have the human factor - someone can start the AP via RetroArch, but then disable it via command line... how do we update the toggle? I guess the bools.localap is set only at startup and then when manually toggling the menu option. or we ignore the changes the user might do on their system (leave the toggle at the last state we have set when switching AP on/off and/or the state we found out during startup), but when going to scan for networks we will not rely on the bools.localap value, but we will really check the status of wifi (is in tether mode -> switch tether off). But the same is for the remaining services - one can switch samba on in RetroArch, then switch it of via command line (systemctl stop smbd), but the toggle will stay "on".

if you guys have any pointers/advice - leave comments here. I will appreciate any kind of help.

@duduke: some question however - as I have s905 box where the wifi driver does not work - I have to use you as my remote hands. if you do not connect to any wifi after booting up and add localap to cmdline, does the AP start as wanted - e.g. does it provide ipv4? or it is vital to first connect to an ipv4 wlan, then disconnect and start the AP? also can you test
```bash
# connmanct disconnect wifi_xxx && systemctl start localap.service
```
of course, replace `wifi_xxx` with the service name and also check that you have the required `/storage/.cache/services/localap.conf` with all required envs.

duduke, 2019-09-20:
Ok, I think I got it. And it's kinda simple:
What seems to happen, is due to the way the localap.service is configured, it runs the ExecStart directive and then immediately ExecStop directive. This happens because it is a Type=Simple systemd service which will automatically run ExecStop after the process configured in ExecStart returns.
I've created a test service that includes a **RemainAfterExit=yes** directive. and now the following happens:
When the service is started, the RPi disconnects from my Home AP and sets up a LocalAP as 192.168.1.1 with DHCP.
When the service is manually stopped, LocalAP goes down and RPi reconnects to my HomeAP - exactly desired behaviour.

I believe localap.service should therefore look like this:
**[Unit]
Description=Local AP via Connman
ConditionKernelCommandLine=|localap
ConditionPathExists=|/storage/.cache/services/localap.conf

[Service]
EnvironmentFile=-/storage/.cache/services/localap.conf
#ExecStart=/usr/bin/sshd -D $SSH_ARGS
ExecStart=/usr/bin/connmanctl tether wifi on $APNAME $PASSWORD
ExecStop=/usr/bin/connmanctl tether wifi off
TimeoutStopSec=1s
StartLimitInterval=0
RemainAfterExit=yes

[Install]
WantedBy=multi-user.target

**

ToKe79, 2019-09-20:
nice.
and what about doing scan / connect to AP from RetroArch during localap running? does it still hang? do we have to do some changes there?

ToKe79, 2019-09-20:
the state of the toggle at RetroArch startup could be determined by the existence of `/storage/.config/system.d/multi-user.target.wants/localap.service` (= service is ENABLED) AND `/storage/.cache/services/localap.conf` (configuration exists).

ToKe79, 2019-09-21:
@duduke I pushed some changes to #792 but I just realized that I forgot to change the wifi AP toggle after switching off the AP when going to wifi scanner - I am not sure if the bool has to be updated manually, or it is updated whenever the menu is changed.
New images will be ready soon at http://nightly.builds.lakka.tv/PRs/pr_792_000095/
Please, re-test and let me know about the results. Thank you!

ToKe79, 2019-09-22:
New images, which should also set `bools.localap_enable` to `false` after going to connect to Wi-Fi network: http://nightly.builds.lakka.tv/PRs/pr_792_000096/
also in the previous build (included also in this) are some messages added, like "Starting Wi-Fi Access Point", "Switching off Wi-Fi Access Point"... it will also show a message when no localap.conf file is found with the arbitrary APNAME and PASSWORD.

ToKe79, 2019-09-22:
the very latest build fails (segfault), I guess I call driver_wifi_tether_status() too early... I will check with RA devs

duduke, 2019-09-22:
I will retest the initial image from scratch just to make sure that all the messing around with connmanctl hasn’t borked anything.

ToKe79, 2019-09-22:
I just pushed new changes - http://nightly.builds.lakka.tv/PRs/pr_792_000097/

ToKe79, 2019-09-22:
sorry, had to interrupt the job, new job started - http://nightly.builds.lakka.tv/PRs/pr_792_000098/

duduke, 2019-11-11:
@ToKe79 see my comments in https://github.com/libretro/Lakka-LibreELEC/pull/792#issuecomment-547294817 I think this can be merged.

Thanks.

ToKe79, 2019-11-11:
The changes are directly in RetroArch, so no need to merge the patch. But I will make separate PR / modify the original one to implement the changes outside of RetroArch, like
- `connman: disable persistent mode for tethering` https://github.com/libretro/Lakka-LibreELEC/pull/792/commits/5dbea277373eb323d44649bb73f628625d46243c
- `localap.service: do not stop after start` https://github.com/libretro/Lakka-LibreELEC/pull/792/commits/028e2152dd50a82e8d3569456b2b1726e1d07f09

ToKe79, 2019-11-11:
And of course - thank you for testing it out.

closed 2020-08-18 by Ntemis