libretro/Lakka-LibreELEC #789: [FEATURE REQUEST] Add support for the Argon One case power button / safe shutdown script

jholloway, 2019-09-17:
The Argon One case from Argon40 has a built-in power button with multiple features that allows for a safe shutdown or reboot. The Argon40 team has made a script available for this, located here: https://download.argon40.com/argon1.sh

I know Lakka already has support for some other cases and their included power buttons and scripts, but I don't believe this one is included. I'd love to see it. I don't yet have my case because they have a shipping backlog, but when I get it I would be happy to help test.

natinusala, 2019-09-17:
It looks like the script installs some dependencies then a python daemon, it's very standard

I think we should provide a generic way for users to use their own shutdown scripts instead of making 74 different images with one for each case

mlnlbrt, 2019-09-20:
There is also a fan speed script. 

@natinusala Basically, I see two design paths here:

1.
* keep one image for each platform (RPi, RPi2, S905x, ...)
* include all possible python libraries for GPIO in all images, maybe more dependencies would be needed to be included by default
* maintain a how-to page with instructions for each case (GPICase, Argon One case, ...); where to put scripts, how to modify autostart.sh and so on
* make users responsible for manually adding case support, using how-to pages, after installing Lakka; this can result in a lot of bug reports, questions, help-me threads...

2.
* add a new BOARD directory for each new case, GPICase is a good example. As far as I see, cases are not generic; they are tighly tied to a particular SBC
* have fine-tuned, plug-and-play images for each SBC-case combination at the cost of +300M of server space for each new supported case...

Maybe there is some other way. :) There are pros and cons for each of them.

natinusala, 2019-09-20:
Well adding a new BOARD for each case isn't maintanable, it adds 2 hours and 400mb to the release cycle

mlnlbrt, 2019-09-20:
BOARD directory is pretty lightweight, just some device-tree files and option tweaks. Probably they won't be modified anymore, so I don't see unmaintainable. Increased build time of a complete release and storage requirements are the only cons here.

natinusala, 2019-09-20:
Well, it does require us to maintain an entire version just for three python scripts and a background service 😅

mlnlbrt, 2019-09-20:
So get rid of squashfs and become RetroPie :)

jholloway, 2019-09-21:
My Argon One case arrived today and I have it installed. If there's any viability for offering these scripts, let me know if I can help test.

mlnlbrt, 2019-09-21:
@natinusala I've looked over the scripts for Argon One case again and you are right, no need for a separate BOARD. That's how I see it:
* put the python scripts and systemd units in a new package / packages
* package remaining dependencies
* modify systemd services from the vendor to include `ConditionKernelCommandLine=`

This way we will have support for Argon One case built in RPI3 project, which could be activated by adding some switch to kernel cmd line. What do you think?

natinusala, 2019-09-21:
That sounds great! 

For other cases where dtbos are needed, we can have the systemd service copy the dtbo to the bootloader partition, alter config.txt and reboot.

So that you add the kernel param, you boot, it copies the files and reboots, and you're all set

ilovelinux, 2019-12-22:
Follow these instruction to install Argon One script on Lakka:
- **1a:** Download [precompiled python libs](https://github.com/libretro/Lakka-LibreELEC/files/3992413/precompiled_libs.zip) of RPi.GPIO and smbus and extract them in /storage/.lib (these are built for RPI4)

**OR**

- **1b:** Build it by yourself downloading [RPi.GPIO](https://pypi.org/project/RPi.GPIO/#files) and [smbus](https://pypi.org/project/smbus/#files) sources, extract them and run `python setup.py build` on the main folder of each two. Then you'll find your precompiled python lib in `build/local-something/` directory. It's important to build the precompiled libs in same RPi that will have to run Lakka. You can easly do it installing a full-featured distro like Raspbian or similar.

- **2:** Use [this script](https://github.com/libretro/Lakka-LibreELEC/files/3992412/argonone-setup-libreelec.zip) to install (it's like the the [official script for LibreELEC](https://www.argon40.com/LEARN/post/assembly-line-how-to-s/argon-one-installation-guide-for-libreelec/) but with different paths for precompiled python libs)

That's all!

Slybo, 2020-02-15:
Thank you ilovelinux. 
How should I go about getting the **.lib** folder and **argonone-setup-libreelec.sh** file copied into /storage?
I was able to do it with chmod and got the fan to work but I think I borked something because now I can't connect my Bluetooth controller.

Mazza2590, 2020-04-18:
> Follow these instruction to install Argon One script on Lakka:
> 
> 1. Download the [precompiled Python libs of RPi.GPIO and smbus](https://github.com/libretro/Lakka-LibreELEC/files/3992413/precompiled_libs.zip) and extract them in /storage/.lib (these are built for RPI4)
>    **OR**
>    build them yourself by downloading [RPi.GPIO](https://pypi.org/project/RPi.GPIO/#files) and [smbus](https://pypi.org/project/smbus/#files) sources, extracting them, and running `python setup.py build` in each of the two folders. If you did it correctly you'll find your precompiled Python lib in `build/local-something/` directory. It's important to build the precompiled libs in the same RPi that will run Lakka. You can easily do it by installing a full-featured distro like Raspbian or similar.
> 2. Use [this script](https://github.com/libretro/Lakka-LibreELEC/files/3992412/argonone-setup-libreelec.zip) to install Argon daemons (it's a copy of the [official script for LibreELEC](https://www.argon40.com/LEARN/post/assembly-line-how-to-s/argon-one-installation-guide-for-libreelec/) with different paths of the precompiled Python libs)
> 
> That's all!

Lakka doesnt seem to have all the necessary Python packages to install this. I followed the steps and ran the script, but the service isn't fully operating properly. I then tried installing smbus manually with pip, which i was able to get working eventually - but i can't get RPi.GPIO working at all. 

Any thoughts?

ilovelinux, 2020-04-18:
Sorry for the late answer @Slybo. The script shouldn't affect bluetooth in any way. What board are you using?

RenHoekNL, 2020-06-03:
I rebuild the functionality of the Python script as a standalone executable. No need to install Python or any other programs anymore. Feel free to give it a try:

https://github.com/RenHoekNL/argonONE

Slybo, 2020-06-04:
Thank you very much. I’m currently lending my Lakka Pi to a coworker. I’ll give the new executable a try when I get it back.

Ray2607, 2020-07-17:
> I rebuild the functionality of the Python script as a standalone executable. No need to install Python or any other programs anymore. Feel free to give it a try:
> 
> https://github.com/RenHoekNL/argonONE

Hello, 
Thank you very much for this.
I want to use it for Lakka but i get a Error.

"Lakka:~/.config # systemctl enable argonONE
Failed to enable unit: File argonONE.service: No such file or directory
Lakka:~/.config # systemctl start argonONE
Failed to start argonONE.service: Unit argonONE.service not found."

The Powerbutton works but only for start, shut down and long press shut down. Reboot dont works. 
I dont know if the fan works, because the Pi is not so warm.

RenHoekNL, 2020-07-17:
> "Lakka:~ /.config # systemctl enable argonONE
> Failed to enable unit: File argonONE.service: No such file or directory

The systemd script didn't go to the right directory.. See the Readme on github and start from
`cat << 'EOF' > /storage/.config/system.d/argonONE.service`

> The Powerbutton works but only for start, shut down and long press shut down. Reboot dont works.
Reboot from the menu. Or just shut down and click again to power on.

(The original script says you can reboot with a double click and shutdown by pressing the button for 3 seconds. This is not true. The long press will start a hardware controlled shutdown regardless from the state your system is in. To ensure a clean shutdown, I repurposed the double click to a shutdown since I can actually detect that and act on it without a timer starting to run in hardware.)

> I dont know if the fan works, because the Pi is not so warm.
It should :) I'll probably add a commandline option to set the fan speed so you can test it.

Ray2607, 2020-07-18:
> Reboot from the menu. Or just shut down and click again to power on.

(The original script says you can reboot with a double click and shutdown by pressing the button for 3 seconds. This is not true. The long press will start a hardware controlled shutdown regardless from the state your system is in. To ensure a clean shutdown, I repurposed the double click to a shutdown since I can actually detect that and act on it without a timer starting to run in hardware.)

Reboot from the Lakka menu works, only the double click on the Button dont works.

The 3 Sec. shutdown and 5 sec. Shutdown works

Ray2607, 2020-07-18:
I tested now your Update. I dont geht an Error now. 
But If i double click the Button the raspi shutdown same like i press it 3 Sec.

RenHoekNL, 2020-07-19:
> 
> The 3 Sec. shutdown and 5 sec. Shutdown works

This is the same thing. The 3 second shutdown is the same as the 5 second shutdown. It's a hardware controlled forced poweroff, is UNSAFE and can lead to corrupted filesystems.
 
> I tested now your Update. I dont geht an Error now.
> But If i double click the Button the raspi shutdown same like i press it 3 Sec.

Yes, that is what I changed.. Instead of a reboot it does a SAFE shutdown. So it waits until the Pi4 indicates it's ready to power off and then does so, not before.

For reboots, use the Lakka menu.

closed 2020-08-18 by Ntemis