libretro/Lakka-LibreELEC #791: S905W Wifi (Broadcom BCM4334) not working

wakawaka99, 2019-09-18:
Using nightly build --> Lakka-S905.arm-nightly-2019-09-17.img
Using hardware H96 mini  --> S905W 2gb/16gb Wifi chipset BCM4334

Scan for wifi showing no AP even though router is less than 3 feet away.
Booting android box into default OS, wifi works as usual.

most roms run fine on that nightly build

wakawaka99, 2019-09-19:
Using stable build -->-S905.arm-2.3.img

same issue

Ntemis, 2019-09-20:
When the port of s905 started s905w didn't exist. The focus was only given to those boards who had Gbit Ethernet and only.New inferior X/W revisions were left to the individual owner to self supporting his/her device. 
In short we wont break others to fix one board.
Best bet is to wait until 9.2 branch is ready.

closed 2019-09-20 by Ntemis

ToKe79, 2019-09-21:
can you test this build?
http://nightly.builds.lakka.tv/PRs/pr_800_000093/S905.arm/
it is based on latest release + some wifi drivers