libretro/Lakka-LibreELEC #793: When using the Ozone menu driver, navigating into any submenu causes RetroArch to crash

jholloway, 2019-09-19:
Lakka version 2.3
Raspberry Pi 4 Model B, 1GB RAM

On a fresh install of Lakka on a RPi4, I changed to the Ozone menu driver in the RetroArch settings, as I prefer it over XMB. I restarted RetroArch to apply the changes. The Ozone menu loads and I can navigate through the top level menu. But when I attempt to go into any deeper level of sub-menu, a graphical glitch happens and then RetroArch restarts.

I can't successfully navigate into any sub-menu level while the Ozone menu driver is active. This means I can't even access the Drivers menu under Settings to change it back. But when using SSH to manually edit the configuration file back to the XMB menu driver and restarting the RetroArch service, RetroArch starts into the XMB menu as expected.

As I said, this is a fresh install that is fairly stock. I only made a few minor changes before attempting the menu driver change. I turned on SSH, Samba, and Bluetooth. I set the system resolution to 1920x1080. That's all I remember changing.

I have attempted to verify this issue by testing RetroArch 1.7.8.4 installed on a fresh Raspbian Buster Lite image, and this issue with the Ozone menu driver does not happen there.

I captured a log of me attempting to navigate into a submenu while the Ozone menu driver was active. It is attached: [lakka-rpi4-ozone-bug-log.txt](https://github.com/libretro/Lakka-LibreELEC/files/3629429/lakka-rpi4-ozone-bug-log.txt)

The final line before core dump seems most relevant:

`retroarch: ../src/broadcom/cle/v3d_packet_helpers.h:74: __gen_uint: Assertion `v <= max' failed.`

Please let me know if there's any additional info I can provide or any troubleshooting steps I can perform to help track down the problem.

natinusala, 2019-09-19:
Ideally I'd need a crash report but I think the issue comes from scissoring. I don't know why the bug isn't here in Raspbian though, are you using the new OpenGL driver?

Mikau28, 2019-11-10:
Can confirm the same issue on 4gb model
https://github.com/libretro/Lakka-LibreELEC/issues/841

Tim-Townsend, 2019-12-24:
I am unable to change back to XMB even after SSH 
Lakka:~ #  nano /storage/.config/retroarch/retroarch.cfg

Then I changed menu_driver to "xmb", still have the issue

sourcerebels, 2020-04-25:
You can restore "xmb" menu driver by accessing the sd card directly from your computer and editing retroarch.cfg. Via ssh I wasn't able to change it.

Falumir, 2020-07-28:
I've been attempting to look into this recently. Unfortunately, a recent stack trace shows that it appears to be an issue with a driver rather than Ozone. The stack trace can be found here: https://pastebin.com/vf8yjrqG

Through trial and error, I've determined that this happens when the sidebar is pushed offscreen. On a hunch that it might have something to do with a negative X coordinate being passed to the driver, I added some logging code to ozone_sidebar.c to try and determine which draw function leads to the driver crashing. The last draw function leading up to the crash was ozone_draw_text at -339.267181, so it's likely not an issue with the X coordinate.

`[INFO] ozone_draw_text - -339.267181
[INFO] ozone_draw_icon - -339.267181
[INFO] ozone_draw_text - -339.267181
retroarch: ../src/broadcom/cle/v3d_packet_helpers.h:74: __gen_uint: Assertion `v <= max' failed.`

I've also determined that a crash occurs when backing out of the quick menu. This is likely related, though I haven't pulled a stack trace yet.

tewald, 2020-08-30:
Using ozone/retroarch compiled on raspbian from source works as expected. When I changed to Lakka the bug reported by OP happens.
Raspberry PI 4 4GB here.

HunterAhlquist, 2020-09-20:
Having the same issue.

gouchi, 2020-09-20:
@Tim-Townsend @sourcerebels You did try like this [comment](https://github.com/libretro/Lakka-LibreELEC/issues/841#issuecomment-568917662) for editing the configuration ?

@tewald How did you compile it ?  On Lakka we compile it with those specific parameters for RPi4 [Source](https://github.com/libretro/Lakka-LibreELEC/blob/master/packages/libretro/retroarch/package.mk#L97)

`--enable-opengles --enable-opengles3 --enable-opengl_core --enable-kms --disable-x11 --disable-videocore --enable-plain_drm`

@Falumir @jholloway You may try to open an issue on [Mesa issue tracker](https://docs.mesa3d.org/bugs.html) to get some feedback as it looks like it is related to the driver ? Similar issue related to [Intel Vulkan driver (anv) with RCPS3](https://gitlab.freedesktop.org/mesa/mesa/-/issues/3351) but it was fixed in RCPS3 at the end.

Philps, 2020-11-12:
New Install LibreElec 9.2.6 with gamestarterAddon on Raspberry Pi4 B 4GB model

RetroArch starts fine and i can navigate up and down, but as soon as i hit enter to get into a submenu it crashes and reloads kodi

Log exceprt
LibreELEC:/ # tail -100 ./storage/.kodi/temp/retroarch.log
RetroArch [ADDON] :: Thu Nov 12 19:20:28 UTC 2020
============================================================
RetroArch [ADDON] :: Total RetroPlayer cores: 11
RetroArch [ADDON] :: Kodi is ready.
RetroArch [ADDON] :: RPi4.arm-9.2.6 - Thu Nov 12 19:20:36 UTC 2020
RetroArch [ADDON] :: Stop kodi...
RetroArch [ADDON] :: Launch Retroarch (Enable debugging and see process output in retroarch_debug.log)
RetroArch [ADDON] :: Restart kodi...

tail ./storage/.kodi/temp/retroarch_debug.log
ALSA lib /home/jails/home/jenkins/LE/build2/workspace/RPi4/LibreELEC.tv/build.LibreELEC-RPi4.arm-9.2.6/alsa-lib-1.1.9/src/seq/seq_hw.c:466:(snd_seq_hw_open) open /dev/snd/seq failed: No such file or directory
[ERROR] [MIDI]: snd_seq_open failed with error -2.
[ERROR] [MIDI]: Initialization failed (list of input devices unavailable).
[INFO] SRAM will not be saved.
[INFO] Loading history file: [/storage/.config/retroarch/content_history.lpl].
[INFO] Loading history file: [/storage/.config/retroarch/content_music_history.lpl].
[INFO] Loading history file: [/storage/.config/retroarch/content_video_history.lpl].
[INFO] Loading history file: [/storage/.config/retroarch/content_image_history.lpl].
[INFO] Loading favorites file: [/storage/.config/retroarch/content_favorites.lpl].
[INFO] [GL]: VSync => on
[INFO] [GL]: VSync => on
[INFO] [KMS]: New FB: 1920x1080 (stride: 7680).
[INFO] [KMS]: New FB: 1920x1080 (stride: 7680).
game.retroarch-RPi: ../src/broadcom/cle/v3d_packet_helpers.h:74: __gen_uint: Assertion `v <= max' failed.

Can upload full log if required.

As a workaround change  menu_driver to "xmb" via ssh

LibreELEC:~ # egrep menu_driver /storage/.config/retroarch/retroarch.cfg
menu_driver = "xmb"

This will load the old UI which works (still some graphical flaws but basic functionality )

Then I changed menu_driver to "xmb",

ToKe79, 2021-06-10:
is this still present in the [latest release](https://le.builds.lakka.tv) / [nightly](https://nightly.builds.lakka.tv/latest/)? if so, please open new issue.

closed 2021-06-10 by ToKe79