libretro/Lakka-LibreELEC #795: [Question] Check emulators running

tonilsz, 2019-09-19:
Hi,

I have a nespi+ case and I use safe shutdown and reset buttons from here:
https://github.com/edus44/lakka-gpio-control

It works perfectly, but I wanted a little update like here:
https://github.com/crcerror/retroflag-picase/blob/master/SafeShutdown.py

what I want is when reset button is pressed, check if we are running any emulator or kodi is open, for this we can inspire in this piece of code:
```
def reboot():
 output = int(subprocess.check_output(['/opt/RetroFlag/multi_switch.sh', '--es-pid']))
 output_rc = int(subprocess.check_output(['/opt/RetroFlag/multi_switch.sh', '--rc-pid']))
 if output_rc:
     os.system("/opt/RetroFlag/multi_switch.sh --closeemu")
 elif output:
     os.system("/opt/RetroFlag/multi_switch.sh --es-restart")
 else:
     os.system("sudo reboot")
```
I know that we don't have ES but I hope that check for some process can be possible

this is the code/pseudocode that I should develop:
```
def reset():
  oldState = True
  while True:
    GPIO.wait_for_edge(resetPin, GPIO.FALLING)

    state = GPIO.input(resetPin)
    if state != oldState and state == False:

      ## state_kodi =  funtionToCheckIfKodiIsRunning()
      ## state_cores =  funtionToCheckIfAnyCoreIsRunning()
      ## if state_kodi:
      ##         os.system("kodi-send --action="Quit"")
      ## elif state_cores:
      ##         FuntionLikeXMBCloseContent
      ## else:
      ##         os.system("pkill retroarch")
      ## time.sleep(1)
```

I hope that it isn't very confused, there is no problem if I have to develop a multi_switch.sh to handle better the events, but first I have to know if its possible and how to find references.

Thanks

natinusala, 2019-09-19:
I actually made a "sleep" thing for Raspberry Pi + ATXRaspi card some time ago

When you pressed the power button when a game was running, it did a savestate before a regular shutdown. On next boot the core and game were reloaded and the savestate was loaded

I think you need to mod RetroArch for what you want to do, to just create and remove a file on content load / close

tonilsz, 2019-09-19:
So I will move to retroarch to, thanks

closed 2019-09-19 by tonilsz