libretro/Lakka-LibreELEC #796: Show menu from emulator

tonilsz, 2019-09-19:
Hi, 

this is related with user experience, I noticed that when have a core open and display the menu (press L3+R3)  if you navigate f.e. change some settings and then go back. Since here all is fine, but if you then want to close the core you (press L3+R3) again, then the xmb is pointing the las place you where ( in this case settings) you have to find runnig core and rom to finish, in some cases there are about 30 cores and handreds of games and in some cases with the same name(different versions).

Nothing is wrong and maybe I'm missing why you take this decision but I think that the lower level user will be greatfull.

Thanks again.

natinusala, 2019-09-19:
This is an issue with RetroArch more than Lakka, can you file an issue there instead?

tonilsz, 2019-09-19:
of course, sorry,

here is it wright?
https://github.com/libretro/RetroArch

closed 2019-09-19 by tonilsz

natinusala, 2019-09-19:
Yes thanks !

ToKe79, 2019-09-19:
You do not have to navigate to Playlist -> ROM.

When using XMB (I am not familiar with other menu-drivers, but should be similar, if not same) and when an emulation is running, in the first / most left menu column (called Lakka in Lakka, RetroArch in standalone RetroArch I think) the first menu item should be the "Quick Menu" (I think it is called), where you have all the same items as in Playlist -> ROM file. So you can close/reload/etc. from there too.

I think staying where you were in the menu between L3+R3 pushes (switching between the emulation and RetroAch) is intended by design.