libretro/Lakka-LibreELEC #802: Lakka - Composite Output to CRT is broken on Pi 3

neosquid, 2019-09-24:
Hello,

Installing the latest LAKKA release refuses to boot to the user interface. SDTV_MODE has been correctly selected in the .config, as I have usually done.

Apparently it is working with an HDMI TV, just not CRT. Others have reported the same problem on the main Lakka forum:

https://forums.libretro.com/t/lakka-composite-output-to-crt-is-broken-on-pi-3/24229

natinusala, 2019-09-24:
Yes, this is a known kernel issue, the next version will have a patch

neosquid, 2019-09-24:
Great!

I hope the patch won't prevent the firmware people are using with Lakka to obtain 240p output from working.

https://filthypants.blogspot.com/2017/03/raspberry-pi-240p-composite-output.html?m=1

MisterFloppy, 2019-09-25:
I've found a "work-around" for v2.3.1 :slight_smile:

I could solve this issue by following the hints regarding updating the firmware files manually given at [https://forum.libreelec.tv/thread/11921 ... #post84908](https://forum.libreelec.tv/thread/11921-raspberry-pi-firmware-update/?postID=84908#post84908) :

I downloaded bootcode.bin, start_x.elf and fixup_x.dat from the RPi firmware github repository at [https://github.com/raspberrypi/firmware ... aster/boot](https://github.com/raspberrypi/firmware/tree/master/boot), renamed start_x.elf to start.elf and fixup_x.dat to fixup.dat and copied the 3 files to the /flash partition - problem solved!

natinusala, 2019-09-25:
I was unsure as to what changed because the `firmware` repo contains a lot of stuff, including stuff that's not related to the firmware at all

For future readers, if you want to use this temporary fix on a RPi4 you need to use the `4x` versions of the files (`fixup4x`, `start4x`, etc)

neosquid, 2019-12-03:
Just a quick note that PI 240p firmware is not working - tested it today. Can boot to the flower, but some unintelligible error message appears at the top of the screen, squashed by my CRT.

gitwinst, 2020-01-25:
YES. This works for the RPi4, composite output to CRT is a go. Thanks @MisterFloppy for the workaround, and also @natinusala for the 4x file tip.

BTW, this was necessary even though I was updated to the latest Lakka 2.3.2 release.

Hiki1, 2020-01-25:
So the composite output is broken in Lakka 2.3.2 on Raspberry Pi 3 . It gets stuck on the boot screen.
Fortunately, the workaround mentioned in this thread helps.

ominari, 2020-04-06:
@MisterFloppy I know this is an old thread but I just found it.  When I try to overwrite the files in the flash folder in WinSCP I get the following error.  scp: /flash/bootcode.bin: Read-only file system.  I know it is a permission issue.  Could you assist me with this please?

Thanks!

natethekvd, 2020-06-15:
thank you so much for this solution ive been looking everywhere

neosquid, 2020-06-16:
Closing issue - I downloaded a beta build and issues with CRT Composite seem to be resolved.

closed 2020-06-16 by neosquid