libretro/Lakka-LibreELEC #803: XU4 - Freezes after a few mins on lakka 2.3, segmentation fault

ttmcmurry, 2019-09-24:
- Which version of Lakka are you using?  **2.3**
- What system hardware are you using?  **XU4**
- What did you do?  **Normal Use**
- What did you expect to happen?  **Operate Normally**
- What happened instead?  **Freezes after a few minutes of use**

Per my post in the lakka forums [here](https://forums.libretro.com/t/lakka-2-3-retroarch-1-7-8-on-xu4-freezing/24248)...

When running Lakka 2.3 for a few minutes, the storage available at /storage stops working and  begins to segfault.  In turn, the image on-screen also freezes.  It's still possible to SSH into Lakka.  

The only data in log.txt (from [here](https://www.lakka.tv/doc/troubleshooting-lakka/), Option B) is the following sentence:

"Segmentation fault"

This doesn't seem to happen with RetroArch 1.7.8 (Not Lakka) on Winx64 or Android 8 or Android 9, only on the XU4.

ToKe79, 2019-09-24:
can you post your nas mount configuration? ofc remove  any credentials.
thanks

ToKe79, 2019-09-24:
also please test if the issue is there without the nas.

ttmcmurry, 2019-09-24:
Sure, the .mount file is below.  The issue was not present in the 9/17 nightly XU3 build (either the 1st or 2nd 9/17 builds), but is present in the final 2.3 build.   I'll reimage it again with 2.3 later tonight and put some data in /storage/roms to see if it continues.

    [Unit]
    Description=cifs mount script
    Requires=network-online.service
    After=network-online.service
    Before=retroarch.service
    
    [Mount]
    What=//192.168.1.50/Emulation
    Where=/storage/roms/nas
    Options=username=retroarch,password=xxxxx,rw
    Type=cifs
    
    [Install]
    WantedBy=multi-user.target

ttmcmurry, 2019-09-25:
Just FYI, the specific with the image I have been using the entire time is Lakka-OdroidXU3.arm-2.3.1.img.gz

I reloaded again from this build via Etcher.  The absolute only setting in Retroarch I modified was to turn on SSH.  I used default directories for everything.  Data was placed in /storage/roms.  BIOS was placed in /tmp/system.  

I swapped in out of some Genesis / SegaCD titles, and left SonicCD playing for about an hour in both game and CD Player mode.  Not a single problem was encountered.  I made save states, loaded save states, reset games, loaded states back.  Created a core default control remap, all saved properly.  Not a single Segfault.  

I will note my storage backend is a Synology DS916+ running DSM 6.2.2 Update 3, and it has remained both online (no reboots) and no updates for the duration of this testing.  I use other SMB shares on the NAS throughout the day in Windows and have zero issues with it on either WiFi or Wired.  Everything seems both kosher and 'usual' on the NAS end.  

Last detail is the XU3 is connected via wired gigabit to an HP 1810 series switch, which is uplinked to an HP 1820 series switch, and both are operating normally.  Lakka shares connections with 7 other wired devices (PS4, Xbox, TV, Roku, Receiver, all entertainment devices) and none of them exhibit issues with network connectivity, or internet.  Aside from the all gigabit home network, my internet uplink is also full duplex gigabit, so network performance is not an issue anywhere.  

I've tried to eliminate as much potential change (there's virtually none) and external factors, including prior Lakka nightly version 9/17 which did work correctly, but I don't have confidence the issue lies in my infrastructure here.

ttmcmurry, 2020-01-02:
Update on this.. as of pr_838_000118, ODroid XU4 continues to freeze.  Typically this occurs when closing a ROM.  I can SSH into the XU4 and issue commands but they tend to do nothing.  The screen is still frozen wherever it was when selecting "close rom" from the quick menu.

If I SSH into the XU4 after lakka freezes, and issue a reboot or shutdown command, it does not execute.  I am forced to power cycle the device.

There are no new PR releases for this device since then (2019-12-03).

closed 2020-08-18 by Ntemis