libretro/Lakka-LibreELEC #804: (RPi3) Dynamic background issues

tonilsz, 2019-09-24:
Hi!

I'm still working in custom theme for lakka and I've experienced some issues, I have about 30+ systems in the menu (for try) and I have set a custom background and changed background directory. Since here all is fine backgrounds display as expected.

The first problem comes on launching, when start the first image displayed is the one set as custom background instread of the image added as (Main Menu), if you move to configuration and back the image is shown well. So I think that it lefts a check for a Main Menu image in the launcher.

The second is little more weird, I have a lot of system loaded so then I have a lot of images, if I move fast to the from first to one, there a moment that UI crashes, first misses all background images and only shows the icons, and then letters start to crash to, finally cores icons disappear to.
I was thinking about the possible reasons of that and I tried to use a different bound images (to save memory) for this I have seen that the images on https://github.com/natinusala/retroarch-assets/tree/master/wallpapers/bichromatic%20pads/1440x900 because they are very light, but the issue still happen. I don't know if I have to set special configuration or remove animations or if Lakka is leaking memory or if is to much for a RPi3.

It remains me only a few images to finish it and I wanted to publish them on Github, but if it doesn't work I don't know if I should publish.

natinusala, 2019-09-24:
> The second is little more weird, I have a lot of system loaded so then I have a lot of images, if I move fast to the from first to one, there a moment that UI crashes, first misses all background images and only shows the icons, and then letters start to crash to, finally cores icons disappear to.

This is a memory issue, the VRAM of the RPi3 is filled with your images. How large are they?

> The first problem comes on launching, when start the first image displayed is the one set as custom background instread of the image added as (Main Menu), if you move to configuration and back the image is shown well. So I think that it lefts a check for a Main Menu image in the launcher.

This is a bug with RetroArch, could you fill an issue there?

tonilsz, 2019-09-24:
> This is a bug with RetroArch, could you fill an issue there?
Ok I move first to RetroArch ;)

> This is a memory issue, the VRAM of the RPi3 is filled with your images. How large are they?
I have tryied with the images becoming with Lakka, the same included in this repository https://github.com/natinusala/retroarch-assets/tree/master/wallpapers/bichromatic%20pads/1440x900 as you can check they are very light, less than 100k every one. my default background is about 400k. how much VRAM is available?

natinusala, 2019-09-24:
I meant in resolution, are they all 1440x900? How many do you load?

You are responsible of how much VRAM is available by tweaking config.txt

tonilsz, 2019-09-24:
well this ones yes are all 1440x900(mine where 1200x768) I loaded all of them, but it happens when you scroll about 20 images. I suppose that they are loaded on demand. 

is there any possible risk if I twerk the VRAM?

natinusala, 2019-09-24:
> I suppose that they are loaded on demand.
I'm not sure about that seeing the errors that you have

> is there any possible risk if I twerk the VRAM?
The worst that can happen is impossibility to boot, in which case you can just revert the change

tonilsz, 2019-09-24:
ok then I will try to expand VRAM and remove background, to use default (orange one I think) to avoid using memory.

closed 2020-08-18 by Ntemis