libretro/Lakka-LibreELEC #806: Endless core-dumps after rebooting without an output available

Enigmahack, 2019-09-24:
- Which version of Lakka are you using?
    [INFO] Built: Sep 21 2019
    [INFO] Version: 1.7.8
    [INFO] Git: ea7e682253

- What system hardware are you using?
ThinkCentre M700 Tiny
256gb SSD, 500gb HDD, 16gb RAM
[INFO] CPU Model Name: Intel(R) Core(TM) i5-6500T CPU @ 2.50GHz
[INFO] Capabilities: MMX MMXEXT SSE1 SSE2 SSE3 SSSE3 SSE4 SSE4.2 AVX AES 

- What did you do?
Boot the system with no display attached, or rebooting via SSH. 

- What did you expect to happen?
Reboot will happen and Lakka will start and be stable

- What happened instead?
Upon reboot completion, Lakka segfaults, dumps a core, repeats segfault/dump in a loop, until SSD is full, still continues dumping cores of 0 bites. It stops crashing and runs fine when I connect an external display. 

In my current setup, the Mini PC is running in a storage cabinet and is connected to a TV via HDMI switcher. The display isn't always available, but I often manage files remotely via SSH and SFTP so require the system to be running.  

Running retroarch in verbose mode remotely gives the following information:

`Lakka:~ # retroarch -v
    [INFO] RetroArch 1.7.8 (Git ea7e682253)
    [INFO] === Build =======================================
    [INFO] CPU Model Name: Intel(R) Core(TM) i5-6500T CPU @ 2.50GHz
    [INFO] Capabilities: MMX MMXEXT SSE1 SSE2 SSE3 SSSE3 SSE4 SSE4.2 AVX AES 
    [INFO] Built: Sep 21 2019
    [INFO] Version: 1.7.8
    [INFO] Git: ea7e682253
    [INFO] =================================================
    [INFO] Environ SET_PIXEL_FORMAT: RGB565.
    [INFO] Redirecting save file to "/storage/savefiles/.srm".
    [INFO] Redirecting savestate to "/storage/savestates/.state".
    [INFO] Version of libretro API: 1
    [INFO] Compiled against API: 1
    [INFO] [Audio]: Set audio input rate to: 30000.00 Hz.
    [INFO] [Video]: Video @ fullscreen
    [INFO] [Video]: Starting threaded video driver ...
    [INFO] [DRM]: Found 6 connectors.
    [INFO] [DRM]: Connector 0 connected: no
    [INFO] [DRM]: Connector 0 has 0 modes.
    [INFO] [DRM]: Connector 1 connected: no
    [INFO] [DRM]: Connector 1 has 0 modes.
    [INFO] [DRM]: Connector 2 connected: no
    [INFO] [DRM]: Connector 2 has 0 modes.
    [INFO] [DRM]: Connector 3 connected: no
    [INFO] [DRM]: Connector 3 has 0 modes.
    [INFO] [DRM]: Connector 4 connected: no
    [INFO] [DRM]: Connector 4 has 0 modes.
    [INFO] [DRM]: Connector 5 connected: no
    [INFO] [DRM]: Connector 5 has 0 modes.
    [WARN] [DRM]: Couldn't get device connector.
    [WARN] [DRM]: Couldn't get device resources.
    [ERROR] [KMS]: Couldn't find a suitable DRM device.
    [INFO] [GL]: Found GL context: null
    [INFO] [GL]: Detecting screen resolution 320x240.
    [INFO] [GL]: Vendor: (null), Renderer: (null).
    [INFO] [GL]: Version: (null).
    Segmentation fault (core dumped)`

Is there not a way to turn off seg fault logging? I've checked all menus and any logging options are disabled however these are still written when Retroarch crashes.

ToKe79, 2019-09-24:
this is expected behaviour - when no display is connected, retroarch does not have any context to render to and segfaults. this can be only avoided using the startup script where you would check for connected display - see proposal in https://github.com/libretro/Lakka-LibreELEC/commit/97bb6a26ffa8e338c2c32b1248d0f057e6b6cd62#diff-bdeac31aa78ce696ec6ebf909e137405 (in this state it is only adapted for the Generic/PC platform, but can be adapted to other platforms too - one must only know where to check for the status of the display connection). this check can be incorporated into `/storage/.config/autostart.sh`, but you must adapt it a little bit - you do not have to rewrite the systemd units as it is done in the commit, but it will use retroarch-autostart.service unit which is started before retroarch.service. but I am not sure if systemd will wait for proper exit of the script. at the end of the script do not start retroarch (`/usr/bin/retroarch`), just use `exit 0`

Enigmahack, 2019-09-25:
Just out of curiosity, where I'll often leave the machine running and switch to other sources is there a good way to have retroarch start when it detects a monitor? The problem with the above is that I'd have to run to a PC to manually start retroarch or reboot the box when I want to use it instead of leaving it running. 

Maybe there's a clever way to have something assigned to the power button? (Like a short-press would launch the startup script above, a long press would trigger a shutdown?)

Pxtl, 2020-03-01:
Wait, does this bug mean that running Lakka on a machine where you might turn the TV off and leave Lakka running (and admin it remotely, possibly including reboots) is not a supported workflow?  Because I just hit this bug and was very confused as to where several gigs of disk space went - especially since the crash-logs are in a hidden directory so a simple du check doesn't find them.

Enigmahack, 2020-03-01:
This is exactly the issue that I was suggesting should be a supported
workflow. Instead I need to turn off my machine if I change HDMI inputs
because I'll run out of diskspace unless I go back and constantly delete
the log files.

It's a pain.

ToKe79, 2020-03-02:
As long as the display is connected during startup, as soon as the RetroArch frontend appears, you should be able to switch inputs on your display/TV. However, if the device reboots for any reason and the device boots without connected display, RetroArch will not find a proper display device (as HDMI will report as disconnected with no video capabilities) and hence segfault.

Maybe it is possible to change the `kernel.core_pattern` via `sysctl` to avoid the core-dumps - you can try:
```
# sysctl kernel.core_pattern=|/dev/null
```
If it does not work (i.e. read only filesystem), you have to rebuild the system with changes like here, but applied for your device (in this example it was done for Rockchip): https://github.com/ToKe79/Lakka-LibreELEC/commit/c51d46b64abafbc0c66fa1e4c20cda4ef46b60f9

natinusala, 2020-03-02:
Even if it doesn't dump the core it will still fill the disk eventually, just slower

Pxtl, 2020-03-02:
It sounds like a bug in RetroArch more than anything.  I mean, @ToKe79's fix of "wait until HDMI is available before starting RetroArch" is definitely an improvement, but RA going full segfault/coredump because of dead HDMI also seems super-wrong.  But as @natinusala points out, even if RA only logged a single line of "can't start because no HDMI found", it would still fill up a disk (assuming an 80-char line per failure, it would fill about a gig per year).

krodelabestiole, 2020-04-04:
a dirty but working solution to the issue :

add the line `sysctl -w 'kernel.core_pattern=|/bin/false'`
to file `/storage/.config/autostart.sh`
and reboot

you can check the value of `kernel.core_pattern` before and after with this command :
`sysctl kernel.core_pattern`
(default value is `/storage/.cache/cores/core.%E.%t.%p` expected is `|/bin/false`)

remove every old dumps with
`rm /storage/.cache/cores/*`

closed 2020-08-18 by Ntemis