libretro/Lakka-LibreELEC #811: lakka 2.3.1 x86_64   cannot boot live from usb stick

mfrigerio17, 2019-09-28:
Hi,
I cannot get a successful first boot in live mode with the newest Lakka, using a usb stick, on a laptop.

During the first boot (the one it is supposed to resize the storage partition, I think) I get the message:
```
*** Error in mount_storage: mount_common: Could not mount UUID=...
### Starting debug shell... type exit to quit ###
sh: can't access tty: job control turned off
# _
```

Using `Lakka-Generic.x86_64-2.3.1`, md5sum verified, flashed onto the usb stick with `dd` (within Ubuntu Linux). I tried two different USB drives, on two different laptops (an old one, a recent one).
Same error.

Typing `exit` brings me back to the same point.
I can see the flower logo at a good resolution.

natinusala, 2019-09-30:
Hi,

Can you use your Ubuntu computer to see the storage partition UUID? Does it match with the error message?

ToKe79, 2019-09-30:
If the second partition (`LAKKA_DISK`) exists on the thumb drive, you could edit syslinux.cfg in the root of the first (`LAKKA`) partition on the USB thumb drive. change
`
disk=UUID=...
`
to
`
disk=LABEL=LAKKA_DISK
`

It could have happened that after the writing the OS changed the UUID (?) or that the UUID cannot be read? Did you `sync` after `dd`? Did you `gunzip` before `dd`?

Or maybe the download was corrupted and the second partition somehow disappeared.

ZachBacon, 2019-09-30:
I can confirm, Live doesn't seem to boot. But strangely enough, just running the installer, it installed just fine, I only found out the live issues by accident as I was too slow to type in the installer command.

mfrigerio17, 2019-10-05:
I tried once more and it worked. The only thing I changed was to create an "mdos" type of partition table, rather than gpt, on the usb key (with GParted). This however should not have any effect as `dd` rewrites the partition table itself... 

For the record:
@natinusala yes, the UUID in syslinux.cfg matched the two partitions on the usb key; in fact, the UUID in the error message was the one of the system partition (LAKKA), not the data partition (LAKKA_DISK)

@ToKe79 in any case, yes I always `gunzip`ed the `.gz` before `dd`, and `sync`ed after

Thank you

infinite-persistence, 2020-07-17:
I'm facing the same issue. @mfrigerio17, can you explain your solution?  Changing to `msdos` in gparted would wipe whatever that was written by `dd` previously, right?

gouchi, 2020-08-16:
Hi,

Do you still reproduce the issue with [nightly build image](https://nightly.builds.lakka.tv/2020-08-10/Generic.x86_64/Lakka-Generic.x86_64-3.0-devel-20200810061340-46502e1.img.gz) ?

Thank you.

mfrigerio17, 2020-08-18:
> I'm facing the same issue. @mfrigerio17, can you explain your solution? Changing to `msdos` in gparted would wipe whatever that was written by `dd` previously, right?

Yes. Rewriting the partition table would wipe out the disk. I did `dd` again afterwards.
As I said, I don't really know why it worked, as the outcome of the last `dd` should not be affected by what you do before, on the partition table

gouchi, 2020-08-18:
Thank you for the feedback so we can close this issue ?

Also you may consider to use [Etcher](https://www.balena.io/etcher/) to ease the burning process to usb.

infinite-persistence, 2020-08-18:
Yeah, I used Etcher in the beginning. 
The solution that I ended up using was to reformat the secondary space myself using `gparted` instead of letting the 2-reboot process do it.

> Thank you for the feedback so we can close this issue ?

Ok with me.

closed 2020-08-18 by gouchi