libretro/Lakka-LibreELEC #812: No info file for mupen64plus

fuzzysloth, 2019-09-28:
Lakka build: Lakka-OdroidXU3.arm-2.3.1.img.gz
Hardware: ODROID-XU4
Symptoms: Mupen64plus disappeared from the core options for N64, the only option is ParaLLEl N64 now.

I've already read [this comment](https://github.com/libretro/Lakka-LibreELEC/issues/705#issuecomment-511558660) and understand that mupen64plus cannot be dropped in favor of mupen64 next. But many N64 games don't work well with parallel n64 either.

Is it an idea to just recover info file for mupen64plus to make this option available again?

PS: I must admit that I didn't nightly build yet because 2.3.1 was released just few days ago. Please feel free to close the ticket if this is already fixed.

natinusala, 2019-09-30:
I think that the core is still there but the info file is indeed missing.

Are you able to load the core and a game manually?

We don't have control over the info files and sometimes they are removed without us noticing.

fuzzysloth, 2019-09-30:
I know it sounds weird.
According to the logs mupen64plus core loads indeed, but in the UI there are only options for parallel64 and the behavior for one particular rom as buggy as it is normally with parallel64 on XU4.

The work-around was recovering mupen64plus.info in the toplevel directory of unionfs so it appears in /tmp/cores and retroarch sees it. Then you can select that core as a default for N64 and the game works.

natinusala, 2019-09-30:
So the core does work, okay. The file will be back in the next update.

fuzzysloth, 2019-09-30:
Well, I can only confirm that it loads, but the experience during the gameplay is not the same if there is no info file.

natinusala, 2019-09-30:
What? Why would it be different

fuzzysloth, 2019-09-30:
I don't know why, but something that can be a reliable indication is that the core crashes when you just close it. When I use mupen64plus with a proper info file with - it closes normally.
As I said - it sounds weird, but the behavior is different with no info file. I cannot reproduce it right now because I'm away from my XU4 box a.t.m. I can do it later.

natinusala, 2019-09-30:
That definitely sounds like a RetroArch bug to me

closed 2019-10-08 by natinusala