libretro/Lakka-LibreELEC #813: fan-control spamming the system log on XU4

fuzzysloth, 2019-09-28:
Lakka build: Lakka-OdroidXU3.arm-2.3.1.img.gz
Hardware: ODROID-XU4

While following systemd journal I see repeatimg messages:

```
Sep 28 17:08:24 Lakka fan-control.sh[286]: cat: can't open '/sys/devices/10060000.tmu/temp': No such file or directory
Sep 28 17:08:24 Lakka fan-control.sh[286]: sh: 78000: unknown operand
Sep 28 17:08:24 Lakka fan-control.sh[286]: sh: 75000: unknown operand
Sep 28 17:08:24 Lakka fan-control.sh[286]: sh: 70000: unknown operand
Sep 28 17:08:24 Lakka fan-control.sh[286]: sh: 68000: unknown operand
Sep 28 17:08:24 Lakka fan-control.sh[286]: sh: 66000: unknown operand
Sep 28 17:08:24 Lakka fan-control.sh[286]: sh: 63000: unknown operand
Sep 28 17:08:24 Lakka fan-control.sh[286]: sh: 60000: unknown operand
Sep 28 17:08:24 Lakka fan-control.sh[286]: sh: 58000: unknown operand
```

This is very annoying when you try to follow retroarch behavior. Also this script doesn't seem to have any effect on the fan - it keeps turning on and off and making noise.

After another check I discovered that the entries that the script tries to access don't exist.

fuzzysloth, 2019-09-28:
[lakka-sys-devices-search.txt](https://github.com/libretro/Lakka-LibreELEC/files/3665663/lakka-sys-devices-search.txt)
Uploaded a result of `find /sys/devices|grep fan`.

natinusala, 2019-09-30:
It looks like @Ntemis updated the kernel without fixing this script

fuzzysloth, 2019-09-30:
Is it possible to fix the script? That would be a nice improvement.

Ntemis, 2019-09-30:
No need for the script on mainline kernel.
I will sent a pr to remove it

tbocek, 2019-10-06:
@Ntemis had the same issue, created a pull request.

fuzzysloth, 2019-10-06:
Just to be sure: is it only kernel related, not hardware (XU4 vs XU3)?

closed 2019-10-06 by Ntemis