libretro/Lakka-LibreELEC #814: [Feature request] OpenGL support for RPi0 / GPi case?

nicosensei, 2019-10-01:
Hi,

I've installed and tested the 2.3.1 build for the GPi case, and it's quite good for 2D games. However performance with PCSX core is very bad, while the same unit can run PS1 games very decently on RetroPie based distros. 

From what I understood, this is because they set up and activate the OpenGL driver for Raspbian.

Is such a driver available for OpenElec? Can we hope to have good PS1 emulation on Pi0 / GPi case?

natinusala, 2019-10-01:
Hey,

PCSX ReARMed doesn't use OpenGL, so the difference is not there

natinusala, 2019-10-01:
Oh maybe

ToKe79, 2019-10-01:
@nicosensei can you put [this](http://nightly.builds.lakka.tv/members/vudiq/rpi0-1_pcsx_rearmed/pcsx_rearmed_libretro.so) version of the core into `/storage/cores` and try again?

nicosensei, 2019-10-02:
Thank you, and sorry for my lack of knowledge 😅

I will try this core and report back.

nicosensei, 2019-10-02:
Hi, I put the .so file in the cores folder on the LAKKA_DISK partition of the SD card. Is this correct? Will it override the default core?

If so, unfortunately the choppiness is still there.

mlnlbrt, 2019-10-02:
Actually, unai is the most suited renderer for devices like RPi0 and I bet RetroPie based distros use the same.

@nicosensei please try this core, place it in `cores` directory on the LAKKA_DISK partition:
http://buildserver3.lakka.tv/pcsx_rearmed_libretro.so

what PS1 games did you try so far? For me Kula World, Worms Armageddon work fine, even Gran Turismo is playable. However, I always disable XA Audio Decoding and CD Audio in core options.

I didn't try standalone pcsx_rearmed core in RetroPie, but I feel it may perform better performance-wise vs libretro version... :(

nicosensei, 2019-10-07:
Hi, I tried the core, but unfortunately it's still too slow to be playable. I activated the unai renderer in the core options. I tested Crash Bandicoot, Blood Omen and Final Fantasy Tactics.

Thanks for the effort!

The standalone core in RetroPie works indeed much faster ... But I still prefer Lakka anytime!

natinusala, 2019-10-07:
If we can figure out why I would love to have it fixed in Lakka

nicosensei, 2019-10-07:
> @nicosensei <https://github.com/nicosensei> do you have ps1 bios in system
> directory?

Yes I have, scph1001.bin precisely.

gouchi, 2019-10-10:
For information, [Retropie libretro-pcsx-rearmed recipe](https://github.com/RetroPie/RetroPie-Setup/blob/master/scriptmodules/libretrocores/lr-pcsx-rearmed.sh).

natinusala, 2019-10-11:
It looks like it's using the `peops` renderer

mlnlbrt, 2019-10-11:
Which is slower than `unai` we are using now, according to pcsx_rearmed readme.txt:

> gpu_unai.so    - Unai's plugin from PCSX4ALL project. Faster than P.E.Op.S.
                 but has some glitches.

I guess there are some custom builds of RetroPie, which have pcsx_rearmed with gles renderer enabled. We can't use it, as it seems that libretro port of pcsx_rearmed don't support it.

@nicosensei which RetroPie build are you using?

nicosensei, 2019-10-11:
I tested Supreme GPi.

-- 
-------------------------------------------------------------------------------------------------------------------------------------------
http://nicoslittleworld.blogspot.fr
http://www.youtube.com/user/NicoSensei
http://soundcloud.com/nikojiro
-------------------------------------------------------------------------------------------------------------------------------------------
(\__/)
(O.o)
(> < ) This is Bunny. Copy Bunny into your signature to help him on his way
to world domination!

closed 2020-08-18 by Ntemis