libretro/Lakka-LibreELEC #817: Yaba Sanshiro is broken under Lakka 2.3.1 x86_64 Generic

iokzero, 2019-10-05:
Using Lakka x86_64 Generic on Intel Iris plus 650 HD graphic (Gen 7 CPU)

Trying to run a Saturn game with Yaba Sanshiro, (The game work under other emulator and is known to work with Yaba Sanshiro)

The Core exist with the following error:
[libretro INFO] PG_NORMAL
[libretro INFO] Compile error in vertex shader. 1
[libretro INFO] Shaderlog:
0:1(10): error: GLSL 3.30 is not supported. Supported versions are: 1.10, 1.20, 1.30, 1.00 ES, 3.00 ES, 3.10 ES, and 3.20 ES

Looks like the graphic driver for x86_64 is out dated, for the core.
The core should remove from this platform until the graphic driver got updated to the support version.

natinusala, 2019-10-05:
Uhm sorry but your GPU is too old to run this core

Ntemis, 2019-10-05:
I don't think so:
https://www.intel.com/content/www/us/en/support/products/98912/graphics-drivers/graphics-for-7th-generation-intel-processors/intel-iris-plus-graphics-650.html

natinusala, 2019-10-05:
But if gl3 isn't available... @iokzero can you give me the full RA logs please?

iokzero, 2019-10-05:
`systemctl stop retroarch.service ; retroarch -v
[INFO] RetroArch 1.7.8 (Git ea7e682253)
[INFO] === Build =======================================
[INFO] CPU Model Name: Intel(R) Core(TM) i7-7567U CPU @ 3.50GHz
[INFO] Capabilities: MMX MMXEXT SSE1 SSE2 SSE3 SSSE3 SSE4 SSE4.2 AVX AES 
[INFO] Built: Sep 16 2019
[INFO] Version: 1.7.8
[INFO] Git: ea7e682253
[INFO] =================================================
[INFO] Environ SET_PIXEL_FORMAT: RGB565.
[INFO] Redirecting save file to "/storage/savefiles/.srm".
[INFO] Redirecting savestate to "/storage/savestates/.state".
[INFO] Version of libretro API: 1
[INFO] Compiled against API: 1
[INFO] [Audio]: Set audio input rate to: 30000.00 Hz.
[INFO] [Video]: Video @ fullscreen
[INFO] [Video]: Starting threaded video driver ...
[INFO] [DRM]: Found 3 connectors.
[INFO] [DRM]: Connector 0 connected: yes
[INFO] [DRM]: Connector 0 has 31 modes.
[INFO] [DRM]: Connector 0 assigned to monitor index: #1.
[INFO] [DRM]: Connector 1 connected: no
[INFO] [DRM]: Connector 1 has 0 modes.
[INFO] [DRM]: Connector 2 connected: no
[INFO] [DRM]: Connector 2 has 0 modes.
[INFO] [DRM]: Mode 0: (3840x2160) 3840 x 2160, 60 Hz
[INFO] [DRM]: Mode 1: (4096x2160) 4096 x 2160, 24 Hz
[INFO] [DRM]: Mode 2: (4096x2160) 4096 x 2160, 24 Hz
[INFO] [DRM]: Mode 3: (3840x2160) 3840 x 2160, 30 Hz
[INFO] [DRM]: Mode 4: (3840x2160) 3840 x 2160, 30 Hz
[INFO] [DRM]: Mode 5: (3840x2160) 3840 x 2160, 25 Hz
[INFO] [DRM]: Mode 6: (3840x2160) 3840 x 2160, 24 Hz
[INFO] [DRM]: Mode 7: (3840x2160) 3840 x 2160, 24 Hz
[INFO] [DRM]: Mode 8: (1920x1080) 1920 x 1080, 60 Hz
[INFO] [DRM]: Mode 9: (1920x1080) 1920 x 1080, 60 Hz
[INFO] [DRM]: Mode 10: (1920x1080i) 1920 x 1080, 60 Hz
[INFO] [DRM]: Mode 11: (1920x1080i) 1920 x 1080, 60 Hz
[INFO] [DRM]: Mode 12: (1920x1080) 1920 x 1080, 50 Hz
[INFO] [DRM]: Mode 13: (1920x1080i) 1920 x 1080, 50 Hz
[INFO] [DRM]: Mode 14: (1920x1080) 1920 x 1080, 30 Hz
[INFO] [DRM]: Mode 15: (1920x1080) 1920 x 1080, 30 Hz
[INFO] [DRM]: Mode 16: (1920x1080) 1920 x 1080, 25 Hz
[INFO] [DRM]: Mode 17: (1920x1080) 1920 x 1080, 24 Hz
[INFO] [DRM]: Mode 18: (1920x1080) 1920 x 1080, 24 Hz
[INFO] [DRM]: Mode 19: (1280x1024) 1280 x 1024, 60 Hz
[INFO] [DRM]: Mode 20: (1280x960) 1280 x 960, 60 Hz
[INFO] [DRM]: Mode 21: (1280x720) 1280 x 720, 60 Hz
[INFO] [DRM]: Mode 22: (1280x720) 1280 x 720, 60 Hz
[INFO] [DRM]: Mode 23: (1280x720) 1280 x 720, 50 Hz
[INFO] [DRM]: Mode 24: (1024x768) 1024 x 768, 60 Hz
[INFO] [DRM]: Mode 25: (800x600) 800 x 600, 60 Hz
[INFO] [DRM]: Mode 26: (720x576) 720 x 576, 50 Hz
[INFO] [DRM]: Mode 27: (720x480) 720 x 480, 60 Hz
[INFO] [DRM]: Mode 28: (720x480) 720 x 480, 60 Hz
[INFO] [DRM]: Mode 29: (640x480) 640 x 480, 60 Hz
[INFO] [DRM]: Mode 30: (640x480) 640 x 480, 60 Hz
[INFO] [GL]: Found GL context: kms
[INFO] [GL]: Detecting screen resolution 3840x2160.
[INFO] [EGL] Found EGL_EXT_platform_base, trying eglGetPlatformDisplayEXT
[INFO] [EGL]: EGL version: 1.4
[INFO] [EGL]: Current context: 0x7f84440a4d80.
[INFO] [KMS]: New FB: 3840x2160 (stride: 15360).
[INFO] [GL]: Vendor: Intel Open Source Technology Center, Renderer: Mesa DRI Intel(R) Iris Plus Graphics 650 (Kaby Lake GT3) .
[INFO] [GL]: Version: 3.0 Mesa 17.1.3.
[INFO] [GL]: Using resolution 3840x2160
[INFO] [GL]: Default shader backend found: glsl.
[INFO] [Shader driver]: Using GLSL shader backend.
[INFO] [GLSL]: Checking GLSL shader support ...
[WARN] [GL]: Stock GLSL shaders will be used.
[INFO] [GLSL]: Found GLSL vertex shader.
[... 215 lines cut ...]
[INFO] [GL]: Loaded 1 program(s).
[INFO] [GL]: Initializing HW render (1024 x 1024).
[INFO] [GL]: Max texture size: 16384 px, renderbuffer size: 16384 px.
[INFO] [GL]: Supports FBO (render-to-texture).
[INFO] [Font]: Using font rendering backend: freetype.
[INFO] [Video]: Graphics driver did not initialize an input driver. Attempting to pick a suitable driver.
[WARN] [udev]: Couldn't open any keyboard, mouse or touchpad. Are permissions set correctly for /dev/input/event*?
[INFO] [udev]: Plugged pad: Logitech Gamepad F310 (1133:49693) on port #0.
[INFO] [udev]: Pad #0 (/dev/input/event5) supports force feedback.
[INFO] [udev]: Pad #0 (/dev/input/event5) supports 16 force feedback effects.
[INFO] [Autoconf]: 284 profiles found.
[INFO] [Autoconf]: selected configuration: /tmp/joypads/udev/Logitech_Gamepad_F310.cfg
[INFO] [Joypad]: Found joypad driver: "udev".
[INFO] [Video]: Found display server: null
[INFO] Found shader "/tmp/shaders/bilinear.glslp"
[INFO] Found shader "/tmp/shaders/nearest.glslp"
[INFO] Found shader "/tmp/shaders/slang-conversion-template.glsl"
[INFO] Found shader "/tmp/shaders/stock.glsl"
[libretro INFO] PG_NORMAL
[libretro INFO] Compile error in vertex shader. 1
[libretro INFO] Shaderlog:
0:1(10): error: GLSL 3.30 is not supported. Supported versions are: 1.10, 1.20, 1.30, 1.00 ES, 3.00 ES, 3.10 ES, and 3.20 ES

Aborted (core dumped)
`

iokzero, 2019-10-05:
It might only affected Intel Integrated Graphic. Nvidia / AMD Graphic may works fine, but I don't have such GPU for testing at the moment. 

Thanks for the help!

natinusala, 2019-10-05:
`[INFO] [GL]: Version: 3.0 Mesa 17.1.3.` It looks like the correct OpenGL version is used, but is your GPU compatible with GLSL 3.30? Can you check?

iokzero, 2019-10-05:
@natinusala not sure where I can find this info, but the iGPU itself should support up to OpenGL 4.5

natinusala, 2019-10-05:
Do you have a way to test the core on the same system on a regular Linux distro?

iokzero, 2019-10-06:
@natinusala 

Don't even know what distro have that emulator, hahahah.
But after some digging I am able to find a way to conform my system can run OpenGL 3.30, 4.2, and 4.5. and this version of the OpenGL should match up with GLSL. 

What I did is that I run the game with Krono, and it stated in the Option, saying it require OpenGL 3.3+ and give you options to choose your OpenGL version Namely: OpenGL 4.2 and Open 4.5, each with fix, and additional capability. And I am able to run the game on any of those setting. 

However, It also have an Option say Video Renderer which also needed OpenGL 3.3+. I am about to run the game with that option, I am able to run the game fine. but if I choose OpenGL CS which require OpenGL 4.2+. I got black screen running the game. But that's another bug I guess. need to look into that to see what's wrong.

natinusala, 2019-10-07:
I guess the core GL version detection is faulty then? There's not much we can do on the Lakka scope

Can you try the WIP Lakka 2.9 versions available here http://nightly.builds.lakka.tv/Lakka-LE9.2/latest/

This version has updated gpu drivers

iokzero, 2019-10-08:
I will give this a try this week, along with other bugs I found (on Generic x86_64). 
If you prefer, in case I found new bugs, I will report them labeled as "Lakka-LE9.2".

iokzero, 2019-10-09:
@natinusala 

Tried the image you suggested. same issue. (along with all other issues I found under 2.3.1 are still there) Currently they are being reported to their respected projects. expect for this one, because I thought it was a lakka problem, but it seems now it is the core itself. I will report the bug as soon as I can. 

With all that said, I wouldn't be able to do further testing for a little bit because I accidentally over write by back up drive with the Lakka image (LOL). so I need to fix that first. Glad that's not the only copy for whatever it is backing up. But now, there's only one copy leave, so got to get that fix. sorry for the off topic.

iokzero, 2020-08-13:
FYI

This problem still exist under the latest night build 08102020

iokzero, 2020-08-15:
I posted this issue on the Yabause / Yaba Sanshiro issues github page, 

@barudreadmon from their team said : "Support for this core has been dropped, use standalone"

https://github.com/libretro/yabause/issues/223

closed 2020-08-18 by Ntemis