libretro/Lakka-LibreELEC #818: Lakka nightly 2019-09-21 has the last working /boot/config.txt

kaiomatico, 2019-10-05:
Hi, I wanted to try out the latest nightly builds for the Raspberry Pi 4 and found out that the latest 2 images are not booting. The folder structure and everything else seems fine on the latest images, but the encoding for the /boot/config.txt seems to be wrong because I cannot open the files with gedit and the flashed micro SD cards are not booting in the Raspberry Pi

Cheers,
- Kai

natinusala, 2019-10-05:
Can you check the integrity of the file you downloaded?

kaiomatico, 2019-10-05:
I just checked, the sha256 sums are fine for the .img.gz downloads. The 2019-10-05 says: Error in mount_storage: mount common: Could not moint LABEL=LAKKA_DISK***

I extracted the .gz and then sudo dd if=Lakka-RPi2.RPi4.arm-nightly-2019-10-05.img of=/dev/sdb bs=1M && sudo sync

The same procedure worked fine for the 2019-09-21 image on the same sd card. I also safely eject

kaiomatico, 2019-10-05:
Huh, sorry, my SD card ready is probably faulty. I just noticed that even the 2.3.1 image had a "corrupted" config.txt on my SD card, but the img mounted as loop device was fine this time. Sorry :+1:

closed 2019-10-05 by kaiomatico