libretro/Lakka-LibreELEC #822: RetroArch crashes when clicking Settings -> Wi-Fi …

fooness, 2019-10-07:
`- Which version of Lakka are you using?`
Lakka-RPi2.RPi4.arm-2.3.1.img.gz

`- What system hardware are you using?`
Raspberry Pi 3B v1.2

`- What did you do?`
Install Lakka, click `Settings` -> `Wi-Fi` …

`- What did you expect to happen?`
Waiting for wireless networks to be found …

`- What happened instead?`
RetroArch crashes and reboots (only RetroArch, not the Lakka, I guess) …

```
Error wifi: Already enabled
retroarch: ../src/broadcom/cle/v3d_packet_helpers.h:74: __gen_uint: Assertion `v <= max' failed.
Aborted (core dumped)
```

[log.txt](https://github.com/libretro/Lakka-LibreELEC/files/3697096/log.txt)

fooness, 2019-10-07:
PS: It seems to work with `Lakka-RPi2.arm-2.3.1.img.gz`, so maybe the RPi2.RPi4 image is not compatible with Raspberry Pi 3?

natinusala, 2019-10-07:
Of course the RPi2.RPi4 image isn't compatible with the Raspberry Pi 3

fooness, 2019-10-07:
Thanks. The naming is a little bit confusing.

closed 2019-10-07 by fooness

natinusala, 2019-10-07:
Yes I agree that the name is confusing, however the website should link to the correct image

fooness, 2019-10-07:
This is correct, yes. Thank you for clarification.

Would it be possible and … maybe smart to rename the images to `Lakka-RPi2.RPi3.arm-2.3.1.img.gz` and `Lakka-RPi4.arm-2.3.1.img.gz`?

natinusala, 2019-10-07:
It's the naming scheme of the build system, `PROJECT.BOARD` and the RPi4 is a board (sub-project) of the RPi2 project.

Lakka 3.0 will have a better naming scheme