libretro/Lakka-LibreELEC #826: PSX (ps1) game format issues Generic PC 2.3.1

Gitlab1111, 2019-10-11:
Currently using  version 2.3.1 Stable for Generic PC 
CPU: Intel Haswell Intel© Core™ i5-4670 CPU @ 3.40GHz × 4
GPU: N/A
RAM: 4gb
SSD 60gb

Im not sure if I can chalk this down to bad roms or user error but for certain formats of **PSX** roms fail to play even when using other **PS1 cores**. I mostly use **PS HW** and sometimes The PSX Rearmed core.
 Another note is that most of my **CCD** formats do not like to be added to the playlist and need to be launched via **Load Content submenu**. I only have one rom which has 2 files which is 1 **mdf** and 1 **mds**. It launches but doesn't add itself to the playlist when using directory scan.

  Examples of Roms and its content:

Folder A has  4 files: ccd, cue, sub and a .img.ecm    These rom fails to play with either psx cores

Folder B has 2 files: 1 cue file and 1 bin.ecm file.   This rom also fails to play

Folder C has 3 files: 1 ccd, 1 sub and 1 img.ecm file.  rom fails to play 

I suspect roms with the .ecm added to it are the culprit because I have a rom in a ccd format that works fine.  It's contents are 4 files which are 1 **ccd**, 1 **cue**, 1 **sub** and 1 .**img** without the **ecm** at the end of it. I noticed the trend of roms with .ecm not working 

Another issue I have been having is that **PSx** cores have been freezing upon closing content. Documentation should mention if .ecm files don't work

ToKe79, 2019-10-11:
iirc for psx you must include the serial in square brackets (e. g. `[SLUS-12345]`) in the file name and use .cue + .bin to be added to a playlist by the scanner. for manual loading check info about the respective core, what it supports.

Gitlab1111, 2019-10-23:
I found that most roms that worked well are in the  bin/cue combo that are usually split in 2 files in a game rom folder. I will try to get around to this .ecm issue and test more roms when I have the chance. 
 @ToKe79 You sent me a link on resolving logs and I think I came up with a configuration that works better on trying to ensure the data is logged and saved.

I had to enable Advanced options and set logging to level 0 (Debug) and level 0 (Debug for core) and Performance core logging. Then after to help ensure logs are saved I use save configuration. The process does require me to reboot PC and open up the log via Linux OS as root.  Question is it preferable that I save the logs as .log or .txt?

Hopefully I can get back to the community with some logs on this matter.

closed 2020-08-18 by Ntemis