libretro/Lakka-LibreELEC #827: PS1 (PSX) eboot rom for PSP not working

Gitlab1111, 2019-10-11:
Currently using version 2.3.1 Stable for Generic PC
CPU: Intel Haswell Intel© Core™ i5-4670 CPU @ 3.40GHz × 4
GPU: N/A
RAM: 4gb
SSD 60gb

Is anything extra needed or any bios for running these ps1 eboot's because so far I couldn't get it to work. Keeping things organized I made a folder for it  under "ps1 eboot"  The rom is labeled  EBOOT.PBP.  The rom also came with a image of the game as well.

Storage/roms/ps1 eboot/Soul of the Samurai [NTSC-U] [SLUS-00822]/SLUS00822

natinusala, 2019-10-14:
I'm not sure if PPSSPP supports the PSX emulator, can you check with the PC version first?

Gitlab1111, 2019-10-21:
I would if I had space to run it on my Linux Mint PC. I did however got those PSP ps1 eboot roms to run on the PSX ReArmed core. I tested it out with the other PSX core HW Beetle but no luck yet.

natinusala, 2019-10-21:
I just checked and PPSSPP doesn't support the PS1 emulator (mainly because it's actually not an emulator)

closed 2019-10-21 by natinusala