libretro/Lakka-LibreELEC #835: Quicknes core does not appear on Raspberry Pi

thezerobit, 2019-11-02:
I'm using the latest 2.3.1 Lakka for Raspberry Pi (RPi.arm).

The only NES cores that appear available in the interface are Fceumm and Nestopia, but the git repository shows that bnes and quicknes should be available, too. 

Quicknes would be especially nice to have since it might actually be fast enough to emulate the NES at playable speeds on the original RPi. Fceumm and Nestopia are too slow, so there's currently no NES cores on Lakka that are playable on Raspberry Pi original models.

Unfortunately, there's not wireless component to the old RPis, and my TV isn't close to my router, so I don't have a way to ssh in for logs.

thezerobit, 2019-11-03:
There is no quicknes source package here: https://sources.lakka.tv/

thezerobit, 2019-11-03:
I downloaded and built the entire Lakka source, and I noticed it did not produce a quicknes .so file. When I try to build quicknes:

    $ DISTRO=Lakka PROJECT=RPi ARCH=arm scripts/build quicknes
      UNPACK   quicknes
      BUILD    quicknes (target)
    make: Entering directory '/home/steve/3rdParty/Lakka-LibreELEC/build.Lakka-RPi.arm-2.3-   devel/quicknes-2cb3bec/libretro'
    make: *** No targets specified and no makefile found.  Stop.
    make: Leaving directory '/home/steve/3rdParty/Lakka-LibreELEC/build.Lakka-RPi.arm-2.3-   devel/quicknes-2cb3bec/libretro'

RobLoach, 2019-11-03:
Thanks a lot for the fix :+1:

closed 2019-11-03 by RobLoach