libretro/Lakka-LibreELEC #837: pkg-config wrong download url

regne, 2019-11-03:
Compiling Lakka for Generic_VK_nvidia x86_64

(I am not sure if this is bug is related to Lakka or LibreELEC. I am sorry about this)

The URL for downloading pkg-config in sources/pkg-config/pkg-config-0.29.1.tar.gz.url points to http://pkg-config.freedesktop.org/releases/pkg-config-0.29.1.tar.gz which doesn't seem to be anymore the freedesktop project website. I am puzzled about this.

You can download the file from github at https://github.com/freedesktop/pkg-config/archive/pkg-config-0.29.1.zip, but its sha256sum does not match the original file.

I downloaded the package from http://mirror.lakka.tv/

natinusala, 2019-11-04:
This is why we keep archives on mirror.lakka.tv :)

I will update the wrong URL in the package, thanks for letting us know

regne, 2019-11-04:
> I will update the wrong URL in the package, thanks for letting us know

Your welcome.

Thank you for Lakka and RetroArch!

natinusala, 2019-11-05:
The freedesktop host looks back up!

closed 2019-11-05 by natinusala