libretro/Lakka-LibreELEC #839: Lakka-Generic.i386-2.3.1 does not boot on EBOX-3350MX (Pentium-compatible)

Oifan, 2019-11-05:
I have downloaded the latest **Lakka i386** image from 2019-09-20 (http://le.builds.lakka.tv/Generic.i386/Lakka-Generic.i386-2.3.1.img.gz), unpacked, `dd`-ed on a USB-thumb drive and inserted it into my **EBOX-3350MX** Mini-PC (http://www.compactpc.com.tw/product.aspx?act=detail&id=12).
After powering on, the Syslinux showed its prompt, and after entering **installer** (I tried also **live**), I got this error message from Lakka kernel:

`SYSLINUX 6.04 EDD 6.04-pre1 Copyright (C) 1994-2015 H. Peter Anvin et al`
`Wait for live session to start or press <TAB> for more options (installer)`
`boot: installer`
`This kernel requires the following features not present on the CPU:`
`cmov`
`Unable to boot - please use a kernel appropriate for your CPU.`

It looks like only **i586**-compatible kernel can run on this Mini-PC.
I can run many **32-bit non-PAE** Linux distros on it (antiX 12, Puppy Linux 5.5, TinyCore Linux 6.4.1, SliTaz 5.0, X-Linux 5.7) as well as MS-DOS 98 and FreeDOS 1.0.

gouchi, 2019-11-07:
[Vortex86 SoC](https://en.wikipedia.org/wiki/Vortex86) will require specific configuration.

Oifan, 2019-11-08:
I have just found more details about the CPU -- it's **Vortex86MX** @933MHz - also called **Xcore86** --> https://bkhome.org/archive/blog2/201001/xcore86-cpu-in-gecko-edubook.html

gouchi, 2019-11-11:
`cat /proc/cpuinfo`

Also you can try to test [RetroArch](https://pkgs.alpinelinux.org/packages?name=retroarch&branch=edge) as there are [some packages](https://pkgs.alpinelinux.org/packages?name=libretro*&branch=edge&repo=testing) for Alpine Linux. You will need to [enable edge branch](https://wiki.alpinelinux.org/wiki/Edge).

Thank you.

Oifan, 2019-11-19:
I managed to get the `cpuinfo` -- see attachment [ebox-info.zip](https://github.com/libretro/Lakka-LibreELEC/files/3866470/ebox-info.zip), but sadly the graphics performance is poor while running XFCE or Awesome in the latest **Alpine**.
Today I managed to start **Salix Live 14.2.1** - the graphics performance in XFCE is better than Alpine but still not good.
Finally I have started **TinyCore Linux 6.4.1** - the graphics performance in **flwm** is OK.
I will search info how to build&run RetroArch on TinyCore Linux, since this seems the best option to me.

gouchi, 2019-11-30:
Thank you for the information.

I am glad you could fine a distribution which is working for your device.

You can read this [documentation](http://docs.libretro.com/development/retroarch/compilation/linux-and-bsd/) about RA compilation.

closed 2019-11-30 by gouchi