libretro/Lakka-LibreELEC #841: Ozone crashes on opening submenu items

Mikau28, 2019-11-08:
If you are sure that it's a bug in Lakka, please answer the following questions:
- Which version of Lakka are you using?
2.3
- What system hardware are you using?
Pi4
- What did you do?
Chose ozone as the menu driver, rebooted.
- What did you expect to happen?
I could traverse menus and change settings values
- What happened instead?
For example just selecting information tab, the menu will briefly load then crash back to ozone. Every menu that opens up a submenu will force lakka to restart. Purging the cfg will revert this back to xmb as you can’t open Submenus to change the setting back.

natinusala, 2019-11-10:
Can you try ozone on Raspbian with the new mesa driver (opengles3)?

Tim-Townsend, 2019-12-24:
I am experiencing the same, I dont have a way to go back to XMB

natinusala, 2019-12-25:
To go back to XML edit (or delete to reset) the RetroArch config file

Tim-Townsend, 2019-12-25:
I was doing so, but the config save on exit was overwriting the menu_driver in the config with 'ozone' every time I would restart retroarch. Ultimately by updating the config to xmb then causing the ozone menu to crash again worked. It failed to do the save config on exit, retaining the xmb in the config. Thanks for coming back to help though!

natinusala, 2019-12-25:
You need to quit Retroarch first by using `systemctl stop retroarch`, then 
edit your config and start RA again with systemctl (same command but start 
instead of stop)

closed 2020-08-18 by Ntemis