libretro/Lakka-LibreELEC #842: Flycast black screen with sound

Mikau28, 2019-11-09:
If you are sure that it's a bug in Lakka, please answer the following questions:
- Which version of Lakka are you using?
2.3 nightly
- What system hardware are you using?
Generic x86_x64
- What did you do?
Launch flycast
- What did you expect to happen?
The game would display video
- What happened instead?
Sound plays as the game is running but it’s a black screen. Tried different games and different time.

gouchi, 2019-11-10:
Please provide some [logs](https://www.lakka.tv/doc/troubleshooting-lakka/#introduction-to-lakka-logs).

Thank you.

Mikau28, 2019-11-11:
> Please provide some [logs](https://www.lakka.tv/doc/troubleshooting-lakka/#introduction-to-lakka-logs).
> 
> Thank you.

https://paste.ubuntu.com/p/4YRbx4h5W6/

natinusala, 2019-11-11:
What is your GPU? What GL version does it support?

Mikau28, 2019-11-11:
> What is your GPU? What GL version does it support?

Intel HD Graphics 2500
OGL 4.0
Alpha sorting is per triangle.

Mikau28, 2019-11-30:
Bought a Lenovo tiny with a 4th gen i5. Flycast runs flawless, might not be a concern for older hardware?

gouchi, 2019-12-05:
Do you reproduce also the issue with [Lakka from LE9.2](http://nightly.builds.lakka.tv/Lakka-LE9.2/latest/Generic.x86_64/) ?

closed 2020-02-08 by Mikau28