libretro/Lakka-LibreELEC #843: No Sound in Emulation Cores but in Menu

Apfel567, 2019-11-11:
Hello dear Community, Im using Lakka generic x86-x64 2.3.1 on an Intel Atom System with 4GB RAM, flashed to a 120gb HDD. I'm trying to get the sound issues worked out, to get better audio I'm using a USB-Soundcard and the sound in the Menu (Backgroun, Select and Cancel Sounds) works fine, but when im starting an Emulating Core (Mame 2010, SNES, N64, SMD and PcEngine tested) no Audio is being outputted. I looked Online for other configuration Methods but only found info to general Sound issues, leaving me to believe this is a bug. Please let me know if you need additional logs or config Files. Thank you in advance.

Apfel567, 2019-11-11:
after a few try's changing the Audio driver I eventually got it working. I Switched from alsathread to tinyalsa - no luck, but after switching back to alsathread the sound worked after an additional restart. I'm not sure what did the trick exactly but the Audio is now working in all Cores.

closed 2019-11-11 by Apfel567

natinusala, 2019-11-12:
For the record the menu sounds aren't reliable and can cause issues such as this one, I would disable them in case any other audio issue occurs

Apfel567, 2019-11-12:
Thank you for the additional Input, I'll keep it in mind. That said I really like the Menu Theme and Sounds, so im really happy that everything works now.