libretro/Lakka-LibreELEC #844: Framecounter/stadistics overlay shows N/A

Xzesstool, 2019-11-13:
Some bugs I have found:
- Framerate counter overlay shows N/A instead of the framerate, and stadistics ram etc... are not shown.
- When selecting "download content" from "Online updater" menú, the meny goes crazy.
-I don't know if it is a bug, but I cannot scan roms from MAME, it blocks on random roms, I also slipted roms in folders of 1000, but again in a random rom it gets stuck. Snes, megadrive scan works perfect even with +5000 rom in a folder.

natinusala, 2019-11-13:
> Framerate counter overlay shows N/A instead of the framerate

Is this solved if you disable threaded video?

> When selecting "download content" from "Online updater" menú, the meny goes crazy.

Crazy in what way?

> I don't know if it is a bug, but I cannot scan roms from MAME, it blocks on random roms

Blocks?

ToKe79, 2019-11-13:
I think I had similar issue in the past, when the folder contained .chd files iirc. I solved it by moving temporarily the .chd files to different (parent) folder and then added them back after the scan of that folder was complete.

natinusala, 2019-11-13:
Isn't scanning done on a separate thread though? It shouldn't block even with heavy CHDs to decompress

Xzesstool, 2019-11-13:
> Is this solved if you disable threaded video?
>
>
> Crazy in what way?
>
>
> Blocks?

Thanks for the replys to all.

- Yes, with threaded video off FPS counter works, but now i realize that vsync doesn't work at least in that mode, frames keep going up and down.

- Rom scanning solved by moving chd files to another folder

- I went to take a photo on crazy menú in "download content" and it was okey now maybe was a server fault when searching for content.

closed 2020-08-18 by Ntemis