libretro/Lakka-LibreELEC #845: RPi4 Firmware version.

acidtech, 2019-11-14:
- Which version of Lakka are you using?  Latest nightly, 11/4
- What system hardware are you using? RPi 4
- What did you do? Setup DPI LCD 18bit mode 2 and PWM audio on GPIO18/19
- What did you expect to happen? Audio to work
- What happened instead? Audio did not work. Conflict in dtoverlay

Because the current master Lakka build is using old RPi 4 firmware(from July 9) the dtoverlay=audremap,args config.txt settings is not available.  So you cant change PWM audio to GPIO18/19 which means GPIO12 conflicts with DPI 18bit mode 2 which means the audio driver does not load.

I've moved to the LE9.2 fork which is using a newer firmware(and kernel) and properly supports the pwm audio remap function but I'd prefer to be using the master.

natinusala, 2019-11-15:
What do you mean by firmware exactly?

acidtech, 2019-11-15:
The RPi Firwmare: https://github.com/raspberrypi/firmware

Nathan Scherdin

natinusala, 2019-11-16:
> This repository contains pre-compiled binaries of the current Raspberry Pi kernel and modules, userspace libraries, and bootloader/GPU firmware.

I need to know what exactly in this repository needs to be updated 😅

acidtech, 2019-11-16:
Sorry about not being more specific before.  I assumed you were familiar
with the RPi build setup.  I'm just getting into Lakk/LibreElec but it
appears the linux package needs to be updated.

In the case of Lakka-LE9.2 it is pulling 4.19.66, in the case of the master
branch it is pulling 4.19.50.  The current stable releas is 4.19.75.

The Raspbery PI firmware/bootloader is included in each release so I
believe all you have to do is change the linux package.mk so it grabs the
latest stable release of linux for RPi.  I dont know how to get the archive
links though.  I know how to get the release links:
https://github.com/raspberrypi/linux/releases.

It may be as easy as pointing to the latest stable release
tar.gz(1.20190925-1), but I assume there are other kernel version specific
steps in building the branch and I'm not familiary enough with
Lakka/LibreElec to know where those may be in the build steps.

Thanks for any help you can give me.

Nathan Scherdin

natinusala, 2019-11-16:
So to be clear we are talking about the kernel and not the firmware, right?

The kernel we are using is built from raspberrypi/linux (and not downloaded from raspberrypi/firmware)

acidtech, 2019-11-17:
The Raspberry Pi foundation fixes bugs by releasing updated firmware
binaries.  They decided what kernel version to compile the firmware
binaries for. There is NO source code to the firmware so if you want the
bug fixes they release, you must use the kernel version they compile for.

Since you are using old firmware with known bugs you have two choices.
Stick with this kernel version and receive no more bug fixes from the
Raspberry Pi Foundation, OR update to the firmware version they are using
to generate their binaries(4.19.75 as of yesterday).  There is no third
choice.

Nathan Scherdin

natinusala, 2019-11-18:
Can't I just update our compiled kernel and the firmware?

acidtech, 2019-11-18:
That is what I suspect, but I'm not familiar enough with Lakka to know for
sure.   If there are any kernel version specific settings that need to be
updated along with changing the linux package.mk file.

Nathan Scherdin

closed 2020-08-18 by Ntemis