libretro/Lakka-LibreELEC #846: no sound through HDMI x86_64 live usb

clach04, 2019-11-15:
- Which version of Lakka are you using? Lakka v2.3.1
- What system hardware are you using? x86_64
- What did you do? booted and tried https://www.lakka.tv/doc/audio-settings/#switching-audio-device-from-the-gui - audio works through front audio jack
- What did you expect to happen? audio to work with hdmi
- What happened instead? audio works through front audio jack

I've asolated this to outside of retroarch. aplay will successful play with audio jack but not hdmi if a wav file is downloaded and tested. e.g.:

    cd /tmp ; wget https://file-examples.com/wp-content/uploads/2017/11/file_example_WAV_1MG.wav
    aplay -vv file_example_WAV_1MG.wav  # works

Here is the list of devices according to aplay:

        Lakka:/tmp # aplay -L
        null
            Discard all samples (playback) or generate zero samples (capture)
        default:CARD=PCH
            HDA Intel PCH, ALC891 Analog
            Default Audio Device
        sysdefault:CARD=PCH
            HDA Intel PCH, ALC891 Analog
            Default Audio Device
        front:CARD=PCH,DEV=0
            HDA Intel PCH, ALC891 Analog
            Front speakers
        surround21:CARD=PCH,DEV=0
            HDA Intel PCH, ALC891 Analog
            2.1 Surround output to Front and Subwoofer speakers
        surround40:CARD=PCH,DEV=0
            HDA Intel PCH, ALC891 Analog
            4.0 Surround output to Front and Rear speakers
        surround41:CARD=PCH,DEV=0
            HDA Intel PCH, ALC891 Analog
            4.1 Surround output to Front, Rear and Subwoofer speakers
        surround50:CARD=PCH,DEV=0
            HDA Intel PCH, ALC891 Analog
            5.0 Surround output to Front, Center and Rear speakers
        surround51:CARD=PCH,DEV=0
            HDA Intel PCH, ALC891 Analog
            5.1 Surround output to Front, Center, Rear and Subwoofer speakers
        surround71:CARD=PCH,DEV=0
            HDA Intel PCH, ALC891 Analog
            7.1 Surround output to Front, Center, Side, Rear and Woofer speakers
        hdmi:CARD=HDMI,DEV=0
            HDA ATI HDMI, HDMI 0
            HDMI Audio Output
        hdmi:CARD=HDMI,DEV=1
            HDA ATI HDMI, HDMI 1
            HDMI Audio Output
        hdmi:CARD=HDMI,DEV=2
            HDA ATI HDMI, HDMI 2
            HDMI Audio Output
        hdmi:CARD=HDMI,DEV=3
            HDA ATI HDMI, HDMI 3
            HDMI Audio Output
        hdmi:CARD=HDMI,DEV=4
            HDA ATI HDMI, HDMI 4
            HDMI Audio Output

gfx card:

        Lakka:/tmp # lspci -nnk | grep -A 3 VGA
        01:00.0 VGA compatible controller [0300]: Advanced Micro Devices, Inc. [AMD/ATI] Device [1002:699f] (rev c7)
                Subsystem: Hewlett-Packard Company Device [103c:8424]
                Kernel driver in use: amdgpu
        01:00.1 Audio device [0403]: Advanced Micro Devices, Inc. [AMD/ATI] Device [1002:aae0]

These work:

    aplay -vv test.wav  -Ddefault:CARD=PCH
    aplay -vv test.wav  -Dsysdefault:CARD=PCH

audio comes out of front audio jack.

Interestingly, this fails:

    Lakka:/tmp # aplay -vv test.wav  -Dfront:CARD=PCH,DEV=0
    aplay: main:807: audio open error: Device or resource busy

if I shutdown Retroarch then front does no longer errors:

    systemctl stop retroarch.service
    aplay -vv test.wav  -Dfront:CARD=PCH,DEV=0

BUT audio is distorted. This maybe a red-herring :(

All the following fail in the same way:

        Lakka:/tmp # aplay -vv test.wav  -Dfront:CARD=PCH,DEV=0
        aplay: main:807: audio open error: Device or resource busy
        Lakka:/tmp # aplay -vv test.wav  -Dsurround21:CARD=PCH,DEV=0
        aplay: main:807: audio open error: Device or resource busy
        Lakka:/tmp # aplay -vv test.wav  -Dsurround40:CARD=PCH,DEV=0
        aplay: main:807: audio open error: Device or resource busy
        Lakka:/tmp # aplay -vv test.wav  -Dsurround41:CARD=PCH,DEV=0
        aplay: main:807: audio open error: Device or resource busy
        Lakka:/tmp # aplay -vv test.wav  -Dsurround50:CARD=PCH,DEV=0
        aplay: main:807: audio open error: Device or resource busy
        Lakka:/tmp # aplay -vv test.wav  -Dsurround51:CARD=PCH,DEV=0
        aplay: main:807: audio open error: Device or resource busy
        Lakka:/tmp # aplay -vv test.wav  -Dsurround71:CARD=PCH,DEV=0
        aplay: main:807: audio open error: Device or resource busy

The hdmi devices do not error BUT audio is silent, i.e. nothing heard through hdmi (and as expected nothing through front audio jack):

        aplay -vv test.wav  -Dhdmi:CARD=HDMI,DEV=0
        aplay -vv test.wav  -Dhdmi:CARD=HDMI,DEV=1
        aplay -vv test.wav  -Dhdmi:CARD=HDMI,DEV=2
        aplay -vv test.wav  -Dhdmi:CARD=HDMI,DEV=3
        aplay -vv test.wav  -Dhdmi:CARD=HDMI,DEV=4

I'm not really sure what else to test/other diagnostics to try.

speaker-test is present in Lakka-LibreELEC and again works with default BUT when ran with hdmi I get errors (unlike with aplay). Also front fails to play too (whether retroarch is runing or not)

        Lakka:/tmp # speaker-test  -Ddefault:CARD=PCH

        speaker-test 1.1.4

        Playback device is default:CARD=PCH
        Stream parameters are 48000Hz, S16_LE, 1 channels
        Using 16 octaves of pink noise
        Rate set to 48000Hz (requested 48000Hz)
        Buffer size range from 2048 to 16384
        Period size range from 1024 to 1024
        Using max buffer size 16384
        Periods = 4
        was set period_size = 1024
        was set buffer_size = 16384
         0 - Front Left
        ^CWrite error: -4,Interrupted system call
        xrun_recovery failed: -4,Interrupted system call
        Transfer failed: Interrupted system call
        Lakka:/tmp # speaker-test  -Dhdmi:CARD=HDMI,DEV=2

        speaker-test 1.1.4

        Playback device is hdmi:CARD=HDMI,DEV=2
        Stream parameters are 48000Hz, S16_LE, 1 channels
        Using 16 octaves of pink noise
        Channels count (1) not available for playbacks: Invalid argument
        Setting of hwparams failed: Invalid argument

Booting from a Lubuntu live USB stick audio works fine over hdmi (older versions of lubuntu would default to front but could be changed to hdmi).

I'm happy to take other diags.  

Not sure if this is similar to issue #321.

gouchi, 2019-11-15:
Hi,

Thank you for the information. Do you reproduce also the issue with [Lakka from LE 9.2](http://nightly.builds.lakka.tv/Lakka-LE9.2/latest/Generic.x86_64/) ?

Also please provide `dmesg` output.

Thank  you.

clach04, 2019-11-16:
@gouchi sadly no change. I pulled down http://nightly.builds.lakka.tv/Lakka-LE9.2/latest/Generic.x86_64/

    Lakka-Generic.x86_64-2.9-devel-20190917212918-de642b3.img.gz	2019-09-17 21:32 	409M	 

Same devices showing (I skipped the config.txt step and jumped straight to using aplay and speaker-test. I did check Lubuntu and noticed its using PulseAudio (aplay shows a bunch more devices). The good news is that the front jack still worked :)

I've attached the dmesg output 
[dmesg.txt](https://github.com/libretro/Lakka-LibreELEC/files/3853687/dmesg.txt) file rather than doing inline due tolength, aplay -L output below to show what looks like the same output:

    Lakka:~ # aplay -L
    null
        Discard all samples (playback) or generate zero samples (capture)
    default:CARD=PCH
        HDA Intel PCH, ALC891 Analog
        Default Audio Device
    sysdefault:CARD=PCH
        HDA Intel PCH, ALC891 Analog
        Default Audio Device
    front:CARD=PCH,DEV=0
        HDA Intel PCH, ALC891 Analog
        Front speakers
    surround21:CARD=PCH,DEV=0
        HDA Intel PCH, ALC891 Analog
        2.1 Surround output to Front and Subwoofer speakers
    surround40:CARD=PCH,DEV=0
        HDA Intel PCH, ALC891 Analog
        4.0 Surround output to Front and Rear speakers
    surround41:CARD=PCH,DEV=0
        HDA Intel PCH, ALC891 Analog
        4.1 Surround output to Front, Rear and Subwoofer speakers
    surround50:CARD=PCH,DEV=0
        HDA Intel PCH, ALC891 Analog
        5.0 Surround output to Front, Center and Rear speakers
    surround51:CARD=PCH,DEV=0
        HDA Intel PCH, ALC891 Analog
        5.1 Surround output to Front, Center, Rear and Subwoofer speakers
    surround71:CARD=PCH,DEV=0
        HDA Intel PCH, ALC891 Analog
        7.1 Surround output to Front, Center, Side, Rear and Woofer speakers
    hdmi:CARD=HDMI,DEV=0
        HDA ATI HDMI, HDMI 0
        HDMI Audio Output
    hdmi:CARD=HDMI,DEV=1
        HDA ATI HDMI, HDMI 1
        HDMI Audio Output
    hdmi:CARD=HDMI,DEV=2
        HDA ATI HDMI, HDMI 2
        HDMI Audio Output
    hdmi:CARD=HDMI,DEV=3
        HDA ATI HDMI, HDMI 3
        HDMI Audio Output
    hdmi:CARD=HDMI,DEV=4
        HDA ATI HDMI, HDMI 4
        HDMI Audio Output

gouchi, 2019-11-17:
Hi,

Thank you for the information. You will have to use Lakka from LE92 because Lakka 2.3.1 kernel is too old for your gpu.

It is already enabled on Lakka from LE92 but can may have to force it by trying to boot with this boot parameters

`live ssh amdgpu.dc=1`

Thank you for your tests.

Thank you.

clach04, 2019-11-20:
Thanks @gouchi, do you mean at some point this will work or that this is expected to work now with LE92 build from 2019-09-17? I just tried this with the boot flags and got the same behavior from `aplay`.

I think this shows the boot flags used:

    Lakka:/tmp # dmesg |grep -i boot
    [    0.000000] smpboot: Allowing 8 CPUs, 0 hotplug CPUs
    [    0.000000] Booting paravirtualized kernel on bare hardware
    [    0.000000] Kernel command line: root=/dev/ram0 rdinit=/init usbcore.autosuspend=-1 BOOT_IMAGE=/KERNEL boot=UUID=2009-2802 disk=UUID=cedc39be-973a-4f04-a39e-4083b6673050 portable quiet vt.global_cursor_default=0 loglevel=2 ssh amdgpu.dc=1
    [    0.032808] smpboot: Max logical packages: 2
    [    0.046015] smpboot: CPU0: Intel(R) Core(TM) i7-7700 CPU @ 3.60GHz (family: 0x6, model: 0x9e, stepping: 0x9)
    [    0.075021] x86: Booting SMP configuration:
    [    0.561019] smpboot: Total of 8 processors activated (57632.59 BogoMIPS)
    [    0.788155] ACPI: \_SB_.PCI0.LPCB.PGEC: Used as boot DSDT EC to handle transactions
    [    0.827181] ACPI: \_SB_.PCI0.LPCB.PGEC: Used as boot DSDT EC to handle transactions and events
    [    0.827261] pci 0000:01:00.0: vgaarb: setting as boot VGA device

    Lakka:/tmp # uname -a
    Linux Lakka 4.11.12-rt14 #1 SMP PREEMPT RT Fri Sep 13 00:18:16 CEST 2019 x86_64 GNU/Linux

Anything else I need to do/test? This isn't urgent so I can do what ever tests are needed.

gouchi, 2019-11-21:
Hi,

It should work **only** with [Lakka LE92](http://nightly.builds.lakka.tv/Lakka-LE9.2/latest/Generic.x86_64/) . May you try to make the test with Lakka LE92 as you should be running kernel 5.1.18 with Lakka LE92.

Thank you.

clach04, 2019-11-22:
Hilariously I used the wrong USB drive for my previous test! Thanks for confirming the kernel number. LE92 without boot flag but explicitly named audio device in config works for retroarch (HDMI device 3) works great.

Interestingly speaker-test does not work and aplay makes a sound but it's corrupted/distorted. But retroarch is fine (once the correct device is set).

Thanks for your help and patience.

closed 2019-11-22 by clach04