libretro/Lakka-LibreELEC #847: No wifi due to missing firmware (rt2800pci)

gionn, 2019-11-16:
Lakka (official): 2.3.1 (Generic.i386)

```
02:00.0 Network controller: Ralink corp. RT3090 Wireless 802.11n 1T/1R PCIe
```

```
[  884.857560] ieee80211 phy0: rt2x00lib_request_firmware: Info - Loading firmware file 'rt2860.bin'
[  884.857630] rt2800pci 0000:02:00.0: Direct firmware load for rt2860.bin failed with error -2
[  884.857641] ieee80211 phy0: rt2x00lib_request_firmware: Error - Failed to request Firmware
```

Is there a way to manual embed the requested firmware as a workaround? Tried scp but filesystem is read-only :eagle:

gouchi, 2019-11-16:
Hi,

It should be included also on [i386](https://github.com/libretro/Lakka-LibreELEC/blob/master/packages/linux-firmware/kernel-firmware/firmwares/i386.dat) as it is linked to [x86_64.dat](https://github.com/libretro/Lakka-LibreELEC/blob/master/packages/linux-firmware/kernel-firmware/firmwares/x86_64.dat) which includes rt2860.bin firmware. What is the output of `/usr/lib/firmware` and `/lib/firmware` ?

Thank you.

gionn, 2019-11-17:
Folder is the same, there is a symlink:

```
Lakka:~ # ls -la /
total 8
drwxrwxr-x   12 root     root           203 Sep 20 11:30 .
drwxrwxr-x   12 root     root           203 Sep 20 11:30 ..
lrwxrwxrwx    1 root     root             8 Sep 20 11:30 bin -> /usr/bin
drwxr-xr-x   16 root     root          3640 Nov 17 15:51 dev
drwxrwxr-x   18 root     root           928 Sep 20 11:32 etc
drwxr-xr-x    3 root     root          4096 Nov 16 18:04 flash
lrwxrwxrwx    1 root     root             8 Sep 20 11:30 lib -> /usr/lib
lrwxrwxrwx    1 root     root            10 Sep 20 11:30 media -> /var/media
dr-xr-xr-x  183 root     root             0 Nov 17 15:51 proc
drwxr-xr-x   12 root     root           400 Nov 17 15:51 run
lrwxrwxrwx    1 root     root             9 Sep 20 11:30 sbin -> /usr/sbin
drwxr-xr-x   24 root     root          4096 Nov 16 19:29 storage
dr-xr-xr-x   12 root     root             0 Nov 17 15:51 sys
drwxrwxrwt   14 root     root           280 Nov 17 15:51 tmp
drwxrwxr-x    9 root     root           138 Sep 20 11:31 usr
drwxr-xr-x    9 root     root           200 Nov 17 15:51 var
```

Most firmwares are there:

> Lakka:~ # ls -l /lib/firmware/ | grep rt
> -rw-rw-r--    1 root     root         10757 Sep 20 11:30 dvb-usb-avertv-a800-02.fw
> -rw-rw-r--    1 root     root          3360 Sep 20 11:30 dvb-usb-tvwalkert.fw
> drwxrwxr-x    2 root     root           583 Sep 20 11:30 libertas
> -rw-rw-r--    1 root     root          8192 Sep 20 11:30 rt2870.bin
> -rw-rw-r--    1 root     root          2048 Sep 20 11:30 rt73.bin
> drwxrwxr-x    2 root     root           202 Sep 20 11:30 rtl_bt
> drwxrwxr-x    2 root     root           934 Sep 20 11:30 rtlwifi

But **rt2860.bin** is missing:
```
Lakka:~ # find /lib/firmware/ -name rt2860.bin
Lakka:~ #
```

Thanks.

gouchi, 2019-12-05:
Hi,

Thank you for the feedback.

@ToKe79 it seems the symbolic link to the [dat file for i386](https://github.com/libretro/Lakka-LibreELEC/blob/master/packages/linux-firmware/kernel-firmware/firmwares/i386.dat) is not [enough](https://github.com/libretro/Lakka-LibreELEC/blob/master/packages/linux-firmware/kernel-firmware/package.mk#L46) ?

ToKe79, 2019-12-05:
@gouchi @gionn is using stable release, the commit with the fix is after 2.3.1 release.
@gionn can you test [latest devel/nightly](http://nightly.builds.lakka.tv/2019-11-04/Generic.i386/) build?

closed 2020-08-18 by Ntemis